nordicsemi/Android-nRF-Mesh-Library #385: provisioning to proxy

Nuclear6, 2021-02-04:
I have a little doubt, why provisioning complete, need rescan mesh proxy service and then connect this device. Why not connect the device directly when completing the provisioning.

roshanrajaratnam, 2021-02-04:
This is because when a node is not provisioned, it is advertising with Mesh Provisioning Service and then the device switches to advertising with Mesh Proxy Service. These two separate services help us to identify nodes that are provisioned and not.

Also after provisioning we need to ensure we are connecting to a node that has the same Node identity or the Network ID, this is advertised with the Mesh Proxy Service.

closed 2021-02-04 by roshanrajaratnam

Nuclear6, 2021-02-04:
I got it, thanks!!!