nordicsemi/Android-nRF-Mesh-Library #389: Time Set Message

fiveagle, 2021-02-05:
Hi,
we are wondering if on Android side is the same regarding https://github.com/NordicSemiconductor/IOS-nRF-Mesh-Library/issues/320 ?
Is it feasible to add it in 3.1?
Or is it fairly simple to implement ourselves like in iOS? Can you give some example?

Thanks a lot.

roshanrajaratnam, 2021-02-07:
Hi @fiveagle the same applies, 3.0 will be submitted Monday morning and we plan to keep the releases and features in sync as much as possible. Pull requests are welcome.

Yes it's pretty simple. You can base it on [GeneicOnOffSetUnacknowledged](https://github.com/NordicSemiconductor/Android-nRF-Mesh-Library/blob/dev/mesh/src/main/java/no/nordicsemi/android/mesh/transport/GenericOnOffSetUnacknowledged.java#L39) and handle the message status [here](https://github.com/NordicSemiconductor/Android-nRF-Mesh-Library/blob/dev/mesh/src/main/java/no/nordicsemi/android/mesh/transport/DefaultNoOperationMessageState.java#L85).

roshanrajaratnam, 2021-12-17:
Hi @fiveagle #448 adds support for Time set message.

closed 2021-12-17 by roshanrajaratnam