nordicsemi/Android-nRF-Mesh-Library #399: Is provisioner check inside BaseMeshNetwork is wrong

Eklow02, 2021-02-15:
---

**Describe the bug**
The method ` public boolean isProvisioner(@NonNull final ProvisionedMeshNode node)` inside the `BaseMeshNetwork` return for all nodes inside the network true.  The method compares the meshUuuid but the id is always the same, because the uuid from the network is set after provisioning. 

**Expected behavior**
The check should compare the `provisioner.getProvisionerUuid()` with the `node.getUuid()`

**Platform details:**
 - Library Version `3.0.2`

roshanrajaratnam, 2021-02-16:
Hi, I did not notice this issue until now and I already released a 3.0.3. I'll take a look at this now.

roshanrajaratnam, 2021-02-16:
It's a stupid bug, I was comparing the network UUIDs instead of their own UUIDs, hence returning always true. Check out the latest Dev branch. You can close the issue, if it works for you.

Eklow02, 2021-02-17:
Yes, I was also a little bit confused 😅. The fix is working for me👍

commit 698eac2d (2021-02-16): Fixes #399. Should check for node's and the provisioner's uuid instead of the network uuid.
closed 2021-02-17 by Eklow02