nordicsemi/Android-nRF-Mesh-Library #400: An error thrown when export mesh network

KcLeung514, 2021-02-16:
---

**Describe the bug**
An error thrown when export mesh network at nRF Mesh App ( Android, v3.0.2 )

**To Reproduce**
Steps to reproduce the behavior:
1. Export mesh network json file from nRF Mesh App ( iOS, v3.0.2 )
2. Import this mesh network file at nRF Mesh App ( Android, v3.0.2 )
3. Try to export mesh network json file from nRF Mesh App ( Android, v3.0.2 )
4. See error

**Expected behavior**
Export mesh network without error.

**Platform details:**
 - Device: Pixel 3 XL
 - OS: Android 11
 - nRF Mesh v3.0.2 ( Android )

**Logs / Screenshots**

[image]

meshNetworkiOS.json
```
{
    "$schema": "http://json-schema.org/draft-04/schema#",
    "appKeys": [
        {
            "boundNetKey": 0,
            "index": 0,
            "key": "16C8BD5597427D6D3BD24F8EC729F45A",
            "name": "control"
        }
    ],
    "groups": [
        {
            "address": "C000",
            "name": "C000",
            "parentAddress": "0000"
        }
    ],
    "id": "http://www.bluetooth.com/specifications/assigned-numbers/mesh-profile/cdb-schema.json#",
    "meshName": "nRF Mesh Network",
    "meshUUID": "21516719-675D-440A-AA5D-693CD6176F55",
    "netKeys": [
        {
            "index": 0,
            "key": "01CA24C73A852420CC6BDC86200B0C3A",
            "minSecurity": "secure",
            "name": "Primary Network Key",
            "phase": 0,
            "timestamp": "2021-02-16T02:16:42Z"
        },
        {
            "index": 1,
            "key": "992924CAFCCD3192B22C6C59B9E41D60",
            "minSecurity": "secure",
            "name": "main",
            "phase": 0,
            "timestamp": "2021-02-16T02:16:42Z"
        }
    ],
    "networkExclusions": [
        {
            "addresses": [
                "0001"
            ],
            "ivIndex": 0
[... 242 lines cut ...]
                    "highAddress": "CE9A",
                    "lowAddress": "CC9B"
                }
            ],
            "allocatedSceneRange": [
                {
                    "firstScene": "3334",
                    "lastScene": "3B33"
                }
            ],
            "allocatedUnicastRange": [
                {
                    "highAddress": "1D9A",
                    "lowAddress": "199B"
                }
            ],
            "provisionerName": "C2EA6790-6FF3-11EB-BC17-0FFE9F546DC1",
            "UUID": "C2EA6790-6FF3-11EB-BC17-0FFE9F546DC1"
        }
    ],
    "scenes": [],
    "timestamp": "2021-02-16T02:16:56Z",
    "version": "1.0.0"
}
```

roshanrajaratnam, 2021-02-16:
Thanks for reporting this issue, await a fix :)

roshanrajaratnam, 2021-02-16:
Version 3.0.3 is now published.

closed 2021-02-16 by roshanrajaratnam