nordicsemi/Android-nRF-Mesh-Library #403: Null Pointer Exception : InternalTransportCallbacks.onMeshPduCreated

Mavericksb, 2021-02-26:
---
Hello,
during Provisioning phase we experienced occasional crashes.

Here you can find the stacktrace log :

**StackTrace
Fatal Exception: java.lang.NullPointerException: Attempt to invoke interface method 'void no.nordicsemi.android.meshprovisioner.InternalTransportCallbacks.onMeshPduCreated(int, byte[])' on a null object reference
       at no.nordicsemi.android.meshprovisioner.transport.MeshMessageState.sendSegmentAcknowledgementMessage(MeshMessageState.java:146)
       at no.nordicsemi.android.meshprovisioner.transport.LowerTransportLayer.sendBlockAck(LowerTransportLayer.java:756)
       at no.nordicsemi.android.meshprovisioner.transport.LowerTransportLayer.lambda$initSegmentedAccessAcknowledgementTimer$0$LowerTransportLayer(LowerTransportLayer.java:704)
       at no.nordicsemi.android.meshprovisioner.transport.-$$Lambda$LowerTransportLayer$OqXRgSPSKQR6i6Ap1Fez71k04mA.run(:12)
       at android.os.Handler.handleCallback(Handler.java:873)
       at android.os.Handler.dispatchMessage(Handler.java:99)
       at android.os.Looper.loop(Looper.java:201)
       at android.app.ActivityThread.main(ActivityThread.java:6815)
       at java.lang.reflect.Method.invoke(Method.java)
       at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:547)
       at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:873)

**Platform details:**
 - OS: Android
 - Nordic Library Version  : 
implementation 'no.nordicsemi.android.support.v18:scanner:1.4.3'
implementation 'no.nordicsemi.android:ble:2.1.1'
implementation 'no.nordicsemi.android:dfu:1.9.0'

roshanrajaratnam, 2021-02-26:
Hi what's the mesh library version you are using and have you made any changes to the internals?

closed 2021-03-22 by roshanrajaratnam

Eklow02, 2021-08-06:
Hi, I get the same error with version 3.1.6

It looks like the error occurs when I try to send several messages in quick succession.