nordicsemi/Android-nRF-Mesh-Library #406: Message reassembly may not be completed yet

kcjian, 2021-03-08:
---

**Describe the bug**
I have a packet of data not received, the log shows `Message reassembly may not be completed yet`

**To Reproduce**
Steps to reproduce the behavior:
1. sendVendorModelMessage:0xF20208FF
2. VendorModelMessageStatus 0xF7C8091C1208FF005515111171777770111111010011DD
3. sendVendorModelMessage:0xF4070EFF0203041E1F
4. received  many `Message reassembly may not be completed yet`

**Expected behavior**
I will receive a data response 0xF7C8091C0C0EFF0200030004001E001F00

**Platform details:**
 - Device: HuaWei P40
 - OS: Android 10
 - Library Version ：3.1.0

The app I developed based on version 2.4.4 can be received normally， I suspect that `LowerTransportLayer.sendBlockAck` has affected the normal data sending and receiving 

**Logs / Screenshots**
```
2021-03-08 20:17:37.601 8802-8802/com.xuntong.meshuart E/MeshService: sendVendorModelMessage:0xF20208FF
2021-03-08 20:17:37.602 8802-8802/com.xuntong.meshuart V/MeshTransport: Src address: 1996
2021-03-08 20:17:37.602 8802-8802/com.xuntong.meshuart V/MeshTransport: Dst address: 0001
2021-03-08 20:17:37.602 8802-8802/com.xuntong.meshuart V/MeshTransport: Key: EFD43206242335D66889D03CDA2DDE9E
2021-03-08 20:17:37.602 8802-8802/com.xuntong.meshuart V/MeshTransport: akf: 1
2021-03-08 20:17:37.602 8802-8802/com.xuntong.meshuart V/MeshTransport: aid: 28
2021-03-08 20:17:37.602 8802-8802/com.xuntong.meshuart V/MeshTransport: aszmic: 0
2021-03-08 20:17:37.602 8802-8802/com.xuntong.meshuart V/MeshTransport: Sequence number: 176
2021-03-08 20:17:37.602 8802-8802/com.xuntong.meshuart V/MeshTransport: Access message opcode: f4
2021-03-08 20:17:37.602 8802-8802/com.xuntong.meshuart V/MeshTransport: Access message parameters: F20208FF
2021-03-08 20:17:37.602 8802-8802/com.xuntong.meshuart V/AccessLayer: Created Access PDU F4C809F20208FF
2021-03-08 20:17:37.602 8802-8802/com.xuntong.meshuart V/UpperTransportLayer: Application nonce: 01000000B01996000100000000
2021-03-08 20:17:37.602 8802-8802/com.xuntong.meshuart V/UpperTransportLayer: Encrypted upper transport pdu: ABE24C0A1E501944103D44
2021-03-08 20:17:37.602 8802-8802/com.xuntong.meshuart V/LowerTransportLayer: Unsegmented Lower transport access PDU 5CABE24C0A1E501944103D44
2021-03-08 20:17:37.602 8802-8802/com.xuntong.meshuart V/NetworkLayer: Encryption key: 93C7471C17A1B9C8A2B6166D79B8231D
2021-03-08 20:17:37.602 8802-8802/com.xuntong.meshuart V/NetworkLayer: Privacy key: 2ABD84D31F57501D1D44525CB10BA231
2021-03-08 20:17:37.602 8802-8802/com.xuntong.meshuart V/NetworkLayer: Sequence Number: 0000B0
2021-03-08 20:17:37.602 8802-8802/com.xuntong.meshuart V/NetworkLayer: Encrypted Network payload: ACC0FF0103445E25CD0BFA26BA18FD3EA736
2021-03-08 20:17:37.602 8802-8802/com.xuntong.meshuart V/VendorModelMessageAckedState: Sending acknowledged vendor model message
2021-03-08 20:17:37.602 8802-8802/com.xuntong.meshuart E/MeshManagerApi: updateNetwork: 1
2021-03-08 20:17:37.603 8802-8802/com.xuntong.meshuart V/BleManager: Writing characteristic 00002add-0000-1000-8000-00805f9b34fb (WRITE COMMAND)
2021-03-08 20:17:37.603 8802-8802/com.xuntong.meshuart D/BleManager: gatt.writeCharacteristic(00002add-0000-1000-8000-00805f9b34fb)
2021-03-08 20:17:37.603 8802-8802/com.xuntong.meshuart D/BluetoothGatt: writeCharacteristic() - uuid: 00002add-0000-1000-8000-00805f9b34fb
2021-03-08 20:17:37.603 8802-9962/com.xuntong.meshuart D/OpenGLRenderer: disableOutlineDraw is true
2021-03-08 20:17:37.604 8802-9909/com.xuntong.meshuart D/BluetoothGatt: onCharacteristicWrite() - Device=F9:FA:4D:**:**:** handle=16 Status=0
2021-03-08 20:17:37.619 8802-10110/com.xuntong.meshuart D/HiTouch_PressGestureDetector: onAttached, package=com.xuntong.meshuart, windowType=1, mHiTouchRestricted=false
2021-03-08 20:17:37.626 8802-9962/com.xuntong.meshuart D/mali_winsys: EGLint new_window_surface(egl_winsys_display *, void *, EGLSurface, EGLConfig, egl_winsys_surface **, EGLBoolean) returns 0x3000
2021-03-08 20:17:37.637 8802-8802/com.xuntong.meshuart I/HwViewRootImpl: removeInvalidNode all the node in jank list is out of time
2021-03-08 20:17:37.638 8802-8802/com.xuntong.meshuart I/BleManager: Data written to 00002add-0000-1000-8000-00805f9b34fb, value: (0x) 00-29-8E-75-C5-A4-C4-EE-AC-C0-FF-01-03-44-5E-25-CD-0B-FA-26-BA-18-FD-3E-A7-36
2021-03-08 20:17:37.638 8802-8802/com.xuntong.meshuart V/MeshManagerApi: MeshNetwork pdu sent: 0x00298E75C5A4C4EEACC0FF0103445E25CD0BFA26BA18FD3EA736
2021-03-08 20:17:37.639 8802-8802/com.xuntong.meshuart D/DecorView: showOrHideHighlightView: hasFocus=true; winMode=1; isMrgNull=true
2021-03-08 20:17:37.639 8802-8802/com.xuntong.meshuart W/InputMethodManager: startInputReason = 1
2021-03-08 20:17:37.845 8802-9909/com.xuntong.meshuart D/BluetoothGatt: onNotify() - Device=F9:FA:4D:**:**:** handle=18
2021-03-08 20:17:37.845 8802-8802/com.xuntong.meshuart I/BleManager: Notification received from 00002ade-0000-1000-8000-00805f9b34fb, value: (0x) 00-29-8E-7B-E3-9B-F1-2B-DF-C6-6D-AA-A5-36-57-52-94-F0-55-6D-C9-9A-DA-4C-F2-DD-04-9A-46-E0
2021-03-08 20:17:37.845 8802-8802/com.xuntong.meshuart V/MeshManagerApi: Received network pdu: 0x00298E7BE39BF12BDFC66DAAA536575294F0556DC99ADA4CF2DD049A46E0
2021-03-08 20:17:37.845 8802-8802/com.xuntong.meshuart V/BaseMeshMessageHandler: TTL for received message: 5
2021-03-08 20:17:37.846 8802-8802/com.xuntong.meshuart V/BaseMeshMessageHandler: Sequence number of received access message: 244
2021-03-08 20:17:37.846 8802-8802/com.xuntong.meshuart V/NetworkLayer: TTL for received message: 5
2021-03-08 20:17:37.846 8802-8802/com.xuntong.meshuart V/NetworkLayer: Dst: 0x1996
2021-03-08 20:17:37.846 8802-8802/com.xuntong.meshuart V/NetworkLayer: Received a segmented access message from: 0001
2021-03-08 20:17:37.846 8802-8802/com.xuntong.meshuart V/LowerTransportLayer: SEG O: 0
2021-03-08 20:17:37.846 8802-8802/com.xuntong.meshuart V/LowerTransportLayer: SEG N: 2
2021-03-08 20:17:37.846 8802-8802/com.xuntong.meshuart V/LowerTransportLayer: Last SeqAuth value 242
2021-03-08 20:17:37.846 8802-8802/com.xuntong.meshuart V/LowerTransportLayer: Current SeqAuth value 244
2021-03-08 20:17:37.846 8802-8802/com.xuntong.meshuart V/LowerTransportLayer: Starting incomplete timer for src: 0001
2021-03-08 20:17:37.846 8802-8802/com.xuntong.meshuart V/BlockAcknowledgementMessage: Block ack value: 1
[... 184 lines cut ...]
2021-03-08 20:17:40.677 8802-8802/com.xuntong.meshuart V/BaseMeshMessageHandler: Sequence number of received access message: 252
2021-03-08 20:17:40.678 8802-8802/com.xuntong.meshuart V/NetworkLayer: TTL for received message: 5
2021-03-08 20:17:40.679 8802-8802/com.xuntong.meshuart V/NetworkLayer: Dst: 0x1996
2021-03-08 20:17:40.680 8802-8802/com.xuntong.meshuart V/NetworkLayer: Received a segmented access message from: 0001
2021-03-08 20:17:40.680 8802-8802/com.xuntong.meshuart V/LowerTransportLayer: SEG O: 1
2021-03-08 20:17:40.680 8802-8802/com.xuntong.meshuart V/LowerTransportLayer: SEG N: 2
2021-03-08 20:17:40.680 8802-8802/com.xuntong.meshuart V/LowerTransportLayer: Last SeqAuth value 244
2021-03-08 20:17:40.680 8802-8802/com.xuntong.meshuart V/LowerTransportLayer: Current SeqAuth value 244
2021-03-08 20:17:40.681 8802-8802/com.xuntong.meshuart V/LowerTransportLayer: Ignoring message since the incomplete timer has expired and all messages have been received
2021-03-08 20:17:40.681 8802-8802/com.xuntong.meshuart V/DefaultNoOperationMessageState: Message reassembly may not be completed yet!
2021-03-08 20:17:41.504 8802-9909/com.xuntong.meshuart D/BluetoothGatt: onNotify() - Device=F9:FA:4D:**:**:** handle=18
2021-03-08 20:17:41.505 8802-8802/com.xuntong.meshuart I/BleManager: Notification received from 00002ade-0000-1000-8000-00805f9b34fb, value: (0x) 00-29-87-16-F1-46-FE-F0-3F-88-72-26-36-5E-96-BF-CA-07-D5-28-39-2D-50-97-84-F9-2F-DC-A9-5C
2021-03-08 20:17:41.506 8802-8802/com.xuntong.meshuart V/MeshManagerApi: Received network pdu: 0x00298716F146FEF03F887226365E96BFCA07D528392D509784F92FDCA95C
2021-03-08 20:17:41.507 8802-8802/com.xuntong.meshuart V/BaseMeshMessageHandler: TTL for received message: 5
2021-03-08 20:17:41.507 8802-8802/com.xuntong.meshuart V/BaseMeshMessageHandler: Sequence number of received access message: 253
2021-03-08 20:17:41.509 8802-8802/com.xuntong.meshuart V/NetworkLayer: TTL for received message: 5
2021-03-08 20:17:41.510 8802-8802/com.xuntong.meshuart V/NetworkLayer: Dst: 0x1996
2021-03-08 20:17:41.511 8802-8802/com.xuntong.meshuart V/NetworkLayer: Received a segmented access message from: 0001
[... cut ...]
2021-03-08 20:17:41.512 8802-8802/com.xuntong.meshuart V/LowerTransportLayer: Last SeqAuth value 244
2021-03-08 20:17:41.512 8802-8802/com.xuntong.meshuart V/LowerTransportLayer: Current SeqAuth value 244
2021-03-08 20:17:41.512 8802-8802/com.xuntong.meshuart V/LowerTransportLayer: Ignoring message since the incomplete timer has expired and all messages have been received
2021-03-08 20:17:41.512 8802-8802/com.xuntong.meshuart V/DefaultNoOperationMessageState: Message reassembly may not be completed yet!
```

kcjian, 2021-03-09:
Use nRFMesh:

app send : `Created Access PDU F4C809000208FF`
app received: `AccessLayer: Received Access PDU F7C809001208FF005515111171777770111111010011DD`
and many `Message reassembly may not be completed yet`

The logs：
```
2021-03-09 09:53:18.866 13875-13875/no.nordicsemi.android.nrfmeshprovisioner V/AudioManager: querySoundEffectsEnabled...
2021-03-09 09:53:18.869 13875-13875/no.nordicsemi.android.nrfmeshprovisioner V/MeshTransport: Src address: 0001
2021-03-09 09:53:18.869 13875-13875/no.nordicsemi.android.nrfmeshprovisioner V/MeshTransport: Dst address: 0003
2021-03-09 09:53:18.869 13875-13875/no.nordicsemi.android.nrfmeshprovisioner V/MeshTransport: Key: A3F854F28F1A810D9E4BC53F6834A55E
2021-03-09 09:53:18.869 13875-13875/no.nordicsemi.android.nrfmeshprovisioner V/MeshTransport: akf: 1
2021-03-09 09:53:18.869 13875-13875/no.nordicsemi.android.nrfmeshprovisioner V/MeshTransport: aid: 44
2021-03-09 09:53:18.869 13875-13875/no.nordicsemi.android.nrfmeshprovisioner V/MeshTransport: aszmic: 0
2021-03-09 09:53:18.869 13875-13875/no.nordicsemi.android.nrfmeshprovisioner V/MeshTransport: Sequence number: 28
2021-03-09 09:53:18.870 13875-13875/no.nordicsemi.android.nrfmeshprovisioner V/MeshTransport: Access message opcode: f4
2021-03-09 09:53:18.870 13875-13875/no.nordicsemi.android.nrfmeshprovisioner V/MeshTransport: Access message parameters: 000208FF
2021-03-09 09:53:18.870 13875-13875/no.nordicsemi.android.nrfmeshprovisioner V/AccessLayer: Created Access PDU F4C809000208FF
2021-03-09 09:53:18.870 13875-13875/no.nordicsemi.android.nrfmeshprovisioner V/UpperTransportLayer: Application nonce: 010000001C0001000300000000
2021-03-09 09:53:18.870 13875-13875/no.nordicsemi.android.nrfmeshprovisioner V/UpperTransportLayer: Encrypted upper transport pdu: E866656E74062138C163A7
2021-03-09 09:53:18.870 13875-13875/no.nordicsemi.android.nrfmeshprovisioner V/LowerTransportLayer: Unsegmented Lower transport access PDU 6CE866656E74062138C163A7
2021-03-09 09:53:18.871 13875-13875/no.nordicsemi.android.nrfmeshprovisioner V/NetworkLayer: Encryption key: A7714CCF02A91A41D7EA66AB19D57353
2021-03-09 09:53:18.871 13875-13875/no.nordicsemi.android.nrfmeshprovisioner V/NetworkLayer: Privacy key: 45DC16BA34D90B18071F0FA583832199
2021-03-09 09:53:18.871 13875-13875/no.nordicsemi.android.nrfmeshprovisioner V/NetworkLayer: Sequence Number: 00001C
2021-03-09 09:53:18.871 13875-13875/no.nordicsemi.android.nrfmeshprovisioner V/NetworkLayer: Encrypted Network payload: D6D120F8CCCA17D2F0637A851E98BC46573B
2021-03-09 09:53:18.871 13875-13875/no.nordicsemi.android.nrfmeshprovisioner V/VendorModelMessageUnackedState: Sending acknowledged vendor model message
2021-03-09 09:53:18.871 13875-13875/no.nordicsemi.android.nrfmeshprovisioner V/BleManager: Writing characteristic 00002add-0000-1000-8000-00805f9b34fb (WRITE COMMAND)
2021-03-09 09:53:18.872 13875-13875/no.nordicsemi.android.nrfmeshprovisioner D/BleManager: gatt.writeCharacteristic(00002add-0000-1000-8000-00805f9b34fb)
2021-03-09 09:53:18.872 13875-13875/no.nordicsemi.android.nrfmeshprovisioner D/BluetoothGatt: writeCharacteristic() - uuid: 00002add-0000-1000-8000-00805f9b34fb
2021-03-09 09:53:18.873 13875-14675/no.nordicsemi.android.nrfmeshprovisioner D/BluetoothGatt: onCharacteristicWrite() - Device=F9:FA:4D:**:**:** handle=16 Status=0
2021-03-09 09:53:18.874 13875-14675/no.nordicsemi.android.nrfmeshprovisioner I/BleManager: Data written to 00002add-0000-1000-8000-00805f9b34fb, value: (0x) 00-5C-7C-AC-15-4A-DF-FD-D6-D1-20-F8-CC-CA-17-D2-F0-63-7A-85-1E-98-BC-46-57-3B
2021-03-09 09:53:18.881 13875-13875/no.nordicsemi.android.nrfmeshprovisioner V/MeshManagerApi: MeshNetwork pdu sent: 0x005C7CAC154ADFFDD6D120F8CCCA17D2F0637A851E98BC46573B
2021-03-09 09:53:19.183 13875-14675/no.nordicsemi.android.nrfmeshprovisioner D/BluetoothGatt: onNotify() - Device=F9:FA:4D:**:**:** handle=18
2021-03-09 09:53:19.183 13875-14675/no.nordicsemi.android.nrfmeshprovisioner I/BleManager: Notification received from 00002ade-0000-1000-8000-00805f9b34fb, value: (0x) 00-5C-9B-25-84-1E-EA-0A-A0-E5-18-48-41-2B-0E-52-8C-49-21-A6-79-12-D6-B8-C7-5A-85-DC-D8-15
2021-03-09 09:53:19.184 13875-13875/no.nordicsemi.android.nrfmeshprovisioner V/MeshManagerApi: Received network pdu: 0x005C9B25841EEA0AA0E51848412B0E528C4921A67912D6B8C75A85DCD815
2021-03-09 09:53:19.184 13875-13875/no.nordicsemi.android.nrfmeshprovisioner V/BaseMeshMessageHandler: TTL for received message: 5
2021-03-09 09:53:19.184 13875-13875/no.nordicsemi.android.nrfmeshprovisioner V/BaseMeshMessageHandler: Sequence number of received access message: 15
2021-03-09 09:53:19.185 13875-14675/no.nordicsemi.android.nrfmeshprovisioner D/BluetoothGatt: onNotify() - Device=F9:FA:4D:**:**:** handle=18
2021-03-09 09:53:19.185 13875-14675/no.nordicsemi.android.nrfmeshprovisioner I/BleManager: Notification received from 00002ade-0000-1000-8000-00805f9b34fb, value: (0x) 00-5C-4F-81-79-2E-51-DD-F6-E1-09-DE-0B-65-D2-B7-BB-80-AD-46-0D-FC-75-BB-78-56-43-91-91-EF
2021-03-09 09:53:19.185 13875-13875/no.nordicsemi.android.nrfmeshprovisioner V/NetworkLayer: TTL for received message: 5
2021-03-09 09:53:19.186 13875-13875/no.nordicsemi.android.nrfmeshprovisioner V/NetworkLayer: Dst: 0x0001
2021-03-09 09:53:19.186 13875-13875/no.nordicsemi.android.nrfmeshprovisioner V/NetworkLayer: Received a segmented access message from: 0003
2021-03-09 09:53:19.186 13875-13875/no.nordicsemi.android.nrfmeshprovisioner V/LowerTransportLayer: SEG O: 0
[... cut ...]
2021-03-09 09:53:22.847 13875-13875/no.nordicsemi.android.nrfmeshprovisioner V/LowerTransportLayer: Last SeqAuth value 15
2021-03-09 09:53:22.847 13875-13875/no.nordicsemi.android.nrfmeshprovisioner V/LowerTransportLayer: Current SeqAuth value 15
2021-03-09 09:53:22.847 13875-13875/no.nordicsemi.android.nrfmeshprovisioner V/LowerTransportLayer: Ignoring message since the incomplete timer has expired and all messages have been received
2021-03-09 09:53:22.847 13875-13875/no.nordicsemi.android.nrfmeshprovisioner V/DefaultNoOperationMessageState: Message reassembly may not be completed yet!
```

kcjian, 2021-03-10:
Line 500 of `LowerTransportLayer` is changed to 2.4.4 and everything is normal

[image]

roshanrajaratnam, 2021-03-10:
Hi Kcjian I will take a look at this. I did some changes to the LowerTransportLayer so may be this was causing the issue. Thanks

closed 2021-03-10 by roshanrajaratnam

kcjian, 2021-03-11:
The test found that there are still the same problems with non direct connected devices

roshanrajaratnam, 2021-03-11:
Are your messages not getting completed?

kcjian, 2021-03-11:
There is no problem with direct connection device.

Use nRFMesh:

app send : Created Access PDU F4C809000208FF
app received: AccessLayer: Received Access PDU F7C809001208FF005515111171777770111111010011DD
and many Message reassembly may not be completed yet

It does not affect the function of nRF mesh, but my own app will send the next message after receiving this reply, and the device will not respond at this time

If I delay sending the next message by 5 seconds after receiving this reply, it will be OK, but I should not.

roshanrajaratnam, 2021-03-11:
Agree, I will try to take a look at this before end of this week.

kcjian, 2021-03-11:
Mesh SDK 3.1.1，thank you very much

roshanrajaratnam, 2021-03-11:
Great, do you mind sharing the hex file? you could also email it to us if you don't want it publicly available here...

kcjian, 2021-03-15:
@roshanrajaratnam 

Here is the simplest DK demo file，
And nRFMesh Log, please have a look.
The test found that the more DK devices, the greater the probability of this problem

[image]
[3.1.1_log.txt](https://github.com/NordicSemiconductor/Android-nRF-Mesh-Library/files/6140877/3.1.1_log.txt)
[Test.zip](https://github.com/NordicSemiconductor/Android-nRF-Mesh-Library/files/6140878/Test.zip)

kcjian, 2021-03-29:
@roshanrajaratnam 
Is there any progress on this issue? I need to solve this problem urgently and need your help，Thank you very much.

roshanrajaratnam, 2021-03-29:
More devices in a mesh network can cause more retransmissions depending on a lot of factors. Starting from connection interval between the proxy client and the proxy node. Not entirely sure how you have formed your network but a simple CompositionDataGet seem to be behaving as expected. I see that you are referring to your vendor model here. Is there any chance you can share it with us?

kcjian, 2021-04-01:
@roshanrajaratnam  The project file is too large, please download it here

[https://pan.baidu.com/s/1JgGPfJVX-WhKlTFy8pB33w ](https://pan.baidu.com/s/1JgGPfJVX-WhKlTFy8pB33w )

 Password: `m82o`

R0m4in-dooz, 2021-05-07:
Hello @roshanrajaratnam . We are facing a problem similar to this nowadays with v3.1.5. Sending segmented msgs and construct back segmented answer is fine with proxy device. However as soon as we try to send such msg to non-proxy devices, it fails to construct the Status, even if all feedback segments has been received. I will create a detailed report during the morning and open a new issue.

roshanrajaratnam, 2021-05-07:
Hi @R0m4in-dooz, sorry for any inconvenience. Some details would be very nice as this can get tricky. Thanks for looking in to this.

roshanrajaratnam, 2021-06-11:
@kcjian let me know here how your test goes!

commit 7dd37f1a (2021-03-10): Fixes #406 which was broken in release 3.0.0 and onwards.
PR #408 (2021-03-10): Release v3.1.1
* This release brings the following changes
  - Fixes #406: Related to broken block acknowledgements in the Lower Transport Layer.
  - Fixes #407 : Adds a check to confirm the old identity key is not null as a Key Refresh Procedure may have not taken place causing the old identity key to be null.