nordicsemi/Android-nRF-Mesh-Library #407: nodeIdentityMatches crashes because of oldIdentityKey being null

Syn38, 2021-03-08:
---

**Describe the bug**
Just after provisioning, we are trying to reconnect to the product as a proxy. So we check the advertising data using the `MeshManagerApi` function `nodeIdentityMatches`.

We managed to find that the `oldIdentityKey` attribute of the `NetworkKey` is **null**

It results as a crash, here is the error output:

> E/AndroidRuntime: FATAL EXCEPTION: main
>     Process: net.linkio.mesh_sig_plugin_example, PID: 3031
>     java.lang.NullPointerException: Attempt to get length of null array
>         at org.spongycastle.crypto.params.KeyParameter.<init>(KeyParameter.java:13)
>         at no.nordicsemi.android.mesh.utils.SecureUtils.encryptWithAES(SecureUtils.java:449)
>         at no.nordicsemi.android.mesh.utils.SecureUtils.calculateHash(SecureUtils.java:439)
>         at no.nordicsemi.android.mesh.MeshManagerApi.nodeIdentityMatches(MeshManagerApi.java:719)
>         at net.linkio.mesh_sig_plugin.ScannerManager.checkIfNodeIdentityMatches(ScannerManager.java:100)
>         at net.linkio.mesh_sig_plugin.ScannerManager$1.onScanResult(ScannerManager.java:71)
>         at no.nordicsemi.android.support.v18.scanner.BluetoothLeScannerCompat$ScanCallbackWrapper.handleScanResult(BluetoothLeScannerCompat.java:491)
>         at no.nordicsemi.android.support.v18.scanner.BluetoothLeScannerImplLollipop$ScanCallbackWrapperLollipop$1$1.run(BluetoothLeScannerImplLollipop.java:250)
>         at android.os.Handler.handleCallback(Handler.java:883)
>         at android.os.Handler.dispatchMessage(Handler.java:100)
>         at android.os.Looper.loop(Looper.java:224)
>         at android.app.ActivityThread.main(ActivityThread.java:7520)
>         at java.lang.reflect.Method.invoke(Native Method)
>         at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:539)
>         at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:950)

**To Reproduce**
Provision a product and scan for proxy.

**Platform details:**
 - Device: Xiaomi Mi 8
 - OS: Android 10
 - Library Version 3.0.2

roshanrajaratnam, 2021-03-09:
Hi, this should fix the issue. Could you give this a spin and see?

Syn38, 2021-03-10:
Hi Roshan,

Thank you for the quick fix! Unfortunately I am using the library with the release version so I'm currently not able to test it properly. Anyway as far as I can see, your fix will do the job!

roshanrajaratnam, 2021-03-10:
Great thanks for the detailed bug report and the confirmation, this was not properly tested. I will publish a version before the end of the day.

PR #408 (2021-03-10): Release v3.1.1
* This release brings the following changes
  - Fixes #406: Related to broken block acknowledgements in the Lower Transport Layer.
  - Fixes #407 : Adds a check to confirm the old identity key is not null as a Key Refresh Procedure may have not taken place causing the old identity key to be null.
commit d3ba6eb2 (2021-03-10): Fixes #407, added a check to see if the old identity key is null because it can be null if a Key refresh procedure has not taken place.
closed 2021-03-10 by roshanrajaratnam