nordicsemi/Android-nRF-Mesh-Library #409: Sensor values are not correctly decoded.

franzney, 2021-03-16:
Since App version 3.1.1 there is support to get detailed sensor information.

Unfortunately the Android version of the App is not showing (decoding) the sensor values correctly!
The IOS version (3.1.0) has this bug not and sensor values are shown correctly.

The following sensor values are tested:

- Ambient temperature / IOS=OK / Android=NO
- Ambient rel. humidity / IOS=OK / Android=NO
- Ambient light level     / IOS=OK / Android=NO
- Motion sensed           / IOS=OK / Android=OK

Looks if the Android implementation only supports the 'motion sensed' property!

Example:
My sensor as implemented 2 properties (temp + humidity)
The attached 2 pictures showing the difference between IOS and Android.

[image]
[image]

Regards,
Fränz

franzney, 2021-03-16:
Android: 3.1.1
Device firmware is based on the Nordic nrf Mesh sensor example. The example only support motion sensed for now!
I have added Temp, Hum, Lux by myself.... - Maybe you can yust change the property ID for testing purpose.

roshanrajaratnam, 2021-03-16:
Ok will do that.

roshanrajaratnam, 2021-03-16:
Hi I just tested the the Sensor Server examples in the nRF Mesh SDK v5..0.0 and I see only a Motion Sensor example on it. Were you referring to the NCS example?

franzney, 2021-03-16:
No, as already mentioned the example only supports the motion property... I have added the temp, hum property by myself...
But the IOS version is handling those correctly... It looks if the Android version only supports the motion property...

Maybe the nordic connect SDK or Zephyr OS has examples to check those... If not you have to modify the nrf sensor example and add those properties for testing....

roshanrajaratnam, 2021-03-16:
Would be really glad if you can share a hex file with me so I can try to take a look at this since I'm assigned with some other priorities.

roshanrajaratnam, 2021-04-12:
Closing due to inactivity, please feel free to re-open.

closed 2021-04-12 by roshanrajaratnam