nordicsemi/Android-nRF-Mesh-Library #414: Delay in sending data

Nuclear6, 2021-04-02:
hi, Why do I need to add a delay every time I send data? These delays are too long, can this time be changed?

roshanrajaratnam, 2021-04-06:
Hi where do you add delays? is it something to do with longer messages with block acknowledgements?

Nuclear6, 2021-04-12:
[image]
[image]
Delays have been added in these places.

roshanrajaratnam, 2021-04-12:
Hi Nuclear you are free to remove these delays. I will remove them in the next release. These were added long time ago to ensure any messages weren't colliding this shouldn't be the case now.

closed 2021-04-12 by roshanrajaratnam