nordicsemi/Android-nRF-Mesh-Library #416: Scene Recall "Read" button  functionality!!

ankit-thealchemist, 2021-04-05:
Hello,

I am using the esp32 micro controller. I created the time scene server and the time scene server setup server model on the element. 
I provisioned the node and bind the keys by using the nrf mesh app. When I click on the scene server, I saw no scene number which are stored on the node(or not reading the scene register state of the scene numbers). So I clicked on the "Read" button, nothing happens. When I check the debug logs on the esp32, when ever I click on the read button, the op code send by the nrf mesh to the node is 0x82 0x41, which belong to the scene get message. Is it intended, because the reply send by the esp32 to the nrf mesh app is 5e000100(current scene number is 0x0001)? 

What I think the correct way should be when ever a "read" button should clicked it should send the Scene register get code ( opcode 0x82 0x44), so the micro controller should send the scene register status (op code 0x82 0x45), by this way we were able to get the all scene number on the nrf mesh app.

Also, scene store functionality is also not working. No opcode has been send by the app to the server. 

Please let me know the correct functionality of the scene model implementation. Thanks in advance!

roshanrajaratnam, 2021-04-06:
Hi Ankit, the app does not support time scene server setup as of now, so I am not sure how you managed to send those messages from the app unless you have added those messages yourself? As far as I remember the only scene related models that are supported in the app are Scene Server and Scene Setup Server.

However there seems to be a bug in the UI, I'll double check and get back to you on this.

roshanrajaratnam, 2021-04-06:
So the idea of the read button to read the current scene (by sending SceneGet) that's on the device. There was some refactoring done on the UI which has caused the list of scenes to be invisible after receiving SceneStatus. I will fix this right away.

closed 2021-04-06 by roshanrajaratnam

ankit-thealchemist, 2021-04-06:
> So the idea of the read button to read the current scene (by sending SceneGet) that's on the device. There was some refactoring done on the UI which has caused the list of scenes to be invisible after receiving SceneStatus. I will fix this right away.

Thanks for the speedy reply. I just want to know that, is there is any way by the app to send the scene register get message, so that the micro controller will  send the scene register status.  Because when I check the debug logs, the app never send the scene register get message. So that nrf mesh app would have all the scene stored in the element.  This is sometimes necessary in the case when scenes are stored in the element while development. Also it will be helpful when there were multiple provisionor in the mesh network. 

Anyway thanks for developing this awesome app. Without this app it wont be possible to understand the ble mesh so easily and so fast!! Keep it up!

roshanrajaratnam, 2021-04-06:
Hi Ankit

I understand your concern, the app UI does not support sending a scene register get as of now. We are on a feature freeze as of now due to some other priorities except for some maintenance on this app so unfortunately you will have to add this message yourself.

Thanks for your feedback really appreciate it.

PR #417 (2021-04-06): Release v3.1.5
This release introduces the following changes.

* Fixes #416 where scenes that are listed not displayed correctly. This most probably broke when migrating from ButterKnife.
* Cleans up a few deprecated functionality from the  mesh library.