nordicsemi/Android-nRF-Mesh-Library #421: NFC support for nRF mesh android application

smohan1098, 2021-04-19:
It would be great if the nRF Mesh android application would support the NFC feature while using OOB authentication during provisioning. The output OOB random number can be read as NFC record and it would be great if this Android app parses the NFC record .

philips77, 2021-04-19:
Hello,
Is the OOB data in some standard form in the NFC record?
We are currently busy with other tasks, but we can add this to feature requests. Unfortunately, I can't promise any dates.

smohan1098, 2021-04-19:
Yes. Just a suggestion that , NFC data can be used as one of the output OOB for authentication. For example the currently supported output OOB from Mesh Profile are {blink, beep ,vibrate, output numeric, output alphanumeric}. In case of blink - a random number is outputted through LED blinks, and the user must give that as input in GUI of the android app. My idea is to turn on NFC scanning at this phase in the device and send the random number as a NFC record to the nRF Mesh app. After this the normal provisioning process continues as in the case of output OOB.  The data type can be numeric unsigned 128-bit value. This value is termed as AuthValue , and it is encoded into an array in big-endian format , which is managed by existing algorithm.

Please Refer the Mesh Profile v1.0.1 ,Section 5.4.3.4, Page 253.

roshanrajaratnam, 2023-05-25:
We don't have this planned as of now, but we gladly accept contribution. :)