nordicsemi/Android-nRF-Mesh-Library #422: iOS/Android MTU value

fiveagle, 2021-04-21:
Hi,
we are facing a strange issue while implementing iOS/Android cross mesh network support.

We successfully import the mesh created from iOS in Android but we can't proxy with nodes provisioned on iOS.
Apparently the nodes still expect a message segmentation based on the iOS MTU and so they refuse the Android segmentation.

`2021-04-21 09:43:03.762 6608-6608/com.xxx.xxx E/MeshManagerApi: Parsing notification failed: 0x010100DBB840445C751109000000006631B8E66B - Incorrect Secure Network Beacon length: 19, expected: 22. Check MTU and Proxy Protocol segmentation.`

If we depower and repower the devices the renegotiate the MTU to 23 (default) and everything gets back to normal.

Is this behaviour expected? There is a way to force an MTU negotiation before sending new proxy messages?
Should we address this during the import?
There is a fix for this issue in library versions released after 2.1.4? Unfortunately we still couldn't manage to plan an update to the next major.

Thanks a lot.

**Platform details:**
 - Device: Any
 - OS: Any
 - Library Version: 2.1.4

roshanrajaratnam, 2021-04-21:
Hi this is not the expected behavior but have not noticed this on our end. Does the MTU request succeed at all on your case?

fiveagle, 2021-04-21:
> Does the MTU request succeed at all on your case?

Right now we don't send any MTU request explicitly

roshanrajaratnam, 2021-04-21:
I would suggest sending the MTU request explicitly or set it to 23 as default (as done in the example app) as it is used in the library to [handle](https://github.com/NordicSemiconductor/Android-nRF-Mesh-Library/blob/46b559cfb35bf4ec451397a27ee62bd501b6064d/app/src/main/java/no/nordicsemi/android/nrfmesh/viewmodels/NrfMeshRepository.java#L497) the gatt level segmentation of the notifications received. From Lollipop onwards you can make use of higher MTU of 517 instead of the 23 and will reduce some traffic in your network as well.

philips77, 2021-04-21:
Remember, that iOS sends MTU request automatically just after connection. Android requires sending it manually using [setMtu(...)](https://github.com/NordicSemiconductor/Android-nRF-Mesh-Library/blob/11c9c8469d3a6399541e4a50e7998b94cb6f1851/app/src/main/java/no/nordicsemi/android/nrfmesh/ble/BleMeshManager.java#L115).
But higher MTU should not be required, as Proxy Protocol is used to send packets in required chunks.

Observe, that the message you receive starts with 0x01. Two MSBits are 0b00, which means this packet should contain the whole packet. Looks like your device thinks the current MTU is higher, while in fact it's 23.

fiveagle, 2021-04-21:
> Observe, that the message you receive starts with 0x01. Two MSBits are 0b00, which means this packet should contain the whole packet. Looks like your device thinks the current MTU is higher, while in fact it's 23.

In fact iOS negotiates at least 66 in my tests. Is it correct that the mesh device still expects 66? Should I close o reset something on iOS side?

fiveagle, 2021-04-21:
Anyway setting MAX mtu makes the trick. Our fork was still requesting default instead of max.

Thanks a lot for the super quick responses!!!

closed 2021-04-21 by fiveagle