nordicsemi/Android-nRF-Mesh-Library #423: Crash after updating app from any version 2.4.4 and below to 3.0 and above

BubblyNetDev, 2021-04-21:
---

**Describe the bug**
Downloaded 2.4.4 and added a device, bound keys, and added subscriptions. Controlled device from 2.4.4 with no issues. Downloaded 3.1.5 and built app. Device TTL unknown and app crashes when trying to send messages.

**To Reproduce**
Steps to reproduce the behavior:
1. Build 2.4.4 and add a device, bind keys, and add subscriptions
2. Update the app to version 3.0 or above
3. Try to control device and app crashes

**Logs / Screenshots**
```
E/AndroidRuntime: FATAL EXCEPTION: main
    Process: no.nordicsemi.android.nrfmeshprovisioner, PID: 9297
    java.lang.NullPointerException: Attempt to invoke virtual method 'int java.lang.Integer.intValue()' on a null object reference
        at no.nordicsemi.android.mesh.transport.MeshTransport.createMeshMessage(MeshTransport.java:145)
        at no.nordicsemi.android.mesh.transport.ConfigMessageState.createAccessMessage(ConfigMessageState.java:69)
        at no.nordicsemi.android.mesh.transport.ConfigMessageState.<init>(ConfigMessageState.java:54)
        at no.nordicsemi.android.mesh.transport.BaseMeshMessageHandler.createConfigMeshMessage(BaseMeshMessageHandler.java:281)
        at no.nordicsemi.android.mesh.transport.BaseMeshMessageHandler.createMeshMessage(BaseMeshMessageHandler.java:247)
        at no.nordicsemi.android.mesh.MeshManagerApi.createMeshPdu(MeshManagerApi.java:891)
        at no.nordicsemi.android.nrfmesh.node.NodeConfigurationActivity.sendMessage(NodeConfigurationActivity.java:392)
        at no.nordicsemi.android.nrfmesh.node.NodeConfigurationActivity.onNodeReset(NodeConfigurationActivity.java:280)
        at no.nordicsemi.android.nrfmesh.node.dialog.DialogFragmentResetNode.lambda$onCreateDialog$0$DialogFragmentResetNode(DialogFragmentResetNode.java:60)
        at no.nordicsemi.android.nrfmesh.node.dialog.-$$Lambda$DialogFragmentResetNode$gCxhC2Iii6008Kq2hz97l_5HhbY.onClick(Unknown Source:2)
        at androidx.appcompat.app.AlertController$ButtonHandler.handleMessage(AlertController.java:167)
        at android.os.Handler.dispatchMessage(Handler.java:107)
        at android.os.Looper.loop(Looper.java:237)
        at android.app.ActivityThread.main(ActivityThread.java:8167)
        at java.lang.reflect.Method.invoke(Native Method)
        at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:496)
        at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:1100)

```

roshanrajaratnam, 2021-04-22:
Hi, yes there have been multiple migrations in between these versions, but should not crash the app when trying to control. I will have to debug this and see. I can't promise any dates as I have some other tasks in my plate.

BubblyNetDev, 2021-04-22:
Okay, I added ttl to the migrations as a fix for now.

roshanrajaratnam, 2021-05-26:
Sorry for the delay but is fixed on dev now.

PR #437 (2021-06-23): Release v3.1.6
This release brings the following changes
* Specify the address of the element instead of the node address when sending message from the SensorServerActivity.java 
* Fixes #423, #418, #333 related to database migration
* Fixes #354 an invalid GATT level segmentation issue. The lib was considering the MTU size instead of the SAR bits in the proxy PDU
* Fixes #427, #406 where segmented messages to be retransmitted being get cleared. This should improve the retransmissions
* Fixes #433 where the app was crashing on Android 6.0 devices due incorrect time stamp formatting.
* Migrates [... cut]
closed 2021-05-31 by roshanrajaratnam