nordicsemi/Android-nRF-Mesh-Library #424: Send a temporary message

xhtlxa, 2021-04-23:
How do I send a temporary message to the device I connected via mViewModel.connect(this, device, false)?

roshanrajaratnam, 2021-04-23:
Hi could you elaborate your issue a little bit more please?

xhtlxa, 2021-04-23:
How to send a message to a device after mViewModel.connect() instead of sending a message to the model under node

philips77, 2021-04-23:
Hello,
Could you clarify what do you mean by temporary message? Or what View Model are you talking? What do you want to achieve? In Bluetooth Mesh you always send messages to all nodes, which may be consumed by some models on some nodes, so I don't know what is your question about.

xhtlxa, 2021-04-23:
How can I send a message to the device without getting the Element and MeshModel of the node?

roshanrajaratnam, 2021-04-23:
What is the message you are trying to send?

xhtlxa, 2021-04-23:
Some byte[] defined by ourselves

roshanrajaratnam, 2021-04-23:
Are you referring to a vendor model message?

xhtlxa, 2021-04-25:
Yes, before you get Element and MeshModel

xhtlxa, 2021-05-18:
solved

closed 2021-05-18 by xhtlxa