nordicsemi/Android-nRF-Mesh-Library #425: Inconsistent UUID formatting between iOS mesh export and Android mesh Export

Cplantijn, 2021-04-30:
Sharing configurations between iOS and Android is problematic since node UUID formats differ between the two platforms. 

Example mesh export from `Android`: 
```json
{
...
  nodes: [
    {
      UUID: "DDDEAD67-B371-61EF-0000-000000000000"
      name: "Christopher's Mesh Node",
      unicastAddress: "0001"
      ....
    }
  ]
....
}
```

Example mesh export from `iOS`:
```json
{
...
  nodes: [
    {
      UUID: "DDDEAD67B37161EF0000000000000000"
      name: "Christopher's Mesh Node",
      unicastAddress: "0001"
      ....
    }
  ]
....
}
```
The UUID is a string in both cases, but the dashes are only present in the Android export.

**Describe the solution you'd like**
Consistent UUID string formatting for both android and iOS. Either the android drops the dashes, or the iOS version adds them in on mesh JSON export.

**Describe alternatives you've considered**
Right now I am just resolving the issue with some string transformations to normalize them all to a format with the dashes. However I think this issue should be resolved in the library itself to make sure mesh JSON exports are cross platform compatible.

philips77, 2021-05-02:
Are you using the latest ios version, which is 3.1.2?

philips77, 2021-05-02:
This has change in nRF Mesh 3.0:
https://github.com/NordicSemiconductor/IOS-nRF-Mesh-Library/commit/e095ffdc34c9dc710b5c0259a290fd735cb01e8b

If your cocoa pods doesn't find newer version, update cocoa pods in Mac.

Cplantijn, 2021-05-06:
@philips77 Thank you for your suggestion. We went ahead and upgraded our iOS mesh library and we are getting expected casings for node UUIDs. Thank you.

closed 2021-05-06 by Cplantijn