nordicsemi/Android-nRF-Mesh-Library #427: Data transmission failure of non directly connected device

kcjian, 2021-05-08:
---

**Describe the bug**
Send data `F1C809CB3132333435363738393031323334353637383930313233343536373839 `. The direct connected device sends successfully and receives a reply. The non direct connected device does not respond

The data length is 33. If I only send data below 33, there is no problem

**Platform details:**
 - Device: HuaWei P40 Pro
 - OS: android 10
 - Library Version 3.1.1

**Logs / Screenshots**

The log of direct connected device is as follows：

```
2021-05-08 16:12:21.667 18506-18506/com.xuntong.meshuart V/MeshTransport: Src address: 1990
2021-05-08 16:12:21.668 18506-18506/com.xuntong.meshuart V/MeshTransport: Dst address: 0017
2021-05-08 16:12:21.668 18506-18506/com.xuntong.meshuart V/MeshTransport: Key: 4E3EDD70BFC8172A06B5A5E531439CD8
2021-05-08 16:12:21.668 18506-18506/com.xuntong.meshuart V/MeshTransport: akf: 1
2021-05-08 16:12:21.668 18506-18506/com.xuntong.meshuart V/MeshTransport: aid: 63
2021-05-08 16:12:21.668 18506-18506/com.xuntong.meshuart V/MeshTransport: aszmic: 0
2021-05-08 16:12:21.668 18506-18506/com.xuntong.meshuart V/MeshTransport: Sequence number: 2964
2021-05-08 16:12:21.669 18506-18506/com.xuntong.meshuart V/MeshTransport: Access message opcode: f1
2021-05-08 16:12:21.669 18506-18506/com.xuntong.meshuart V/MeshTransport: Access message parameters: 123132333435363738393031323334353637383930313233343536373839
2021-05-08 16:12:21.670 18506-18506/com.xuntong.meshuart V/AccessLayer: Created Access PDU F1C809123132333435363738393031323334353637383930313233343536373839
2021-05-08 16:12:21.670 18506-18506/com.xuntong.meshuart V/UpperTransportLayer: Application nonce: 0100000B941990001700000000
2021-05-08 16:12:21.671 18506-18506/com.xuntong.meshuart V/UpperTransportLayer: Encrypted upper transport pdu: E3E5FA564A21C38693B50D12874920A37901027EED6F09A1027AFADC366AB1C99B2455AEFA
2021-05-08 16:12:21.672 18506-18506/com.xuntong.meshuart V/LowerTransportLayer: Segmented Lower transport access PDU: FF2E5003E3E5FA564A21C38693B50D12 0 of 4
2021-05-08 16:12:21.674 18506-18506/com.xuntong.meshuart V/LowerTransportLayer: Segmented Lower transport access PDU: FF2E5023874920A37901027EED6F09A1 1 of 4
2021-05-08 16:12:21.674 18506-18506/com.xuntong.meshuart V/LowerTransportLayer: Segmented Lower transport access PDU: FF2E5043027AFADC366AB1C99B2455AE 2 of 4
2021-05-08 16:12:21.674 18506-18506/com.xuntong.meshuart V/LowerTransportLayer: Segmented Lower transport access PDU: FF2E5063FA 3 of 4
2021-05-08 16:12:21.675 18506-18506/com.xuntong.meshuart V/NetworkLayer: Encryption key: 06358353B92CC8027DD1BAFD2006D648
2021-05-08 16:12:21.675 18506-18506/com.xuntong.meshuart V/NetworkLayer: Privacy key: 68751462A589E2F9DE9400E7A2D466C4
2021-05-08 16:12:21.676 18506-18506/com.xuntong.meshuart V/NetworkLayer: Sequence Number: 000B94
2021-05-08 16:12:21.677 18506-18506/com.xuntong.meshuart V/NetworkLayer: Encrypted Network payload: 61E6EF16B6713C3F6A3C3A288C01CE5031CA5C1C06E0
2021-05-08 16:12:21.678 18506-18506/com.xuntong.meshuart V/NetworkLayer: Sequence Number: 000B95
2021-05-08 16:12:21.679 18506-18506/com.xuntong.meshuart V/NetworkLayer: Encrypted Network payload: 7DF3A46E6000058130AD31777DF7AE4F91169B76C63E
2021-05-08 16:12:21.680 18506-18506/com.xuntong.meshuart V/NetworkLayer: Sequence Number: 000B96
2021-05-08 16:12:21.681 18506-18506/com.xuntong.meshuart V/NetworkLayer: Encrypted Network payload: A1C04ACFD344145165BAA79D6B24DD117DE9093480EB
2021-05-08 16:12:21.682 18506-18506/com.xuntong.meshuart V/NetworkLayer: Sequence Number: 000B97
2021-05-08 16:12:21.683 18506-18506/com.xuntong.meshuart V/NetworkLayer: Encrypted Network payload: 5E64CEB1217A55C8558AFF
2021-05-08 16:12:21.684 18506-18506/com.xuntong.meshuart V/VendorModelMessageAckedState: Sending acknowledged vendor model message
2021-05-08 16:12:21.686 18506-18506/com.xuntong.meshuart V/BleManager: Writing characteristic 00002add-0000-1000-8000-00805f9b34fb (WRITE COMMAND)
2021-05-08 16:12:21.686 18506-18506/com.xuntong.meshuart D/BleManager: gatt.writeCharacteristic(00002add-0000-1000-8000-00805f9b34fb)
2021-05-08 16:12:21.707 18506-18506/com.xuntong.meshuart I/BleManager: Data written to 00002add-0000-1000-8000-00805f9b34fb, value: (0x) 00-61-1A-56-93-F7-5F-BA-61-E6-EF-16-B6-71-3C-3F-6A-3C-3A-28-8C-01-CE-50-31-CA-5C-1C-06-E0
2021-05-08 16:12:21.707 18506-18506/com.xuntong.meshuart V/MeshManagerApi: MeshNetwork pdu sent: 0x00611A5693F75FBA61E6EF16B6713C3F6A3C3A288C01CE5031CA5C1C06E0
2021-05-08 16:12:21.707 18506-18506/com.xuntong.meshuart V/BleManager: Writing characteristic 00002add-0000-1000-8000-00805f9b34fb (WRITE COMMAND)
2021-05-08 16:12:21.707 18506-18506/com.xuntong.meshuart D/BleManager: gatt.writeCharacteristic(00002add-0000-1000-8000-00805f9b34fb)
2021-05-08 16:12:21.722 18506-18506/com.xuntong.meshuart I/BleManager: Data written to 00002add-0000-1000-8000-00805f9b34fb, value: (0x) 00-61-31-87-D4-B0-AB-11-7D-F3-A4-6E-60-00-05-81-30-AD-31-77-7D-F7-AE-4F-91-16-9B-76-C6-3E
2021-05-08 16:12:21.722 18506-18506/com.xuntong.meshuart V/MeshManagerApi: MeshNetwork pdu sent: 0x00613187D4B0AB117DF3A46E6000058130AD31777DF7AE4F91169B76C63E
2021-05-08 16:12:21.722 18506-18506/com.xuntong.meshuart V/BleManager: Writing characteristic 00002add-0000-1000-8000-00805f9b34fb (WRITE COMMAND)
2021-05-08 16:12:21.722 18506-18506/com.xuntong.meshuart D/BleManager: gatt.writeCharacteristic(00002add-0000-1000-8000-00805f9b34fb)
2021-05-08 16:12:21.725 18506-18506/com.xuntong.meshuart I/BleManager: Data written to 00002add-0000-1000-8000-00805f9b34fb, value: (0x) 00-61-66-6E-80-63-48-8A-A1-C0-4A-CF-D3-44-14-51-65-BA-A7-9D-6B-24-DD-11-7D-E9-09-34-80-EB
2021-05-08 16:12:21.725 18506-18506/com.xuntong.meshuart V/MeshManagerApi: MeshNetwork pdu sent: 0x0061666E8063488AA1C04ACFD344145165BAA79D6B24DD117DE9093480EB
2021-05-08 16:12:21.725 18506-18506/com.xuntong.meshuart V/BleManager: Writing characteristic 00002add-0000-1000-8000-00805f9b34fb (WRITE COMMAND)
2021-05-08 16:12:21.725 18506-18506/com.xuntong.meshuart D/BleManager: gatt.writeCharacteristic(00002add-0000-1000-8000-00805f9b34fb)
2021-05-08 16:12:21.728 18506-18506/com.xuntong.meshuart I/BleManager: Data written to 00002add-0000-1000-8000-00805f9b34fb, value: (0x) 00-61-5E-EE-26-CD-FC-6C-5E-64-CE-B1-21-7A-55-C8-55-8A-FF
2021-05-08 16:12:21.728 18506-18506/com.xuntong.meshuart V/MeshManagerApi: MeshNetwork pdu sent: 0x00615EEE26CDFC6C5E64CEB1217A55C8558AFF
2021-05-08 16:12:21.992 18506-18506/com.xuntong.meshuart I/BleManager: Notification received from 00002ade-0000-1000-8000-00805f9b34fb, value: (0x) 00-61-C7-5A-59-94-0D-D3-3C-2D-30-47-11-5A-CE-8E-D1-C5-37-BA-7C-E3-A5-16-07
2021-05-08 16:12:21.993 18506-18506/com.xuntong.meshuart V/MeshManagerApi: Received network pdu: 0x0061C75A59940DD33C2D3047115ACE8ED1C537BA7CE3A51607
2021-05-08 16:12:21.993 18506-18506/com.xuntong.meshuart V/BaseMeshMessageHandler: TTL for received message: 8
2021-05-08 16:12:21.993 18506-18506/com.xuntong.meshuart V/BaseMeshMessageHandler: Sequence number of received access message: 8193
2021-05-08 16:12:21.994 18506-18506/com.xuntong.meshuart V/NetworkLayer: TTL for received message: 8
2021-05-08 16:12:21.994 18506-18506/com.xuntong.meshuart V/DefaultNoOperationMessageState: Acknowledgement payload: 0000000F
2021-05-08 16:12:21.994 18506-18506/com.xuntong.meshuart V/BlockAcknowledgementMessage: Segment 0 of 3 received by peer
2021-05-08 16:12:21.994 18506-18506/com.xuntong.meshuart V/BlockAcknowledgementMessage: Segment 1 of 3 received by peer
2021-05-08 16:12:21.994 18506-18506/com.xuntong.meshuart V/BlockAcknowledgementMessage: Segment 2 of 3 received by peer
2021-05-08 16:12:21.994 18506-18506/com.xuntong.meshuart V/BlockAcknowledgementMessage: Segment 3 of 3 received by peer
[... 145 lines cut ...]
2021-05-08 16:10:19.169 18506-18506/com.xuntong.meshuart V/BaseMeshMessageHandler: Sequence number of received access message: 24795
2021-05-08 16:10:19.170 18506-18506/com.xuntong.meshuart V/NetworkLayer: TTL for received message: 7
2021-05-08 16:10:19.172 18506-18506/com.xuntong.meshuart V/DefaultNoOperationMessageState: Acknowledgement payload: 00000003
2021-05-08 16:10:19.173 18506-18506/com.xuntong.meshuart V/BlockAcknowledgementMessage: Segment 0 of 3 received by peer
2021-05-08 16:10:19.173 18506-18506/com.xuntong.meshuart V/BlockAcknowledgementMessage: Segment 1 of 3 received by peer
2021-05-08 16:10:19.173 18506-18506/com.xuntong.meshuart V/BlockAcknowledgementMessage: Segment 2 of 3 not received by peer
[... cut ...]
2021-05-08 16:10:20.272 18506-18506/com.xuntong.meshuart V/BaseMeshMessageHandler: TTL for received message: 7
2021-05-08 16:10:20.273 18506-18506/com.xuntong.meshuart V/BaseMeshMessageHandler: Sequence number of received access message: 24796
2021-05-08 16:10:20.274 18506-18506/com.xuntong.meshuart V/NetworkLayer: TTL for received message: 7
2021-05-08 16:10:20.276 18506-18506/com.xuntong.meshuart V/DefaultNoOperationMessageState: Acknowledgement payload: 00000007
2021-05-08 16:10:20.276 18506-18506/com.xuntong.meshuart V/BlockAcknowledgementMessage: Segment 0 of 0 received by peer

```

Nuclear6, 2021-05-12:
Is the direct connected device set proxy white list, include provisioner address?
The non direct connected device will adv response, these data  will arrive first the direct connected device, and then check dst address in white list. if in white list, the direct connected device will send data to Provisioner.
Hope it helps you!!!

kcjian, 2021-05-19:
@Nuclear6 No, I am not allowed to add provisioner address to the proxy white list in the app

roshanrajaratnam, 2021-05-19:
Duplicate of #426

roshanrajaratnam, 2021-05-26:
@kcjian  could you test this issue against the latest dev branch please? I have fixed an underlying GATT level segmentation issue where sometimes the lib was waiting for messages when it should not.

roshanrajaratnam, 2021-06-09:
Closing due to inactivity and we can continue the discussion on #426

kcjian, 2021-06-10:
@roshanrajaratnam 
I tested nrfmesh version 3.1.5, but the problem still exists

When successful：send 32 bytes
Send：       F1C8091234567890123456789012345678901234567890123456789012345678
Received：F3C8091234567890123456789012345678901234567890123456789012345678
[image]

On failure：Add one byte, the data reaches 33 bytes
Send:          F1C809123456789012345678901234567890123456789012345678901234567890
Received：time out
[image]

[log_successful.txt](https://github.com/NordicSemiconductor/Android-nRF-Mesh-Library/files/6628954/log_successful.txt)
[log_failed.txt](https://github.com/NordicSemiconductor/Android-nRF-Mesh-Library/files/6628955/log_failed.txt)

**There's no problem with direct connected device**
Send：       F1C809123456789012345678901234567890123456789012345678901234567890
Received：F3C809123456789012345678901234567890123456789012345678901234567890

[log_successful_direct_device_with_33_bytes.txt](https://github.com/NordicSemiconductor/Android-nRF-Mesh-Library/files/6628963/log_successful_direct_device_with_33_bytes.txt)

roshanrajaratnam, 2021-06-10:
did you test the version from dev branch?

kcjian, 2021-06-10:
@roshanrajaratnam Yes, the result is the same

roshanrajaratnam, 2021-06-10:
Could you share your fw ? if it's based on nRF 5Sdk could you share the fw bundled with the softdevice please?

kcjian, 2021-06-10:
Yes, I'm planning to make the simplest demo, including a vendor model, which can reply to whatever I send. It may be provided to you tomorrow or next week

roshanrajaratnam, 2021-06-10:
great that would be very helpful to debug this.

kcjian, 2021-06-11:
[image]

- Vendor Modek : 0x00590000
- OpCode: 0xC1
- Parameters: 0x123456789012345678901234567890123456789012345678901234567890

A minimum of two devices are required

[light_switch_server_nrf52832_xxAA_s132_7.2.0.hex.zip](https://github.com/NordicSemiconductor/Android-nRF-Mesh-Library/files/6635339/light_switch_server_nrf52832_xxAA_s132_7.2.0.hex.zip)

roshanrajaratnam, 2021-06-11:
Great will take a look at this now, is this bundled with the soft device?

roshanrajaratnam, 2021-06-11:
Hi @kcjian so I managed to reproduce your issue and after some hours of debugging found the root cause.

kcjian, 2021-06-11:
Can you send and receive successfully on non direct connected devices？

roshanrajaratnam, 2021-06-11:
yes it works for devices when they are not connected directly. This was more of an issue on the network layer causing this to happen.

roshanrajaratnam, 2021-06-11:
@kcjian oops I missed one file in my commit :( . Check the latest commit :)

kcjian, 2021-06-11:
That's right, I can receive the correct reply，Thanks for your work.

And Can you take a look at this problem that still exists

[#406](https://github.com/NordicSemiconductor/Android-nRF-Mesh-Library/issues/406)

The same happens in non direct connected devices.

roshanrajaratnam, 2021-06-11:
I believe this should also be fixed with the current fix.

kcjian, 2021-06-11:
But I tested the above test method, which shows that this problem still exists,I'll test it again

[image]

roshanrajaratnam, 2021-06-11:
ok let's move the conversion to the other issue. I will close this now.

commit fc88c3d7 (2021-06-11): Fixes #427 where the NetworkLayerPdu were getting incorrectly reset when creating segmented messages to be retransmitted
commit dd4fec68 (2021-06-17): Fixes #427 where the NetworkLayerPdu were getting incorrectly reset when creating segmented messages to be retransmitted
PR #437 (2021-06-23): Release v3.1.6
This release brings the following changes
* Specify the address of the element instead of the node address when sending message from the SensorServerActivity.java 
* Fixes #423, #418, #333 related to database migration
* Fixes #354 an invalid GATT level segmentation issue. The lib was considering the MTU size instead of the SAR bits in the proxy PDU
* Fixes #427, #406 where segmented messages to be retransmitted being get cleared. This should improve the retransmissions
* Fixes #433 where the app was crashing on Android 6.0 devices due incorrect time stamp formatting.
* Migrates [... cut]
closed 2021-06-11 by roshanrajaratnam