nordicsemi/Android-nRF-Mesh-Library #431: Provisioning a node that was previously deleted.

Cplantijn, 2021-05-19:
Hello, I am trying to figure out the `networkExclusions` object that gets populated after a node is removed. When I remove a node, I see that it's address is added to the list. If I try to provision that same node again on the same network, I am unable to send messages to it to start the provisioning process.

How can I clear the `networkExclusions` list when I want to re-provision an unprovisioned/reset/deleted node? Is that even the right approach? I've combed over the example app and could not find any examples of things clearing up. What if I have just one node that I want to unprovision (the proxy node) and re-provision. Should I reset the entire network after that?

I see some things about updating the network IV, but I cant find any APIs/methods to do that. 

Any advice would be appreciated. Thanks.

R0m4in-dooz, 2021-05-19:
Hello @Cplantijn . From my understanding you should not try to reuse the addresses that are in network exclusions.
You should assign a new address, getting it via `nextAvailableUnicastAddress` method

roshanrajaratnam, 2021-05-19:
What @R0m4in-dooz  mentions is correct. According to the specification an address can be re-used only after the current iv index is updated. Check section 3.10.7 on mesh profile spec 1.0.1.

Cplantijn, 2021-05-19:
Thank you for the responses. I currently call ...
```kotlin
// Kotlin
val unicast = meshNetwork.nextAvailableUnicastAddress(elementCount, provisioner)
meshNetwork.assignUnicastAddress(unicast)
```
during a device's `PROVISIONING_CAPABILITIES` state during setup (at `onProvisioningStateChanged()`). I do see that the new node gets an updated unicast address (for example it was 002 the first time, then deleted, then reprovisioned at 003). However transmission after that is not picked up by the mesh. For example, the `ConfigCompositionDataGet()` message that is fired after `onProvisioningCompleted()`, disconnect, and reconnect on the node identity advertisement is not picked up. 

I am not sending a `ConfigNodeReset()` message to the node before it is removed from the mesh. Is that what is hanging me up? Thanks.

roshanrajaratnam, 2021-05-20:
Well it's a bit strange that it's not getting picked up. Just to confirm I believe you are sending it to the correct address?

Cplantijn, 2021-05-21:
I am definately getting a different address on the second provisioning attempt. `ConfigCompositionDataGet` is getting sent however

Cplantijn, 2021-05-21:
I can absolutely confirm that the device unicast address is being incremented during the second provision. For example, the first unicast address is 2, then after unprovisioning the node and deleting it from the mesh, the next provisioning attempt has the unicast as address 3. Both times are using `nextAvailableUnicastAddress` and `assignUnicastAddress` in `override fun onProvisioningStateChanged()`

Cplantijn, 2021-05-21:
Thanks for the help, all! It was an issue with my app code. Once I removed the node I still had it marked for removal, so when it was added the second time it was instantly wiped out. Thanks for all the help!

closed 2021-05-21 by Cplantijn