nordicsemi/Android-nRF-Mesh-Library #432: Potential ConcurrentException in MeshNetworkDb:InsertNetworkAsyncTask()

Mavericksb, 2021-05-25:
---
Hello, I'd like to report a second bug found during importing a MeshNetwork (The first one and solution has been post under BubblyNetDev post named "Migration issue in 3.1.5", not sure if it has been read and is planned to be fixed in next releases).

By the way, In class MeshNetworkDb.Java

**Problem :**  
In class : 
private static class InsertNetworkAsyncTask extends AsyncTask<Void, Void, Void> {
> ...
> ...
> if (!meshNetwork.nodes.isEmpty()) {
>     nodesDao.insert(meshNetwork.nodes);
> }
> 
> if (meshNetwork.groups != null) {
>     groupsDao.insert(meshNetwork.groups);
> }
> 
> if (meshNetwork.scenes != null) {
>     scenesDao.insert(meshNetwork.scenes);
> }
> ...
}

This can cause troubles down the line, in our case it caused several (still rare) ConcurrentExceptions : 
Fatal Exception: java.lang.RuntimeException: An error occurred while executing doInBackground()
       at android.os.AsyncTask$4.done(AsyncTask.java:415)
     ...
Caused by java.util.ConcurrentModificationException
       at java.util.ArrayList$Itr.next(ArrayList.java:860)
       at androidx.room.EntityInsertionAdapter.insert(EntityInsertionAdapter.java:95)
       at no.nordicsemi.android.meshprovisioner.data.ProvisionedMeshNodesDao_Impl.insert(ProvisionedMeshNodesDao_Impl.java:303)
       at no.nordicsemi.android.meshprovisioner.MeshNetworkDb$InsertNetworkAsyncTask.doInBackground(MeshNetworkDb.java:307)
       at no.nordicsemi.android.meshprovisioner.MeshNetworkDb$InsertNetworkAsyncTask.doInBackground(MeshNetworkDb.java:271)
...

**Cause :** 
meshNetwork.nodes meshNetwork.groups e meshNetwork.scenes are arrayList. They can be modified from outside the MeshNetworkDb.java for any reason, it shouldn't happen, but if it happens lit eads to ConcurrentException.

**Solution**
I fixed the problem cloning the arrayList before passing it to the Dao, this makes sure we never pass an array which could be potentially modified from outside:
if (!meshNetwork.nodes.isEmpty()) {
    ArrayList<ProvisionedMeshNode> copyNodes = new ArrayList<>(meshNetwork.nodes);
    nodesDao.insert(copyNodes);
}

roshanrajaratnam, 2021-05-25:
Hi, thanks for reporting this. Asynctask has been deprecated in the library. However like you mentioned the migration can cause an issue. I will take a look.

saty9, 2021-06-10:
We are hitting this repeatedly when making multiple changes to the network in quick succession

FluffyBunniesTasteTheBest, 2022-02-07:
Any update on this issue? A few of our users are facing something quite similar:

```
java.util.ConcurrentModificationException: 
  at java.util.ArrayList$Itr.next (ArrayList.java:860)
  at androidx.room.EntityInsertionAdapter.insert (EntityInsertionAdapter.java:95)
  at no.nordicsemi.android.mesh.data.ProvisionersDao_Impl.insert (ProvisionersDao_Impl.java:167)
  at no.nordicsemi.android.mesh.MeshNetworkDb.lambda$insertNetwork$0 (MeshNetworkDb.java:149)
  at no.nordicsemi.android.mesh.-$$Lambda$MeshNetworkDb$cvDKr66SNZXIthGwE9B7B-3EKhs.run (Unknown Source:16)
```

Mavericksb, 2022-02-07:
> Any update on this issue? A few of our users are facing something quite similar:
>   at no.nordicsemi.android.mesh.MeshNetworkDb.lambda$insertNetwork$0 (MeshNetworkDb.java:149)

What is the line of code at MeshNetworkDb.java:149
?
If there is an array passed to an insert() method, just copy the array first, and then pass the copy to the insert method instead of the original array, to avoid concurrency.

roshanrajaratnam, 2022-02-08:
@FluffyBunniesTasteTheBest sorry, been a bit busy with some other project and I just got back from my paternity leave. I will start looking in to these one by one!

Mavericksb, 2022-02-08:
> Beside of just copying the `ArrayList`, I've also delayed the call that's causing the `ConcurrentModificationException`.

Ok, BTW adding a delay shouldn't be necessary.

> Google Play, btw., also logged some related `SQLiteConstraintException`s:
> 
> android.database.sqlite.SQLiteConstraintException: 
> ...

SQLiteConstrainException isn't related to the previous problem, it means that you are passing an Object (or list of Objects) that violate the constraints that are defined in Table constructor.
As far as I can tell you are trying to insert a Provisioner Node.
It's hard to help without any more detail on the Table you are trying to write on, its definition, and the objects you are passing.
One of the most common causes (but it is just an example) could be that the Provisioner Table defined in MeshNetworkDb class needs an UNIQUE Key, and you are trying to insert an object (A Provisioner Node in this case) with an UNIQUE ID which already exists in the Table.

You need to do some debugging to find out where the problem lies. Honestly I didn't face it on our App, sorry I can't help more than this.

FluffyBunniesTasteTheBest, 2022-02-08:
@Mavericksb Thank you very much for the clarification! 
> As far as I can tell you are trying to insert a Provisioner Node.

Yes. During startup, the app creates a bunch of provisioners, to enable multiple app instances to operate the mesh.  

> It's hard to help without any more detail on the Table you are trying to write on, its definition, and the objects you are passing.

That's all handled by the library, but...

> SQLiteConstrainException isn't related to the previous problem, it means that you are passing an Object (or list of Objects) that violate the constraints that are defined in Table constructor.

... that perfectly explains the cause of the `SQLiteConstraintException`:

If the app crashes because of a `ConcurrentException`, it tries to re-create the provisioners again during the next start, which, the way it looks, then causes `SQLiteConstrainException` because some of the provisioners already exist.

Thank you very much for sharing this! You've just saved me from quite some sleepless nights...

FluffyBunniesTasteTheBest, 2022-02-12:
Just for the records, there's one more situation where the `ConcurrentModificationException` can occur and crash the app:
```
java.util.ConcurrentModificationException: 
  at java.util.ArrayList$Itr.next (ArrayList.java:860)
  at androidx.room.EntityDeletionOrUpdateAdapter.handleMultiple (EntityDeletionOrUpdateAdapter.java:86)
  at no.nordicsemi.android.mesh.data.ProvisionerDao_Impl.update (ProvisionerDao_Impl.java:232)
  at no.nordicsemi.android.mesh.MeshNetworkDb.lambda$update$16 (MeshNetworkDb.java:276)
  at no.nordicsemi.android.mesh.-$$Lambda$MeshNetworkDb$kyOqCg3Q_F167aHhj0mXWofVRzM.run (Unknown Source:4)
  at java.util.concurrent.ThreadPoolExecutor.runWorker (ThreadPoolExecutor.java:1167)
  at java.util.concurrent.ThreadPoolExecutor$Worker.run (ThreadPoolExecutor.java:641)
  at java.lang.Thread.run (Thread.java:920)
```

roshanrajaratnam, 2022-02-14:
@FluffyBunniesTasteTheBest are you facing this issue on the latest version as well?

R0m4in-dooz, 2022-02-14:
To answer your question @roshanrajaratnam, yes it occurs on latest version (3.1.6). Actually we have been facing `ConcurrentModficationException` crashes pretty much from the beginning of our usage of this library (smth like v2.4 was the first version we used)

The places where such crash can happen :
- `ProvisionerDao`
- `GroupDao` (or `GroupEntityDao`?)
- `networkExclusionsToJson` (this one may be our fault tho)

We could not find time to try to fix this issue as we mostly see it only when using `ConfigNodeReset`, which is not a feature our end users use a lot. But we have been collecting stack traces for future work. I may post them here if you want to see what are our crash occurrences ?

roshanrajaratnam, 2022-02-14:
I will use the approach that @Mavericksb has suggested since it's a working solution. All updates and inserts will be done after copying the data in to a new list.

roshanrajaratnam, 2022-02-14:
So while debugging I found that the reason for this to happen only in the above mentioned `Dao` objects is that, an unmodifiable list not being passed for the insert operation.

FluffyBunniesTasteTheBest, 2022-02-14:
@roshanrajaratnam yes, happens with 3.1.6.

R0m4in-dooz, 2022-02-17:
Hello @roshanrajaratnam !
So I've pulled the new release to test the fixes, and it seems better, thanks for the update 🥳
Though the only place I flagged as "maybe our fault" is the one I can still get lol.. Of course you didn't apply the fix to the network exclusions list (by my fault?). But actually, I think it needs the fix too. ([here](https://github.com/NordicSemiconductor/Android-nRF-Mesh-Library/pull/477/files#diff-3b0cb53dd6778865c232f01f254f3d831b1f90ee5addc85a5308d4f915b0e832L194))
```
V/MeshManagerApi(13645): Received network pdu: 0x0076CBD44F998565E7172186B5E3E4DC7C33FB2F5C
V/BaseMeshMessageHandler(13645): Sequence number of received Network PDU: 23
V/NetworkLayer(13645): TTL for received message: 127
V/NetworkLayer(13645): Dst: 0x499C
D/LowerTransportLayer(13645): IV Index of received message: 0
D/LowerTransportLayer(13645): SeqAuth: 23
V/AccessLayer(13645): Received Access PDU 804A
D/fr.dooz.nordic_nrf_mesh.DoozMeshManagerCallbacks(13645): onNetworkUpdated
D/fr.dooz.nordic_nrf_mesh.DoozMeshManagerCallbacks(13645): onNetworkUpdated
D/fr.dooz.nordic_nrf_mesh.DoozMeshManagerCallbacks(13645): onNetworkUpdated
D/fr.dooz.nordic_nrf_mesh.DoozMeshManagerCallbacks(13645): onNetworkUpdated
D/DoozMeshStatusCallbacks(13645): onMeshMessageReceived
D/DoozMeshStatusCallbacks(13645): received a ConfigNodeResetStatus
E/AndroidRuntime(13645): FATAL EXCEPTION: pool-21-thread-1
E/AndroidRuntime(13645): Process: fr.dooz.app, PID: 13645
E/AndroidRuntime(13645): java.util.ConcurrentModificationException
E/AndroidRuntime(13645): 	at java.util.ArrayList$Itr.next(ArrayList.java:860)
E/AndroidRuntime(13645): 	at com.google.gson.internal.bind.CollectionTypeAdapterFactory$Adapter.write(CollectionTypeAdapterFactory.java:96)
E/AndroidRuntime(13645): 	at com.google.gson.internal.bind.CollectionTypeAdapterFactory$Adapter.write(CollectionTypeAdapterFactory.java:61)
E/AndroidRuntime(13645): 	at com.google.gson.internal.bind.TypeAdapterRuntimeTypeWrapper.write(TypeAdapterRuntimeTypeWrapper.java:69)
E/AndroidRuntime(13645): 	at com.google.gson.internal.bind.MapTypeAdapterFactory$Adapter.write(MapTypeAdapterFactory.java:208)
E/AndroidRuntime(13645): 	at com.google.gson.internal.bind.MapTypeAdapterFactory$Adapter.write(MapTypeAdapterFactory.java:145)
E/AndroidRuntime(13645): 	at com.google.gson.Gson.toJson(Gson.java:735)
E/AndroidRuntime(13645): 	at com.google.gson.Gson.toJson(Gson.java:714)
E/AndroidRuntime(13645): 	at com.google.gson.Gson.toJson(Gson.java:669)
E/AndroidRuntime(13645): 	at com.google.gson.Gson.toJson(Gson.java:649)
E/AndroidRuntime(13645): 	at no.nordicsemi.android.mesh.MeshTypeConverters.networkExclusionsToJson(MeshTypeConverters.java:157)
E/AndroidRuntime(13645): 	at no.nordicsemi.android.mesh.MeshNetworkDb.lambda$update$3(MeshNetworkDb.java:200)
E/AndroidRuntime(13645): 	at no.nordicsemi.android.mesh.-$$Lambda$MeshNetworkDb$lIm-UMtxdFiQaNxV0ONcZDAI6eo.run(Unknown Source:4)
E/AndroidRuntime(13645): 	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1162)
E/AndroidRuntime(13645): 	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:636)
E/AndroidRuntime(13645): 	at java.lang.Thread.run(Thread.java:764)
E/AndroidRuntime(13645): FATAL EXCEPTION: pool-21-thread-2
E/AndroidRuntime(13645): Process: fr.dooz.app, PID: 13645
E/AndroidRuntime(13645): java.util.ConcurrentModificationException
E/AndroidRuntime(13645): 	at java.util.ArrayList$Itr.next(ArrayList.java:860)
E/AndroidRuntime(13645): 	at java.util.Collections$UnmodifiableCollection$1.next(Collections.java:1084)
E/AndroidRuntime(13645): 	at androidx.room.EntityDeletionOrUpdateAdapter.handleMultiple(EntityDeletionOrUpdateAdapter.java:86)
E/AndroidRuntime(13645): 	at no.nordicsemi.android.mesh.data.ProvisionedMeshNodesDao_Impl.update(ProvisionedMeshNodesDao_Impl.java:305)
E/AndroidRuntime(13645): 	at no.nordicsemi.android.mesh.MeshNetworkDb.lambda$update$6(MeshNetworkDb.java:227)
E/AndroidRuntime(13645): 	at no.nordicsemi.android.mesh.-$$Lambda$MeshNetworkDb$P_OWMUQ0TcsIRsw0eLJvpw-4dpE.run(Unknown Source:16)
E/AndroidRuntime(13645): 	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1162)
E/AndroidRuntime(13645): 	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:636)
E/AndroidRuntime(13645): 	at java.lang.Thread.run(Thread.java:764)
```

roshanrajaratnam, 2022-02-17:
Thanks for your feedback. You are correct I did forget to include network exclusions in the latest release.

roshanrajaratnam, 2022-02-17:
Could you try this please?

R0m4in-dooz, 2022-02-17:
I just had it again :/ Will try more times again to see

roshanrajaratnam, 2022-02-17:
ok, I haven't released yet. Please try it and let me know thanks.

R0m4in-dooz, 2022-02-17:
This time, I had it when force deleting node (ie. not sending ConfigNodeReset)
```
D/DoozMeshNetwork(19440): should delete the nodeId : 18960
D/fr.dooz.nordic_nrf_mesh.DoozMeshManagerCallbacks(19440): onNetworkUpdated
E/AndroidRuntime(19440): FATAL EXCEPTION: pool-21-thread-1
E/AndroidRuntime(19440): Process: fr.dooz.app, PID: 19440
E/AndroidRuntime(19440): java.util.ConcurrentModificationException
E/AndroidRuntime(19440): 	at java.util.ArrayList$Itr.next(ArrayList.java:860)
E/AndroidRuntime(19440): 	at com.google.gson.internal.bind.CollectionTypeAdapterFactory$Adapter.write(CollectionTypeAdapterFactory.java:96)
E/AndroidRuntime(19440): 	at com.google.gson.internal.bind.CollectionTypeAdapterFactory$Adapter.write(CollectionTypeAdapterFactory.java:61)
E/AndroidRuntime(19440): 	at com.google.gson.internal.bind.TypeAdapterRuntimeTypeWrapper.write(TypeAdapterRuntimeTypeWrapper.java:69)
E/AndroidRuntime(19440): 	at com.google.gson.internal.bind.MapTypeAdapterFactory$Adapter.write(MapTypeAdapterFactory.java:208)
E/AndroidRuntime(19440): 	at com.google.gson.internal.bind.MapTypeAdapterFactory$Adapter.write(MapTypeAdapterFactory.java:145)
E/AndroidRuntime(19440): 	at com.google.gson.Gson.toJson(Gson.java:735)
E/AndroidRuntime(19440): 	at com.google.gson.Gson.toJson(Gson.java:714)
E/AndroidRuntime(19440): 	at com.google.gson.Gson.toJson(Gson.java:669)
E/AndroidRuntime(19440): 	at com.google.gson.Gson.toJson(Gson.java:649)
E/AndroidRuntime(19440): 	at no.nordicsemi.android.mesh.MeshTypeConverters.networkExclusionsToJson(MeshTypeConverters.java:157)
E/AndroidRuntime(19440): 	at no.nordicsemi.android.mesh.MeshNetworkDb.lambda$update$3(MeshNetworkDb.java:200)
E/AndroidRuntime(19440): 	at no.nordicsemi.android.mesh.-$$Lambda$MeshNetworkDb$lIm-UMtxdFiQaNxV0ONcZDAI6eo.run(Unknown Source:4)
E/AndroidRuntime(19440): 	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1162)
E/AndroidRuntime(19440): 	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:636)
E/AndroidRuntime(19440): 	at java.lang.Thread.run(Thread.java:764)
```
my exclusion list is pretty long (like 100+ addresses), maybe it can impact the async task ?

R0m4in-dooz, 2022-02-17:
I gotta move to some others tasks. I will test again tomorrow or next week 👍

roshanrajaratnam, 2022-02-17:
But the library has deprecated asynctasks in the DB operations. I will debug more. Thanks for testing.

roshanrajaratnam, 2022-02-19:
@R0m4in-dooz could you export and share your network with me? I can test against your large exclusion list!

R0m4in-dooz, 2022-02-19:
Sure, there you go :
```json
{
    "meshUUID": "1C64B16B-6E61-4511-9CC5-E8627442CAB4",
    "version": "1.0.0",
    "provisioners": [
        {
            "allocatedGroupRange": [
                {
                    "highAddress": "EA36",
                    "lowAddress": "E741"
                }
            ],
            "allocatedUnicastRange": [
                {
                    "highAddress": "699A",
                    "lowAddress": "619B"
                }
            ],
            "UUID": "E65C18BA-8C51-41F7-AB5C-B1FABDD95A29",
            "allocatedSceneRange": [
                {
                    "lastScene": "8333",
                    "firstScene": "7B34"
                }
            ],
            "provisionerName": "DooZ Mesh Provisioner"
        },
        {
            "UUID": "4311516C-760C-4DB1-A6F2-8D62B7D6D351",
            "allocatedGroupRange": [
                {
                    "highAddress": "E154",
                    "lowAddress": "DE5F"
                }
            ],
            "provisionerName": "DooZ Mesh Provisioner",
            "allocatedSceneRange": [
                {
                    "firstScene": "6334",
                    "lastScene": "6B33"
                }
            ],
            "allocatedUnicastRange": [
                {
                    "highAddress": "519A",
                    "lowAddress": "499B"
                }
            ]
        },
        {
            "allocatedSceneRange": [
                {
                    "lastScene": "4333",
                    "firstScene": "3B34"
                }
            ],
            "allocatedGroupRange": [
                {
                    "highAddress": "D286",
                    "lowAddress": "CF91"
                }
            ],
            "provisionerName": "DooZ Mesh Provisioner",
            "allocatedUnicastRange": [
                {
                    "highAddress": "299A",
                    "lowAddress": "219B"
                }
            ],
[... 3392 lines cut ...]
    "meshName": "843055B3-B7C4-471C-A14C-E7F6947CE09C",
    "appKeys": [
        {
            "boundNetKey": 0,
            "index": 0,
            "key": "D46CBB582C39CA8F7D179E942C720826",
            "name": "Application Key 1"
        },
        {
            "key": "930519C76C26FBF3C5349CFA29C20F95",
            "boundNetKey": 0,
            "name": "Application Key 2",
            "index": 1
        },
        {
            "index": 2,
            "key": "2A390DF9458D7006710607702C9FC507",
            "boundNetKey": 0,
            "name": "Application Key 3"
        }
    ],
    "scenes": [],
    "timestamp": "2022-02-18T13:04:09Z"
}
```

R0m4in-dooz, 2022-02-25:
well, bad news : I had crash upon node reset status and it didn't occur on network exclusions handling
```
V/MeshManagerApi(22342): Received network pdu: 0x00762B15ED7007453044CDCCD444D691BA06F89C5D
V/BaseMeshMessageHandler(22342): Sequence number of received Network PDU: 114741
V/NetworkLayer(22342): TTL for received message: 127
V/NetworkLayer(22342): Dst: 0x61F3
D/LowerTransportLayer(22342): IV Index of received message: 0
D/LowerTransportLayer(22342): SeqAuth: 114741
V/AccessLayer(22342): Received Access PDU 804A
D/fr.dooz.nordic_nrf_mesh.DoozMeshManagerCallbacks(22342): onNetworkUpdated
I/chatty  (22342): uid=10311(u0_a311) fr.dooz.app identical 2 lines
D/fr.dooz.nordic_nrf_mesh.DoozMeshManagerCallbacks(22342): onNetworkUpdated
D/DoozMeshStatusCallbacks(22342): onMeshMessageReceived
D/DoozMeshStatusCallbacks(22342): received a ConfigNodeResetStatus
E/AndroidRuntime(22342): FATAL EXCEPTION: pool-21-thread-2
E/AndroidRuntime(22342): Process: fr.dooz.app, PID: 22342
E/AndroidRuntime(22342): java.util.ConcurrentModificationException
E/AndroidRuntime(22342): 	at java.util.ArrayList$Itr.next(ArrayList.java:860)
E/AndroidRuntime(22342): 	at java.util.Collections$UnmodifiableCollection$1.next(Collections.java:1084)
E/AndroidRuntime(22342): 	at androidx.room.EntityDeletionOrUpdateAdapter.handleMultiple(EntityDeletionOrUpdateAdapter.java:86)
E/AndroidRuntime(22342): 	at no.nordicsemi.android.mesh.data.ProvisionedMeshNodesDao_Impl.update(ProvisionedMeshNodesDao_Impl.java:305)
E/AndroidRuntime(22342): 	at no.nordicsemi.android.mesh.MeshNetworkDb.lambda$update$6(MeshNetworkDb.java:227)
E/AndroidRuntime(22342): 	at no.nordicsemi.android.mesh.-$$Lambda$MeshNetworkDb$P_OWMUQ0TcsIRsw0eLJvpw-4dpE.run(Unknown Source:16)
E/AndroidRuntime(22342): 	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1162)
E/AndroidRuntime(22342): 	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:636)
E/AndroidRuntime(22342): 	at java.lang.Thread.run(Thread.java:764)
```
I didn't carefully check my network JSON maybe there is an issue inside, but i don't see why. Plus this crash not only occur with my network as we can see some users are facing it thanks to monitoring tools..
[image]

roshanrajaratnam, 2022-02-25:
Any node that is reset gets added to the exclusion list. I still haven't had time to test against your json file. I will push a fix, could you test it for me on your network?

FluffyBunniesTasteTheBest, 2022-03-11:
Two of our (~500 active) users experience the same crash @R0m4in-dooz reported. Google Play console shows:
```
java.util.ConcurrentModificationException: 
  at java.util.ArrayList$Itr.next (ArrayList.java:860)
  at java.util.Collections$UnmodifiableCollection$1.next (Collections.java:1089)
  at androidx.room.EntityDeletionOrUpdateAdapter.handleMultiple (EntityDeletionOrUpdateAdapter.java:86)
  at no.nordicsemi.android.mesh.data.ProvisionedMeshNodesDao_Impl.update (ProvisionedMeshNodesDao_Impl.java:317)
  at no.nordicsemi.android.mesh.MeshNetworkDb.lambda$update$6 (MeshNetworkDb.java:233)
  at no.nordicsemi.android.mesh.MeshNetworkDb$$ExternalSyntheticLambda2.run (Unknown Source:16)
  at java.util.concurrent.ThreadPoolExecutor.runWorker (ThreadPoolExecutor.java:1167)
  at java.util.concurrent.ThreadPoolExecutor$Worker.run (ThreadPoolExecutor.java:641)
  at java.lang.Thread.run (Thread.java:923)
```
Beside that, Google Play also logged a few `SQLiteConstraintException` that could be related - both exceptions have a call of `ProvisionedMeshNodeDao_Impl.insert()` in their stack trace.
```
android.database.sqlite.SQLiteConstraintException: 
  at android.database.sqlite.SQLiteConnection.nativeExecuteForLastInsertedRowId (Native Method)
  at android.database.sqlite.SQLiteConnection.executeForLastInsertedRowId (SQLiteConnection.java:1217)
  at android.database.sqlite.SQLiteSession.executeForLastInsertedRowId (SQLiteSession.java:790)
  at android.database.sqlite.SQLiteStatement.executeInsert (SQLiteStatement.java:88)
  at androidx.sqlite.db.framework.FrameworkSQLiteStatement.executeInsert (FrameworkSQLiteStatement.java:51)
  at androidx.room.EntityInsertionAdapter.insert (EntityInsertionAdapter.java:64)
  at no.nordicsemi.android.mesh.data.ProvisionedMeshNodeDao_Impl.insert (ProvisionedMeshNodeDao_Impl.java:301)
  at no.nordicsemi.android.mesh.MeshNetworkDb.lambda$insert$19 (MeshNetworkDb.java:315)
  at no.nordicsemi.android.mesh.MeshNetworkDb$$ExternalSyntheticLambda9.run (Unknown Source:4)
  at java.util.concurrent.ThreadPoolExecutor.runWorker (ThreadPoolExecutor.java:1167)
  at java.util.concurrent.ThreadPoolExecutor$Worker.run (ThreadPoolExecutor.java:641)
  at java.lang.Thread.run (Thread.java:923)
```

roshanrajaratnam, 2022-03-11:
@FluffyBunniesTasteTheBest  did you test against version 3.1.7? Although I have not had time to test the the reset issue @R0m4in-dooz has faced yet.

FluffyBunniesTasteTheBest, 2022-03-11:
@roshanrajaratnam Thanks for your help! 
> did you test against version 3.1.7?

Not yet, the current app in production is still based on 3.1.6, but I've merged some of the fixes [
9bd1a82](https://github.com/NordicSemiconductor/Android-nRF-Mesh-Library/commit/9bd1a82b67877820c3e73b8e620dcbe1bb55a68f) between 3.1.6 and 3.1.7. 

Unfortunately I don't even know what triggered the crash. Problem is that privacy is very important to our target audience, so there's no analytics/crash reporting implemented other than what Android provides, which, unfortunately, keep us clueless about how to provoke the issue. Anyway, I'll publish an update with 3.1.7 next week.

FluffyBunniesTasteTheBest, 2022-03-16:
@roshanrajaratnam Thanks for your help! 

Unfortunately it happens with 3.1.7 as well. This time from line 195 at ProvisionersDao_Impl.java. The previous crash happened at line 317 at ProvisionedMeshNodesDao_Impl.java.

```
java.util.ConcurrentModificationException
    at java.util.ArrayList$Itr.next(ArrayList.java:860)
    at java.util.Collections$UnmodifiableCollection$1.next(Collections.java:1089)
    at androidx.room.EntityInsertionAdapter.insert(EntityInsertionAdapter.java:95)
    at no.nordicsemi.android.mesh.data.ProvisionersDao_Impl.insert(ProvisionersDao_Impl.java:159)
    at no.nordicsemi.android.mesh.MeshNetworkDb.lambda$insertNetwork$0(MeshNetworkDb.java:151)
    at no.nordicsemi.android.mesh.MeshNetworkDb$$ExternalSyntheticLambda29.run(Unknown Source:16)
```

Would be nice if this could be fixed soon.

roshanrajaratnam, 2022-03-17:
@FluffyBunniesTasteTheBest  will take a look at this issue today and get back to you!

roshanrajaratnam, 2022-03-17:
question here though, what do you mean by adding nodes? are you referring to provisioning them in to the network?

roshanrajaratnam, 2022-03-22:
@FluffyBunniesTasteTheBest I think I have found where the `ConcurrentModificationException` is  being thrown during the import and also when removing an item from the list of `networkExclusions`. Could you give this a spin please?

daretobeorjan, 2022-03-23:
Hi, watching this issue since we have similar problems.

Doesn't the new getNetworkExclusions also need to make a copy of the ArrayList(s) that is the value of the map as well?

roshanrajaratnam, 2022-03-23:
Hi yes I am working on this at the moment. I would say that returning an unmodifiable list should work when inserting to db. What is your opinion on this?

daretobeorjan, 2022-03-23:
Yeah, I agree. I did a quick test with a re-implementation of getNetworkExclusions as such:
```
        return networkExclusions.entrySet().stream()
                .collect(Collectors.toMap(Map.Entry::getKey, e -> new ArrayList(e.getValue())));
```
It seemed to get rid of the concurrent modification crashes completely. Probably shouldn't use streams, but I mostly work with Kotlin so it's the quickest way to prototype something for me.

roshanrajaratnam, 2022-03-23:
Good point. So it could be that the arraylist within the map is not an unmodifiableList causing this crash. Unfortunately cannot use streams on my end due to API limitations. I will push a fix soon, if you can please give it a test because I am having a hard time reproducing this.

roshanrajaratnam, 2022-03-23:
@R0m4in-dooz I think I may have found the issue relating to the `ConcurrentModificationException` when sending a ConfigNodeReset According my logs it seems like a timing issue following update method is called for , currently the following update is called when sending a message and I notice that for larger networks the db update is not completed even until the ConfigNodeResetStatus is received causing this ConcurrentModificationException to be thrown when calling  `network.getNodes()`. The only way around it seemed to be @Mavericksb solution of copying the list in to a new list because creating an unmodifiable list seems to not work in this case due to timing in my opinion.

https://github.com/NordicSemiconductor/Android-nRF-Mesh-Library/blob/cb127405c5acc859ea39e360bf195944afe587ec/mesh/src/main/java/no/nordicsemi/android/mesh/MeshNetworkDb.java#L211-L231

In addition to this, I've also made the `getNetworkExclusions()` return an unmodifiable map containing an unmodifiable list of addresses for a given IV Index.

R0m4in-dooz, 2022-03-23:
Hello @roshanrajaratnam !
Ok i may have time to test it this week. I should pull latest dev HEAD I suppose ?

closed 2022-03-28 by roshanrajaratnam

FluffyBunniesTasteTheBest, 2022-03-30:
@roshanrajaratnam please excuse the delay, it's a little busy right now. 

The `ConcurrentModificationException` unfortunately still happens with version 3.1.8: 

```
E/AndroidRuntime( 7576): java.util.ConcurrentModificationException
E/AndroidRuntime( 7576): 	at java.util.ArrayList$Itr.next(ArrayList.java:860)
E/AndroidRuntime( 7576): 	at java.util.Collections$UnmodifiableCollection$1.next(Collections.java:1089)
E/AndroidRuntime( 7576): 	at com.google.gson.internal.bind.CollectionTypeAdapterFactory$Adapter.write(CollectionTypeAdapterFactory.java:96)
E/AndroidRuntime( 7576): 	at com.google.gson.internal.bind.CollectionTypeAdapterFactory$Adapter.write(CollectionTypeAdapterFactory.java:61)
E/AndroidRuntime( 7576): 	at com.google.gson.internal.bind.TypeAdapterRuntimeTypeWrapper.write(TypeAdapterRuntimeTypeWrapper.java:69)
E/AndroidRuntime( 7576): 	at com.google.gson.internal.bind.MapTypeAdapterFactory$Adapter.write(MapTypeAdapterFactory.java:208)
E/AndroidRuntime( 7576): 	at com.google.gson.internal.bind.MapTypeAdapterFactory$Adapter.write(MapTypeAdapterFactory.java:145)
E/AndroidRuntime( 7576): 	at com.google.gson.Gson.toJson(Gson.java:735)
E/AndroidRuntime( 7576): 	at com.google.gson.Gson.toJson(Gson.java:714)
E/AndroidRuntime( 7576): 	at com.google.gson.Gson.toJson(Gson.java:669)
E/AndroidRuntime( 7576): 	at com.google.gson.Gson.toJson(Gson.java:649)
E/AndroidRuntime( 7576): 	at no.nordicsemi.android.mesh.MeshTypeConverters.networkExclusionsToJson(MeshTypeConverters.java:157)
E/AndroidRuntime( 7576): 	at no.nordicsemi.android.mesh.MeshNetworkDb.lambda$update$6(MeshNetworkDb.java:223)
E/AndroidRuntime( 7576): 	at no.nordicsemi.android.mesh.MeshNetworkDb$$ExternalSyntheticLambda1.run(Unknown Source:16)
E/AndroidRuntime( 7576): 	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1167)
E/AndroidRuntime( 7576): 	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:641)
E/AndroidRuntime( 7576): 	at java.lang.Thread.run(Thread.java:919)
```

It's quite easy to provoke the crash using a test routine that:
1. Scans for unprovisioned devices and takes the first device found.
2. Provisions the device.
3. Sends a `ConfigNodeReset` message to unprovision the node.

I'm using a one second delay between each of the steps and it repeatedly crashes between 30-50 iterations.

roshanrajaratnam, 2022-03-31:
@FluffyBunniesTasteTheBest sorry it's been a tricky one to fix here, I see that this is happening when updating the db. I have included a fix that copies the network exclusions to a new map before updating the db. Please give it a shot and let me know how it goes.

FluffyBunniesTasteTheBest, 2022-03-31:
@roshanrajaratnam Thanks for the quick fix! 

~~It looks promising so far: 120 iterations without any issues.~~ Argh... Murphy's Law! Crash just while writing these  lines...

```
E/AndroidRuntime(10318): java.util.ConcurrentModificationException
E/AndroidRuntime(10318): 	at java.util.ArrayList$Itr.next(ArrayList.java:860)
E/AndroidRuntime(10318): 	at java.util.Collections$UnmodifiableCollection$1.next(Collections.java:1089)
E/AndroidRuntime(10318): 	at com.google.gson.internal.bind.CollectionTypeAdapterFactory$Adapter.write(CollectionTypeAdapterFactory.java:96)
E/AndroidRuntime(10318): 	at com.google.gson.internal.bind.CollectionTypeAdapterFactory$Adapter.write(CollectionTypeAdapterFactory.java:61)
E/AndroidRuntime(10318): 	at com.google.gson.internal.bind.TypeAdapterRuntimeTypeWrapper.write(TypeAdapterRuntimeTypeWrapper.java:69)
E/AndroidRuntime(10318): 	at com.google.gson.internal.bind.MapTypeAdapterFactory$Adapter.write(MapTypeAdapterFactory.java:208)
E/AndroidRuntime(10318): 	at com.google.gson.internal.bind.MapTypeAdapterFactory$Adapter.write(MapTypeAdapterFactory.java:145)
E/AndroidRuntime(10318): 	at com.google.gson.Gson.toJson(Gson.java:704)
E/AndroidRuntime(10318): 	at com.google.gson.Gson.toJson(Gson.java:683)
E/AndroidRuntime(10318): 	at com.google.gson.Gson.toJson(Gson.java:638)
E/AndroidRuntime(10318): 	at com.google.gson.Gson.toJson(Gson.java:618)
E/AndroidRuntime(10318): 	at no.nordicsemi.android.mesh.MeshTypeConverters.networkExclusionsToJson(MeshTypeConverters.java:157)
E/AndroidRuntime(10318): 	at no.nordicsemi.android.mesh.MeshNetworkDb.lambda$update$6(MeshNetworkDb.java:221)
E/AndroidRuntime(10318): 	at no.nordicsemi.android.mesh.MeshNetworkDb$$ExternalSyntheticLambda1.run(Unknown Source:16)
E/AndroidRuntime(10318): 	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1167)
E/AndroidRuntime(10318): 	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:641)
E/AndroidRuntime(10318): 	at java.lang.Thread.run(Thread.java:919)
```
Happened with dev branch at a0774c137c4a5b76ed6066b53d40f611394d2f26.

commit 9bd1a82b (2022-02-15): Fixes #432 by passing an unmodifiable list for insert and update operations.
PR #477 (2022-02-16): Release v3.1.7
This release includes the following changes.
* Fixes #432
* Fixes #438
* Fixes #441 
* #448 Adds support for Scheduler, Generic Transition time, PowerUpBehaviour, GenericProperties (Admin, User, Manufacturer) and Time Messages
* #456 maintains consistency.
* Fixes #457 
* Fixes #458, where the provisioner's sequence number seems to get reset when sending a message to 0xFF
* Fixes #461 and #462 that was causing the app/library to crash when de-obfuscating a received message.
* #464 adds support for Android 12 `BLUETOOTH_SCAN`, `BLUETOOTH_CONNECT` permissions and splash screen. Th [... cut]
commit ec0f15c0 (2022-02-17): Fixes #432 network exclusions list must be unmodifiable.
commit f276d094 (2022-03-22): Contains a possible fix #432 by using an unmodifiable list when importing a mesh network.