nordicsemi/Android-nRF-Mesh-Library #433: Unknown pattern character 'X'

DevHachi, 2021-06-17:
---

**Describe the bug**
When trying to load the mesh beacon data the app crashes when trying to get mesh beacon data from scan record.

**To Reproduce**
Steps to reproduce the behavior:
1. Scan for device
2. call meshManagerApi.getMeshBeaconData with scan record bytes

**Expected behavior**
should not crash.

**Platform details:**
 - Device: Huawei P8
 - OS: Android 6.0
 - Library Version: 3.1.5

**Logs / Screenshots**
```java.lang.ExceptionInInitializerError
        at no.nordicsemi.android.mesh.MeshManagerApi.isMeshBeacon(MeshManagerApi.java:651)
        at no.nordicsemi.android.mesh.MeshManagerApi.getMeshBeaconData(MeshManagerApi.java:666)
        at com.quimron.ble_mesh.BleMesh.identifyNode(BleMesh.kt:275)
        at com.quimron.ble_mesh.BleMesh.access$identifyNode(BleMesh.kt:27)
        at com.quimron.ble_mesh.BleMesh$addDeviceToMesh$2$1.invoke(BleMesh.kt:188)
        at com.quimron.ble_mesh.BleMesh$addDeviceToMesh$2$1.invoke(BleMesh.kt:187)
        at com.quimron.ble_mesh.BleMesh.connectToDevice$lambda-26$lambda-25(BleMesh.kt:367)
        at com.quimron.ble_mesh.BleMesh.lambda$NcKof_1TOBSGM-QidILhLgmLnZg(BleMesh.kt)
        at com.quimron.ble_mesh.-$$Lambda$BleMesh$NcKof_1TOBSGM-QidILhLgmLnZg.onRequestCompleted(lambda)
        at no.nordicsemi.android.ble.Request.lambda$notifySuccess$1$Request(Request.java:1203)
        at no.nordicsemi.android.ble.-$$Lambda$Request$ep6TD_94xCe5LaEafH2PrarR3yc.run(lambda)
        at android.os.Handler.handleCallback(Handler.java:743)
        at android.os.Handler.dispatchMessage(Handler.java:95)
        at android.os.Looper.loop(Looper.java:150)
        at android.app.ActivityThread.main(ActivityThread.java:5546)
        at java.lang.reflect.Method.invoke(Native Method)
        at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:794)
        at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:684)
     Caused by: java.lang.IllegalArgumentException: Unknown pattern character 'X'
        at java.text.SimpleDateFormat.validatePatternCharacter(SimpleDateFormat.java:323)
        at java.text.SimpleDateFormat.validatePattern(SimpleDateFormat.java:312)
        at java.text.SimpleDateFormat.<init>(SimpleDateFormat.java:365)
        at no.nordicsemi.android.mesh.utils.MeshParserUtils.<clinit>(MeshParserUtils.java:47)
        at no.nordicsemi.android.mesh.MeshManagerApi.isMeshBeacon(MeshManagerApi.java:651) 
        at no.nordicsemi.android.mesh.MeshManagerApi.getMeshBeaconData(MeshManagerApi.java:666) 
        at com.quimron.ble_mesh.BleMesh.identifyNode(BleMesh.kt:275) 
        at com.quimron.ble_mesh.BleMesh.access$identifyNode(BleMesh.kt:27) 
        at com.quimron.ble_mesh.BleMesh$addDeviceToMesh$2$1.invoke(BleMesh.kt:188) 
        at com.quimron.ble_mesh.BleMesh$addDeviceToMesh$2$1.invoke(BleMesh.kt:187) 
        at com.quimron.ble_mesh.BleMesh.connectToDevice$lambda-26$lambda-25(BleMesh.kt:367) 
        at com.quimron.ble_mesh.BleMesh.lambda$NcKof_1TOBSGM-QidILhLgmLnZg(BleMesh.kt) 
        at com.quimron.ble_mesh.-$$Lambda$BleMesh$NcKof_1TOBSGM-QidILhLgmLnZg.onRequestCompleted(lambda) 
        at no.nordicsemi.android.ble.Request.lambda$notifySuccess$1$Request(Request.java:1203) 
        at no.nordicsemi.android.ble.-$$Lambda$Request$ep6TD_94xCe5LaEafH2PrarR3yc.run(lambda) 
        at android.os.Handler.handleCallback(Handler.java:743) 
        at android.os.Handler.dispatchMessage(Handler.java:95) 
        at android.os.Looper.loop(Looper.java:150) 
        at android.app.ActivityThread.main(ActivityThread.java:5546) 
        at java.lang.reflect.Method.invoke(Native Method) 
        at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:794) 
        at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:684) ```

roshanrajaratnam, 2021-06-17:
Thanks for reporting and this should be caught in the app.

roshanrajaratnam, 2021-06-17:
Hi what firmware are you using to test this?

DevHachi, 2021-06-17:
You mean the firmware of the lights?
If so I can't tell you because I don't know. Those are Toran lights.

DevHachi, 2021-06-17:
From my perspective it seems like this is the culprit:
`private static final SimpleDateFormat SDF = new SimpleDateFormat("yyyy-MM-dd'T'HH:mm:ssX", Locale.US);`
Because 'X' is not supported on devices before nougat, which an android 6 device definitely is.

roshanrajaratnam, 2021-06-17:
this is indeed hte culprit. I will push a fix later today

roshanrajaratnam, 2021-06-21:
@DevHachi check this please. the last fix introduced a timestamp parsing issue however now is handled properly and maintains compatibility between both Android and iOS.

PR #437 (2021-06-23): Release v3.1.6
This release brings the following changes
* Specify the address of the element instead of the node address when sending message from the SensorServerActivity.java 
* Fixes #423, #418, #333 related to database migration
* Fixes #354 an invalid GATT level segmentation issue. The lib was considering the MTU size instead of the SAR bits in the proxy PDU
* Fixes #427, #406 where segmented messages to be retransmitted being get cleared. This should improve the retransmissions
* Fixes #433 where the app was crashing on Android 6.0 devices due incorrect time stamp formatting.
* Migrates [... cut]
closed 2021-06-23 by roshanrajaratnam