nordicsemi/Android-nRF-Mesh-Library #439: How to list and select meshes?

Rawa, 2021-06-30:
Scenario: I have one device and want to maintain multiple meshes, how would I use the library?

It seems like the database for this library supports having multiple meshes, however from the entry point of the `MeshManagerApi` there is only `loadMeshNetwork()` which will generate a mesh if none exists. 

What is the intended way of selecting an existing mesh? There doesn't seem to be any function to list all the meshes in the database? Are one supposed to store the meshes outside of this library in JSON format and then load it in once wanting to use it (import and export mesh by JSON)?

roshanrajaratnam, 2021-06-30:
Hi @Rawa, you are correct the database was designed in such a way that multiple mesh networks can be stored. As of now we don't have the intended API ready to list multiple mesh networks yet. This is something I have in the backlog. 

You are correct, this is one way to do it until the API is ready.

Rawa, 2021-07-01:
Ok, thanks for the reply. :)

closed 2021-07-01 by Rawa