nordicsemi/Android-nRF-Mesh-Library #440: Question : select node as provisioner

R0m4in-dooz, 2021-07-06:
Hello !

Is it possible to elevate a previously provisioned node to be a provisioner of the n/w ?

roshanrajaratnam, 2021-07-06:
Hey, if you are referring to a physical node, I am not sure if it's possible. A provisioner is a special node/client that is a part of the network that can provision without having to be provisioned. I don't think the mesh profile specification states anything about this either.

R0m4in-dooz, 2021-07-06:
Alright ! Yes it was more of a naive question, I didn't check spec. We have a special node that will need to be provisioner so we can improve its capabilities. Then ok, we'll see how to handle. Thx

closed 2021-07-06 by R0m4in-dooz

roshanrajaratnam, 2021-07-06:
Well one more thing is that, this is kind of an application specific feature and there is nothing stopping you from implementing such a feature. :)

R0m4in-dooz, 2021-07-06:
Yes of course, I just wanted to quick check with you if it was smth that the ADK was prepared for. My main concern about the feature request for our app, is to always be in sync with what this special node will do in the network... From what I understood with embedded team, it may be autonomous and I'm afraid that our app will loose track of changes in the network and could cause huge bugs. We will have to work hard for that ;)
Thanks