nordicsemi/Android-nRF-Mesh-Library #441: Prevent nextAvailableUnicastAddress() from returning an already assigned address.

NanaDada, 2021-07-18:
---

**Describe the bug**
`nextAvailableUnicastAddress()` will return an already assigned unicast address. 

A new node is added to a mesh by:
1. Provisioning the device.
2. Connecting to the new node and...
3. (only if the connection is established) assigning and binding an app key by sending `ConfigDefaultTtlGet`, `ConfigNetworkTransmitSet`, `ConfigAppKeyAdd`, `ConfigModelAppBind`.
    
If step 1. succeeds, but step 2. fails, then the library gets inconsistent with the actual mesh. The effect is that `nextAvailableUnicastAddress()` will return an already (at step 1.) assigned unicast address.  

This happens because the library updates its internal list of nodes (`BaseMeshNetwork.nodes`) in response of the commands `ConfigDefaultTtlGet`, `ConfigNetworkTransmitSet`, `ConfigAppKeyAdd`, `ConfigModelAppBind`. Now if the connection to the mesh fails (step 2.), then non of those commands are sent and hence the library does not update its internal list of nodes. If a new node is now provisioned, `nextAvailableUnicastAddress()` will returns an already assinged address. (The same address as the node that failed to connect).

**To Reproduce**
The easiest way to reproduce this is by provisioning (and assigning app keys to) a bunch of devices in a loop. If one device fails to connect, then the following device will get the same address assigned, causing a "unicast address already in use" error to show up in logcat.

**Expected behavior**
`nextAvailableUnicastAddress()` must not return an already used unicast address. The only way to fix this is if the library updates its internal list of nodes as soon as a new node is provisioned - i.e. before connecting to the newly provisioned node.

**Platform details:**
Happens on all Android versions.

roshanrajaratnam, 2021-07-20:
Hi will have to look at this after getting back from vacation.

NanaDada, 2021-07-25:
Another way to reproduce the issue is by provisioning multiple devices without ever connecting to a node.

Rawa, 2021-09-14:
We are having this issue as well, is there a workout to avoid this issue?

philips77, 2021-09-14:
Hello,
Currently we are busy with another project, but will get back to mesh in coming weeks.

roshanrajaratnam, 2021-10-30:
@lenguyenthanh I have a few more fixes to do before I release a snapshot. You could clone this repo as it is and give it a shot.

roshanrajaratnam, 2021-12-09:
Closing due to inactivity and is not reproduceable anymore on my end.

PR #477 (2022-02-16): Release v3.1.7
This release includes the following changes.
* Fixes #432
* Fixes #438
* Fixes #441 
* #448 Adds support for Scheduler, Generic Transition time, PowerUpBehaviour, GenericProperties (Admin, User, Manufacturer) and Time Messages
* #456 maintains consistency.
* Fixes #457 
* Fixes #458, where the provisioner's sequence number seems to get reset when sending a message to 0xFF
* Fixes #461 and #462 that was causing the app/library to crash when de-obfuscating a received message.
* #464 adds support for Android 12 `BLUETOOTH_SCAN`, `BLUETOOTH_CONNECT` permissions and splash screen. Th [... cut]
closed 2021-12-09 by roshanrajaratnam