nordicsemi/Android-nRF-Mesh-Library #442: Could not get unknown property 'sonatypeStagingProfileId' for object

123yangu, 2021-07-29:
Caused by: groovy.lang.MissingPropertyException: Could not get unknown property 'sonatypeStagingProfileId' for object of type io.github.gradlenexus.publishplugin.NexusRepository.

philips77, 2021-07-29:
Yes, true. That's needed only for releasing. The issue only happens when you try to build the project without some constants in local.properties file. If you clone the repo and add the mesh module to your project this issue should not occur. If you, however, want to compile this project for some reason, just remove/comment out the references to release scripts in build.gradle files.

philips77, 2021-07-29:
E.g. this lines:
https://github.com/NordicSemiconductor/Android-nRF-Mesh-Library/blob/e03b1eba6cf4d7bb41b1fa36d3ce8055e8df4208/build.gradle#L54-L56

closed 2021-07-29 by philips77