nordicsemi/Android-nRF-Mesh-Library #443: How to operate one mesh network with multiple devices/app instances at the same time?

NanaDada, 2021-08-11:
**Describe the bug**
Unable to use a mesh network on multiple Android devices at the same time.

**To Reproduce**
Steps to reproduce the behavior:
1. Export a mesh on one Android device by calling `MeshManagerApi.exportMeshNetwork()` 
2. Import the mesh on another Android device by calling `MeshManagerApi.importMeshNetwork()`
3. Connect with both devices to the mesh network.
4. Only one device can use the mesh network. Commands from the second device a ignored.

**Expected behavior**
Both devices should be able to use the mesh.

**Platform details:**
 - Device: All
 - OS: Android
 - Library Version (or Commit) 3.1.6

---

Imagine you want to operate a Mesh network with multiple Android devices. How do you copy the mesh across devices, so that 
all devices can use the mesh at the same time (i.e. in parallel)?

To share the mesh across devices, my app calls `MeshManagerApi.exportMeshNetwork()` and `MeshManagerApi.importMeshNetwork()`. This is basically working: Reboot the mesh, start the app on any of the devices and you can connect and use the mesh. So far, so good.

The problem arises as soon as a second instance of the app connects to the mesh: Then all commands sent from the 
second instance are simply ignored. Now the question is why, and how to fix it?

Do you have any ideas what could be wrong here? Is it sufficient to just call `MeshManagerApi.exportMeshNetwork()` and `MeshManagerApi.importMeshNetwork()`, or is there anything else that needs to be done to allow multiple instances access the mesh at the same time?

roshanrajaratnam, 2021-08-11:
So do you use the same provisioner from the second instance? if so the messages sent from the second instance would be ignore because of the sequence numbers keep incrementing. So you would have to create a new provisioner on the second device with a different address and this should help I believe.

NanaDada, 2021-08-19:
Thanks for the hint! I've ended up adding a bunch of provisioners and a setting to chose one.

closed 2021-08-19 by NanaDada