nordicsemi/Android-nRF-Mesh-Library #446: ProvisionedBaseMeshNode.setNodeName() doesn't change a node's name.

NanaDada, 2021-08-19:
---

**Describe the bug**

The name of a node can be changed by calling `ProvisionedBaseMeshNode.setNodeName()`. This however only effects the calling app. The actual mesh node isn't effected by it. 

**Expected behavior**
The node whose name was changed should advertise itself with the new name.

**Platform details:**
 - Device: Any
 - OS: Android
 - Library Version: 3.1.6

roshanrajaratnam, 2021-08-20:
Hi, to update the provisioner you need to use Provisioner.setProviosionerName and then call., Network.updateProvisioner with the provisioner object.

closed 2021-08-20 by roshanrajaratnam

NanaDada, 2021-08-20:
Dear Roshan Rajaratnam,

Thanks for your help! 

Just to clarify a little further: After changing a node's name, I'd like to have the node advertise itself with the newly changed name. So if you scan for BLE devices with another smartphone, the new name should be visible. 

The node whose name should be changed is a simple light bulb that only contains a generic level/onoff server model - i.e. it is not a provisioner, so I got this intuition that `Provisioner.setProviosionerName()` and `Network.updateProvisioner()` might not lead to the desired effect.

roshanrajaratnam, 2021-08-20:
You are correct, as of now there is no way to change the advertised name of a device.

NanaDada, 2021-08-20:
Thanks for the clarification. 

Imagine somebody wants to implement it anyway, what needs to be done? How would you tackle that requirement?

NanaDada, 2021-08-24:
> It's information for the user stored in the internal database within the app.

This sounds like a huge design flaw...

Imagine you change some node names from Node1, Node2, Node3 to Kitchen, Attic, Basement. 
Now, if you're operating the mesh network with more than one smartphone, and you want to have the new names applied to each of them, (which seems like an obvious idea considering we're talking about a mesh network), you have to either:
* change the names on every smartphone or
* export the mesh and re-import it on every smartphone.

What's bothering me is how do you explain to your clients/customers, that you're having an app to operate a mesh network that can't sync device names over the network? From a user's point of view, it just doesn't make much sense...

It seems this issue should be re-opened.

roshanrajaratnam, 2021-08-24:
I get your point but this is how the mesh specification is defined. You are free to implement a we service that could sync the exported mesh network across devices. Remember this library supports comfiguring and controlling mesh networks but depending on the application you are free to extend your own functionality since this requirement is application dependant.

NanaDada, 2021-08-24:
Dear Roshan Rajaratnam,

thanks for the clarification. 

> this is how the mesh specification is defined.

Predetermined inconsistency. Makes it look like the mesh specification was just halfheartedly tinkered together. But OK, let's implement a _Proprietary-Node-Name-Server-Model_ then.

philips77, 2021-08-24:
Mind, that provisioners (phones) need to be synchronized using the Internet. For example if a new Node is added this can only be shared this way. So the names can be shared the same way.
In mesh it is more important what the node does, than what is the node name. Recommended way of sending messages is by group address, to be able to replace the node with a different one and subscribing the new one to the same groups, instead of reconfiguring all existing nodes to publish to new unicast address. So you don't call nodes, but rather groups. And also, renaming a group also need sharing network configuration via Internet.

NanaDada, 2021-08-24:
Hi Philips,
Thanks for the additional clarification. 

Unfortunately we can't use the internet to synchronize the provisioners - the niche we're serving simply doesn't allow it. So I must find another way. Something like this might work:

1. The node gets a proprietary-node-data-server-model (PNDSM) implemented. It saves node specific data at the node's flash memory and provides the data upon request. 
2. Every time any of the node's data changes, it's saved by the PNDSM. This includes the node's name, the goups it's subscribed/publishing to along with the group names. Just anything that's needed to recover a node's latest configuration.
3. Group data is especially interesting here, as we need a sequence number or timestamp to be able to find a group's last modification. It could happen that group data gets inconsistent across nodes - it's even very likely considering a mesh could be spread over various locations. 
4. Every time an app connects, it broadcasts a "hello" message to all nodes, which respond with all of the stored data. 
5. In case a node replies newer data than the app is aware of (like a changed name), the app updates its data.
6. If a node replies some older data then the app is aware of, then the node's PNDSM gets updated.
7. Both app and nodes are happily synchronized ever after.

Well, at least in theory... 

Having the node configuration data spread across the node itself (1), in the PNDSM (2), in the mesh library (3) and in the app itself (4), is getting messy and very error prone. Its going to be an ugly hack par excellence, but it should be doable. 

Roshan, Aleksander, you're the experts, what's your point on it? Beside maintaining the node data at four different places, do you see any other obvious flaws in that approach? 

Your advise is very much appreciated,
Thank you!

roshanrajaratnam, 2021-08-25:
Hi, one other thing is that you can look in to storing the json (or atleast some information that you think is important for your usecase) on the proxy node of the network. This way you don't need to send a hello to all the nodes. You could sync the information on the phone with the proxy to see if there have been any changes.