nordicsemi/Android-nRF-Mesh-Library #454: Better handling of secure network beacon

DomiR, 2021-11-17:
On android handling of the secure network beacon is not implemented in a way that it handles ivIndex updates correctly.

---

**Describe the bug**
When receiving a secure network beacon with an increased ivIndex this library does not seem to validate the secure network beacon. Instead it calculates a local network beacon (potentially with a lesser ivIndex, which then will fail) and compares the authenticationValues. This comparison should always fail as soon as the iv indice are not the same any more. Instead we need to use the beacon key (which is derived from the network key) to caclulate the hash for the data from the received beacon and compare it to the authenticationValue from the received beacon to validate it.

https://github.com/NordicSemiconductor/Android-nRF-Mesh-Library/blob/e03b1eba6cf4d7bb41b1fa36d3ce8055e8df4208/mesh/src/main/java/no/nordicsemi/android/mesh/MeshManagerApi.java#L298

The iOS lib does this correctly:
https://github.com/NordicSemiconductor/IOS-nRF-Mesh-Library/blob/f4a80ab641534449b331dbed4583be613dcb7b0e/nRFMeshProvision/Classes/Layers/Network%20Layer/SecureNetworkBeacon.swift#L76

**To Reproduce**
Steps to reproduce the behavior:
1. Use the devboard with ivindex update
2. Increase ivindex
3. Debug receiving secure network beacon

**Expected behavior**
Should be handled the same way as iOS.

**Platform details:**
 - Device: All
 - OS: Any
 - Library Version: Master (e03b1eb)

**Logs / Screenshots**
-

roshanrajaratnam, 2021-11-17:
@DomiR thanks for catching this. I will take a look in to this soon currently a bit busy with some other project. We would gladly accept a pull request if you get around it.

roshanrajaratnam, 2021-12-10:
Hey @DomiR i'm testing the iv index update and it seems to authenticate the SNB for an increased IV index. The local `SecureNetworkBeacon`  is generated based on the received `SecureNetworkBeacon`s iv index so this allows us to generate the authentication values to match the SNBs. How are you testing this?

roshanrajaratnam, 2022-02-14:
@DomiR  any updates on this?

DomiR, 2022-02-14:
I just checked this again and saw that you're right. I guess I missed that you use the ivIndex from the receiving message to calculate the "localSecureNetworkBeacon". I probably was on an older version of this repo where this was calculated using the network ivIndex. Sorry for wasting your time and thanks for looking into this 🙏 

I think a small refactor to move this authentication into the SecureBeacon class would be nice and would need 3 lines of code. In SecureNetworkBeacon add: 
```java
public boolean authenticateWithNetKey(final byte[] n) {
        final byte[] netId = SecureUtils.calculateK3(n);
        final byte[] calcAuthValue = SecureUtils.calculateAuthValueSecureNetBeacon(n, flags , netId, ivIndex.getIvIndex());
        return Arrays.equals(calcAuthValue, authenticationValue);
}
```

closed 2022-02-14 by DomiR