nordicsemi/Android-nRF-Mesh-Library #457: send unacknowledged message should not open DialogFragmentTransactionStatus

xiangxiangjiang, 2021-11-23:
unacknowledged message will not get the response from the remote node according to the ble mesh specification.
when I send unacknowledged message to the remote node, the app show the DialogFragmentTransactionStatus with R.string.operation_timed_out after a few seconds, it seems to wait for respons message.

roshanrajaratnam, 2021-11-23:
@xiangxiangjiang could you confirm if the message is received on the node? Is it sent via a proxy node?

xiangxiangjiang, 2021-11-24:
yes, I'm sure. I use the TI's simplelink launchpad cc2652r with sdk simplelink_cc13x2_26x2_sdk_5_20_00_52. My device was set as relay and proxy node, the node received the unacknowledged  message that I send using vendor model controls in Vendor Model Actvity with Acknoledged Message checkbox unchecked and printed the message on the screen. Because the message from mobile is unack type, the node will not send the response message to the originor. And after I read the android app source code, I find that the sendmessage method start a runable to wait for response whatever the message type is ack or unack. 

刘道明|daoming北京理工大学
手机：18811553599
***@***.***

roshanrajaratnam, 2021-12-07:
Hi is your vendor model message a segmented message? The runnable you are referring to is related to segmented messages.

roshanrajaratnam, 2021-12-08:
Also you have a point here. Will fix this!

PR #477 (2022-02-16): Release v3.1.7
This release includes the following changes.
* Fixes #432
* Fixes #438
* Fixes #441 
* #448 Adds support for Scheduler, Generic Transition time, PowerUpBehaviour, GenericProperties (Admin, User, Manufacturer) and Time Messages
* #456 maintains consistency.
* Fixes #457 
* Fixes #458, where the provisioner's sequence number seems to get reset when sending a message to 0xFF
* Fixes #461 and #462 that was causing the app/library to crash when de-obfuscating a received message.
* #464 adds support for Android 12 `BLUETOOTH_SCAN`, `BLUETOOTH_CONNECT` permissions and splash screen. Th [... cut]
closed 2022-02-16 by roshanrajaratnam