nordicsemi/Android-nRF-Mesh-Library #458: Sequence number is not properly updated when broadcasting segmented VendorModelMessageAcked messages.

FluffyBunniesTasteTheBest, 2021-11-29:
**Describe the bug**

When broadcasting a segmented `VendorModelMessageAcked`, the sequence number seems to be incremented only once, causing the firmware to discard subsequent messages. 

Let's start with the **OK** case: 
Here's an example of a `VendorModelMessageAcked` that's transmitted in three segments. The message was sent to a single node twice in a row, and the following was logged: 

```
 V/MeshTransport: Src address: 0001
 V/MeshTransport: Dst address: 0002
 ...
 V/MeshTransport: Sequence number: 10
 V/MeshTransport: Access message opcode: 3
 V/MeshTransport: Access message parameters: 04110101220202330303440404414243444546474800
 V/AccessLayer: Created Access PDU C3123404110101220202330303440404414243444546474800
 V/UpperTransportLayer: Application nonce: 010000000A0001000200000000
 V/UpperTransportLayer: Encrypted upper transport pdu: 09FFF90E242966A8FE2544C976E15A5AEB765BBFA1313BC79D0BF607CD
 V/LowerTransportLayer: Segmented Lower transport access PDU: D900280209FFF90E242966A8FE2544C9 0 of 3
 V/LowerTransportLayer: Segmented Lower transport access PDU: D900282276E15A5AEB765BBFA1313BC7 1 of 3
 V/LowerTransportLayer: Segmented Lower transport access PDU: D90028429D0BF607CD 2 of 3
 ...
 V/NetworkLayer: Sequence Number: 00000A
 V/NetworkLayer: Encrypted Network payload: 17EFFF4AD3E23CE2FDF98B1CAAAB2FB29D26E09E0AF6
 V/NetworkLayer: Sequence Number: 00000B
 V/NetworkLayer: Encrypted Network payload: 82FF850037EA33E3B0E908B10FD243C75AB69133F62A
 V/NetworkLayer: Sequence Number: 00000C
 V/NetworkLayer: Encrypted Network payload: 83210847FB2EACEF08029906E3A76A
 V/VendorModelMessageAckedState: Sending acknowledged vendor model message

 /*
  * Message is sent a second time:
  */
 V/MeshTransport: Src address: 0001
 V/MeshTransport: Dst address: 0002
 ...
 V/MeshTransport: Sequence number: 13
 V/MeshTransport: Access message opcode: 3
 V/MeshTransport: Access message parameters: 04110101220202330303440404414243444546474800
 V/AccessLayer: Created Access PDU C3123404110101220202330303440404414243444546474800
 V/UpperTransportLayer: Application nonce: 010000000D0001000200000000
 V/UpperTransportLayer: Encrypted upper transport pdu: 5C17C41C7E50D80BEAB0DC93E45F1B0F9BCA8365028C7110FC92F92460
 V/LowerTransportLayer: Segmented Lower transport access PDU: D90034025C17C41C7E50D80BEAB0DC93 0 of 3
 V/LowerTransportLayer: Segmented Lower transport access PDU: D9003422E45F1B0F9BCA8365028C7110 1 of 3
 V/LowerTransportLayer: Segmented Lower transport access PDU: D9003442FC92F92460 2 of 3
 ...
 V/NetworkLayer: Sequence Number: 00000D
 V/NetworkLayer: Encrypted Network payload: C3CD33738C97E1FD8A5DB555C145012FAFC197BCA2D3
 V/NetworkLayer: Sequence Number: 00000E
 V/NetworkLayer: Encrypted Network payload: E4D0A55B45E85DBD68BAC9DB758E680DE97CBAD1D191
 V/NetworkLayer: Sequence Number: 00000F
 V/NetworkLayer: Encrypted Network payload: 9739202BD93CD1A8FBA57D79A5D1F7
 V/VendorModelMessageAckedState: Sending acknowledged vendor model message
```

Please note that the sequence number is incremented as expected from **0x0a** to **0x0c** and from **0x0d** to **0x0f**.

When **broadcasting** the same message twice in a row, it's getting interesting:

 ```
V/MeshTransport: Src address: 0001
 V/MeshTransport: Dst address: FFFF
 ...
 V/MeshTransport: Sequence number: 10
 V/MeshTransport: Access message opcode: 3
 V/MeshTransport: Access message parameters: 04110101220202330303440404414243444546474800
 V/AccessLayer: Created Access PDU C3123404110101220202330303440404414243444546474800
 V/UpperTransportLayer: Application nonce: 010000000A0001FFFF00000000
 V/UpperTransportLayer: Encrypted upper transport pdu: 000ECE8DFE0500E5030002787274D7D049AB314CEBA274B9B95D127AC4
 V/LowerTransportLayer: Segmented Lower transport access PDU: E9002802000ECE8DFE0500E503000278 0 of 3
 V/LowerTransportLayer: Segmented Lower transport access PDU: E90028227274D7D049AB314CEBA274B9 1 of 3
 V/LowerTransportLayer: Segmented Lower transport access PDU: E9002842B95D127AC4 2 of 3
 ...
 V/NetworkLayer: Sequence Number: 00000A
 V/NetworkLayer: Encrypted Network payload: CC930017BC1E28A454CA2BAE679F8E9BC0F9719DF2C1
 V/NetworkLayer: Sequence Number: 00000B
 V/NetworkLayer: Encrypted Network payload: B7617BFD317347A4AB38E1B1DBE535EDFD01245135FE
 V/NetworkLayer: Sequence Number: 00000C
 V/NetworkLayer: Encrypted Network payload: E960C6C08463A3EFD7C1EB5E25547F
 V/VendorModelMessageAckedState: Sending acknowledged vendor model message
 
  /*
   * Message is sent a second time:
   */
 V/MeshTransport: Src address: 0001
 V/MeshTransport: Dst address: FFFF
 ...
 V/MeshTransport: Sequence number: 11
 V/MeshTransport: Access message opcode: 3
 V/MeshTransport: Access message parameters: 04110101220202330303440404414243444546474800
 V/AccessLayer: Created Access PDU C3123404110101220202330303440404414243444546474800
 V/UpperTransportLayer: Application nonce: 010000000B0001FFFF00000000
 V/UpperTransportLayer: Encrypted upper transport pdu: 7962F8A29FA7928C8B734B8FE7693169231E3CB198965D2540CEB9FA48
 V/LowerTransportLayer: Segmented Lower transport access PDU: E9002C027962F8A29FA7928C8B734B8F 0 of 3
 V/LowerTransportLayer: Segmented Lower transport access PDU: E9002C22E7693169231E3CB198965D25 1 of 3
 V/LowerTransportLayer: Segmented Lower transport access PDU: E9002C4240CEB9FA48 2 of 3
 ...
 V/NetworkLayer: Sequence Number: 00000B
 V/NetworkLayer: Encrypted Network payload: B7617BFD35534CB2844A37BD7825553CC2378A087E65
 V/NetworkLayer: Sequence Number: 00000C
 V/NetworkLayer: Encrypted Network payload: E960C6C08003FDDBF4D20C86AB76D27D0EDD96A6026D
 V/NetworkLayer: Sequence Number: 00000D
 V/NetworkLayer: Encrypted Network payload: 617082EE3F663FD9261BD2268F2671
 V/VendorModelMessageAckedState: Sending acknowledged vendor model message
```

What's interesting here is that the sequence numbers of the first and second call overlap (**0x0a** to **0x0c**, and **0x0b** to **0x0d**), causing the firmware to happily discard any segments whose sequence number has already been used (0x0b and 0x0c of the second call). 

**To Reproduce**
Steps to reproduce the behavior:
1. Create a firmware that contains a vendor model that expects some data. The NCS chat example can easily be modified for this.
2. Send a `VendorModelMessageAcked` message to the firmware using the broadcast (0xffff) address.
   Ensure the message's payload is big enough to be sent in segments.
3. Send the same message a second time.   
4. At the second call, the firmware will log a warnings like:
    `<wrn> bt_mesh_transport: Replay: src 0x0001 dst 0xffff seq 0x00000b`
   Roughly around the same time, the library will log already used sequence numbers similar to the (second) logs above.

**Expected behavior**
Broadcasting a `VendorModelMessageAcked` should properly update the sequence number - i.e. sequence numbers must not be re-used.

**Platform details:**
 - Device: Any.
 - OS: Any (most likely). Android 10 for sure.
 - Library Version: 3.1.6

roshanrajaratnam, 2021-11-30:
Hi are you reproducing this via the sample app? how do you send the same message a second time?

FluffyBunniesTasteTheBest, 2021-11-30:
> are you reproducing this via the sample app?

It happens during automated tests of my own app. 

> how do you send the same message a second time?

Same as the first time: 
1. Waiting for one second to ensure the mesh has processed any previously sent messages. 
2. Instantiating a new `VendorModelMessageAcked` instance.
2. Filling the message with the same data as at the first call.
3. Sending the message by calling: `MeshManagerApi.createMeshPdu(0xffff, myVendorModelMessageAckedIntance);`

Summary:
- This **happens only when broadcasting a vendor-message** (i.e. when sending it to the broadcast address 0xffff). 
- Everything is working as expected when sending the vendor-message directly to a node.
- Even when broadcasting, **both messages are received by the mesh/nodes**, but the firmware will drop the second message because of the already used sequence numbers.
- The above logs indicate that, when broadcasting, `NetworkLayer's` sequence number is incremented only once per **vendor** message, rather than once per sent segment.

roshanrajaratnam, 2021-11-30:
Ok thanks for the detailed info. I'll try to take a look at this as soon as possible.

FluffyBunniesTasteTheBest, 2021-11-30:
Thanks for your help!

> could you test against the sample app as well?

Sure, but the network layer is quite far down the stack, and the library nicely hides it behind a comfortable high level interface - great work btw - so I'd be very surprised if it behaves different at the sample app. Anyway, please give me a few days...

FluffyBunniesTasteTheBest, 2021-11-30:
Unfortunately the same with the sample app... 

Here's a bare minimum firmware application based on NCS's chat example to help reproducing the issue. It just prints the payload at the NRF Terminal: [vendor-message-test.zip](https://github.com/NordicSemiconductor/Android-nRF-Mesh-Library/files/7626888/vendor-message-test.zip)

The java code to send a message to the firmware looks like this:

```
void broadcastSegmentedVendorMessage() {
    final int BROADCAST_ADDRESS = 0xffff;
    final int VENDOR_ID = 0x22ee;
    final int MODEL_ID = 0x22ee0001;
    final int OPCODE = 0x01;
    final String payload = "Hello World! This is some payload that is too long to be send in one segment...\0";

    final ApplicationKey appKey = mViewModel.getNetworkLiveData().getMeshNetwork().getAppKey(0);
    VendorModelMessageAcked message = new VendorModelMessageAcked(appKey, MODEL_ID, VENDOR_ID, OPCODE, payload.getBytes(StandardCharsets.US_ASCII));
    mViewModel.getMeshManagerApi().createMeshPdu(BROADCAST_ADDRESS, message);
}

```
Please note that I've picked an arbitrary vendor id (0x22ee). 

**Now, to reproduce the issue:**

1. Add `broadcastSegmentedVendorMessage()` to `NodeConfigurationActivity.java` and call it in response of a button tap.
2. Set `minSdkVersion` to 19 in build.grade of the example app (Needed for the ASCII-Charset conversion). 
3. Run the attached firmware application on a nrf52832 dev kit.
3. Lauch the example app, provision the firmware, and bind an app key to the vendor model.
4. In the example app, push the button that calls `broadcastSegmentedVendorMessage()`. The first time the button is pushed will cause the firmware to log:
    `I: Hello World! This is some payload that is too long to be send in one segment...`
    Pushing the button a second time will just log:
```
    W: Replay: src 0x0001 dst 0xffff seq 0x00000b
    W: Replay: src 0x0001 dst 0xffff seq 0x00000c
    W: Replay: src 0x0001 dst 0xffff seq 0x00000d
    W: Replay: src 0x0001 dst 0xffff seq 0x00000e
    W: Replay: src 0x0001 dst 0xffff seq 0x00000f
    W: Replay: src 0x0001 dst 0xffff seq 0x000010
    W: Replay: src 0x0001 dst 0xffff seq 0x000011
```

roshanrajaratnam, 2021-11-30:
Thanks again, I should have some time to take a look at this tomorrow!

roshanrajaratnam, 2021-12-07:
Sorry for the delayed reply. I did some digging in to your issue. First I would advise against sending acknowledged messages to a group address as this can cause a lot of traffic in the mesh network. In your case you are sending it to all nodes and you will get a status back from all nodes in your network. However it's up to you.

Currently I am testing against the NCS vendor model sample and I don't see an issue relating to incrementing the sequence number. I am using the sample app with the address hardcoded to `0xFFFF` and pressing the send button. I extracted some logs from Android studio console that shows the incremented sequence number.
```
2021-12-07 12:37:39.787 13306-13306/no.nordicsemi.android.nrfmeshprovisioner V/MeshTransport: Src address: 0001
2021-12-07 12:37:39.787 13306-13306/no.nordicsemi.android.nrfmeshprovisioner V/MeshTransport: Dst address: FFFF
2021-12-07 12:37:39.787 13306-13306/no.nordicsemi.android.nrfmeshprovisioner V/MeshTransport: Key: 1C5549DE75CB8425A4CA1E7AAD1188E9
2021-12-07 12:37:39.787 13306-13306/no.nordicsemi.android.nrfmeshprovisioner V/MeshTransport: akf: 1
2021-12-07 12:37:39.787 13306-13306/no.nordicsemi.android.nrfmeshprovisioner V/MeshTransport: aid: 59
2021-12-07 12:37:39.787 13306-13306/no.nordicsemi.android.nrfmeshprovisioner V/MeshTransport: aszmic: 0
2021-12-07 12:37:39.787 13306-13306/no.nordicsemi.android.nrfmeshprovisioner V/MeshTransport: Sequence number: 24
2021-12-07 12:37:39.787 13306-13306/no.nordicsemi.android.nrfmeshprovisioner V/MeshTransport: Access message opcode: b
2021-12-07 12:37:39.787 13306-13306/no.nordicsemi.android.nrfmeshprovisioner V/MeshTransport: Access message parameters: 00
2021-12-07 12:37:39.787 13306-13306/no.nordicsemi.android.nrfmeshprovisioner V/AccessLayer: Created Access PDU CB590000
```

After pressing the button a second time.
```
2021-12-07 12:37:41.232 13306-13306/no.nordicsemi.android.nrfmeshprovisioner V/MeshTransport: Src address: 0001
2021-12-07 12:37:41.232 13306-13306/no.nordicsemi.android.nrfmeshprovisioner V/MeshTransport: Dst address: FFFF
2021-12-07 12:37:41.232 13306-13306/no.nordicsemi.android.nrfmeshprovisioner V/MeshTransport: Key: 1C5549DE75CB8425A4CA1E7AAD1188E9
2021-12-07 12:37:41.232 13306-13306/no.nordicsemi.android.nrfmeshprovisioner V/MeshTransport: akf: 1
2021-12-07 12:37:41.232 13306-13306/no.nordicsemi.android.nrfmeshprovisioner V/MeshTransport: aid: 59
2021-12-07 12:37:41.232 13306-13306/no.nordicsemi.android.nrfmeshprovisioner V/MeshTransport: aszmic: 0
2021-12-07 12:37:41.232 13306-13306/no.nordicsemi.android.nrfmeshprovisioner V/MeshTransport: Sequence number: 25
2021-12-07 12:37:41.233 13306-13306/no.nordicsemi.android.nrfmeshprovisioner V/MeshTransport: Access message opcode: b
2021-12-07 12:37:41.233 13306-13306/no.nordicsemi.android.nrfmeshprovisioner V/MeshTransport: Access message parameters: 00
2021-12-07 12:37:41.233 13306-13306/no.nordicsemi.android.nrfmeshprovisioner V/AccessLayer: Created Access PDU CB590000
```

FluffyBunniesTasteTheBest, 2021-12-07:
It seems like you're trying to reproduce with a payload that fits into one segment, which won't trigger the issue. 

Please try again with a bigger payload - the [java snippet from above](https://github.com/NordicSemiconductor/Android-nRF-Mesh-Library/issues/458#issuecomment-982752740) is copy&past-ready. So is the provided bare-minimum-example firmware. Should be easy to reproduce with it.

roshanrajaratnam, 2021-12-07:
Can confirm it's reproduceable. Will dig in to this.

roshanrajaratnam, 2021-12-07:
So I am also testing against the NCS chat example the only difference are the model id and opcodes which should not matter and I don't see this issue happening via the sample app.

NanaDada, 2021-12-07:
Interesting...

Can you please provide the logged messages from `MeshTransport`, `AccessLayer`, `UpperTransportLayer`, `LowerTransportLayer` and `NetworkLayer`? Basically the same log messages as in this threads first message.

Please note that you must send the segmented payload twice - the first call works as intended, subsequent calls trigger the issue.

FluffyBunniesTasteTheBest, 2021-12-07:
Interesting...

Can you please provide the logged messages from `MeshTransport`, `AccessLayer`, `UpperTransportLayer`, `LowerTransportLayer` and `NetworkLayer`? Basically the same log messages as in this threads first message.

Please note that you must send the segmented payload twice - the first call works as intended, subsequent calls trigger the issue.

roshanrajaratnam, 2021-12-07:
Hey I see you are referring to the network layer sequence number. This seems like an issue, let me check on this again.

roshanrajaratnam, 2021-12-08:
@FluffyBunniesTasteTheBest Could you give this a test and confirm if everything looks fine on your end?

FluffyBunniesTasteTheBest, 2021-12-10:
@roshanrajaratnam Thank you for the fast fix. I'll have a look at it in the next few days.

roshanrajaratnam, 2021-12-10:
I have already merged this to develop so you can check it there!

FluffyBunniesTasteTheBest, 2021-12-15:
Works well. Thank you for the fast fix!

PR #477 (2022-02-16): Release v3.1.7
This release includes the following changes.
* Fixes #432
* Fixes #438
* Fixes #441 
* #448 Adds support for Scheduler, Generic Transition time, PowerUpBehaviour, GenericProperties (Admin, User, Manufacturer) and Time Messages
* #456 maintains consistency.
* Fixes #457 
* Fixes #458, where the provisioner's sequence number seems to get reset when sending a message to 0xFF
* Fixes #461 and #462 that was causing the app/library to crash when de-obfuscating a received message.
* #464 adds support for Android 12 `BLUETOOTH_SCAN`, `BLUETOOTH_CONNECT` permissions and splash screen. Th [... cut]
closed 2022-02-16 by roshanrajaratnam