nordicsemi/Android-nRF-Mesh-Library #461: Bug in parseMeshPduNotifications

Rawa, 2021-12-17:
---

**Describe the bug**
parseMeshPduNotifications get stuck in loop if node is not in provisioner's config

**To Reproduce**
Setup a mesh,
Delete node X from mesh
Connect to node X as Proxy (whom is not in config anymore)
Configure it as Proxy

Device will get stuck in loop of parsing message.

**Expected behavior**
Unsure of expected behavior, since node is not in config, it should probably not be returned but method should anyhow not get stuck.

**Platform details:**
 - Device: Pixel 5
 - OS: Android 12L
 - Library Version (or Commit) 3.1.6

**Logs / Screenshots**
```
12-16 14:17:51.427 24143 24253 V BaseMeshMessageHandler: TTL for received message: 13
12-16 14:17:51.427 24143 24253 V BaseMeshMessageHandler: TTL for received message: 63
12-16 14:17:51.427 24143 24253 V BaseMeshMessageHandler: TTL for received message: 46
12-16 14:17:51.427 24143 24253 V BaseMeshMessageHandler: TTL for received message: 6
12-16 14:17:51.427 24143 24253 V BaseMeshMessageHandler: TTL for received message: 125
12-16 14:17:51.427 24143 24253 V BaseMeshMessageHandler: TTL for received message: 43
... about 40000 lines of same print
12-16 14:18:27.592 24330 24377 V BaseMeshMessageHandler: TTL for received message: 44
12-16 14:18:27.592 24330 24377 V BaseMeshMessageHandler: TTL for received message: 10
12-16 14:18:27.593 24330 24377 V BaseMeshMessageHandler: TTL for received message: 50
12-16 14:18:27.593 24330 24377 V BaseMeshMessageHandler: Sequence number of received access message: 3756400

```

roshanrajaratnam, 2021-12-17:
@Rawa Thanks for reporting this issue. Do you delete from the list of nodes in the app?  what do you mean by Configure it as proxy  after connecting to it?

Rawa, 2021-12-17:
@roshanrajaratnam Thanks for the quick reply.

I did not use the Nrf Mesh app but the library directly, the node was deleted without unprovisioning which means it still has the configuration of network and app key, then the connection was made as a proxy to that very same deleted node since it was still advertising with network ID etc. 

By configuring the proxy I meant setting up a inclusion filter on it and add an address that I listen to. Upon receiving the status message the library won't find the node and thus just immediately continue in this loop.

Regarding the loop:
Currently the loop will always be true `ivIndex <= ivIndex + 1` which essentially means we are bruteforcing IVIndex and not just looking at IVIndex and IVIndex-1? E.g if we are unable to find https://github.com/NordicSemiconductor/Android-nRF-Mesh-Library/blob/e03b1eba6cf4d7bb41b1fa36d3ce8055e8df4208/mesh/src/main/java/no/nordicsemi/android/mesh/transport/BaseMeshMessageHandler.java#L117 we will also continue to loop over the IVIndex?

I'm not sure what the correct behavior should be for a provisioner when it is unable to find the node in the provisioner config.

roshanrajaratnam, 2021-12-17:
Ok let me test with that setup. Also you do have a point about reconnecting to deleted nodes with the same configuration. The deletion here is allows to get rid of a node that may have failed to operate.

> Regarding the loop:
Currently the loop will always be true ivIndex <= ivIndex + 1 which essentially means we are bruteforcing IVIndex and not just looking at IVIndex and IVIndex-1? E.g if we are unable to find getMatchingK2Output we will also continue to loop over the IVIndex?

You can see we consider both IV Index and also IV Index - 1
https://github.com/NordicSemiconductor/Android-nRF-Mesh-Library/blob/e03b1eba6cf4d7bb41b1fa36d3ce8055e8df4208/mesh/src/main/java/no/nordicsemi/android/mesh/transport/BaseMeshMessageHandler.java#L105

> I'm not sure what the correct behavior should be for a provisioner when it is unable to find the node in the provisioner config. But this looping doesn't make much sense when it is unable to find the node at all? It should probably at least break the while loop?

Ideally it should reach here https://github.com/NordicSemiconductor/Android-nRF-Mesh-Library/blob/e03b1eba6cf4d7bb41b1fa36d3ce8055e8df4208/mesh/src/main/java/no/nordicsemi/android/mesh/transport/BaseMeshMessageHandler.java#L162

Let me give this a test.

roshanrajaratnam, 2021-12-17:
So this is my setup

- I have a switch and a light server. 
- I configured the switch to turn on and off the light server. 
- Then connected to the switch, configured the inclusion filter with light server address
- Deleted the switch from the list of nodes (Note the sample app is still connected to the switch)
- Turn on and off from the switch

I don't see the issue with looping here though...

R0m4in-dooz, 2021-12-17:
Hello there !

We also face this issue from time to time (because of bad n/w handling). If a provisioned node is ever deleted from database but is not really unprovisioned this "infinite loop" is indeed hapenning and may stuck the host device (phone, tablet,...).

We face it for instance if we delete a LPN from the network, but the friend didn't forward the ConfigNodeReset for some reason. If this LPN is publishing in the network, then this library will print hundreds (if not thousands) logs `TTL for received message` and eventually will print a really high seq num.

R0m4in-dooz, 2021-12-17:
not sure if it is high priority bug though, because it's due to desync of the database against reality, usually due to some bad handling from the library consumer...

roshanrajaratnam, 2021-12-17:
so do you mean delete or a reset of the node?

R0m4in-dooz, 2021-12-17:
for our case, it only happens when we receive a msg from unknown node. We face it if we *delete* the node without having a ConfigNodeResetStatus using the good API to send a node reset

roshanrajaratnam, 2021-12-17:
OK let me dig in to this more because this is something I've not seen before.

R0m4in-dooz, 2021-12-17:
there is also another bug related to this. It **crashes** the app if the library tries to send a block ack to an unknown node, due to calling incrementSeqNum to null node (`NullPointerException`).

Again, for me it's not that high priority (but I'll let you define them 🤣) as it is usually due to bad database handling from the library consumer

roshanrajaratnam, 2021-12-17:
@R0m4in-dooz  may be you should create a new issue?

R0m4in-dooz, 2021-12-17:
#462 created, I think related to this issue because I faced both issues while an unknown was publishing to the current connected network

roshanrajaratnam, 2021-12-17:
@Rawa @R0m4in-dooz you succeeded proving me wrong!

R0m4in-dooz, 2021-12-17:
lol not sure if it is a great achievement 😅 Have fun fixing this !

roshanrajaratnam, 2021-12-17:
@Rawa could you confirm if this works for you?

Rawa, 2021-12-20:
Just read through your commit, this seems more correct, but I also think there should be a print if the node was not found because this mean you have a configuration that is not in sync. 

Another caveat is that this message will not be processed fully by the library and fail silently. E.g in my case I try to setup a proxy filter (e.g set exclusion filter), I will not receive a message back after this message have been processed by the library. The MeshManagerApi would log
```
V/MeshManagerApi: Received proxy configuration message: 0x025878B62D4F2A838ACBABD067B6A32286872A75DB65
```
but none of the `MeshStatusCallbacks` would be notified. The user of the library can only assume it failed after the 30 seconds timeout specified in the mesh profile specification. 

I'm not sure what Mesh Profile states about these types of messages. The node itself is setup with the correct network keys/application keys it, and it would function correctly in a mesh, but it is not in the Provisioner Config so it should probably not been connected to in the first place since it was not disregarded correctly after removal from the provision config. But at the same time with the usecase of multiple provisioners modifiying the mesh, it might happen anyway?

roshanrajaratnam, 2021-12-20:
> Just read through your commit and also tested it, this seems more correct, but I also think there should be a print if the node was not found because this mean you have a configuration that is not in sync.

Great that it works, I will look in to adding some kind of log.

> An important caveat is that this message will not be processed fully by the library and fail silently. E.g in my case I try to setup a proxy filter (e.g set exclusion filter), I will not receive a message back after this message have been processed by the library. The MeshManagerApi would log
> `V/MeshManagerApi: Received proxy configuration message: 0x025878B62D4F2A838ACBABD067B6A32286872A75DB65`
> but none of the MeshStatusCallbacks would be notified. The user of the library can only assume it failed after the 30 seconds timeout specified in the mesh profile specification.

As of there is no way of notifying the user of any messages sent from an unknown node, but this can be done by simply notifying the user by adding a callback.  I will look in to this.

> I'm not sure what Mesh Profile states about these types of messages. The node itself is setup with the correct network keys/application keys it, and it would function correctly in a mesh, but it is [...]

As you may know, the correct approach to delete a node from the network is by sending the `ConfigNodeReset`. Upon receiving the status `ConfigNodeReset`, the node is deleted from the list of nodes and it's address is added to the exclusion list. However simply deleting a node will not delete any of it's network configuration. All it does is removing from the network i.e. list of nodes and adding it's address to the exclusion list. The idea of adding this delete API is to allow the user to delete a node from the network that may have failed due to a hardware malfunction etc. Therefore you will always be able to connect to it if the node came back to life. There is nothing much we can do about it. There's always a possibility of multiple provisioners modifying a network but that's more like an application specific feature that a user has the freedom to define and implement.

Rawa, 2021-12-20:
> Great that it works, I will look in to adding some kind of log.

👍 

> 
> As of now there is no way of notifying the user of any messages sent from an unknown node, but this can be done by simply notifying the user by adding a callback. I will look in to this.

Yes, something like this seems like it would suitable.

> As you may know, the correct approach to delete a node from the network is by sending the `ConfigNodeReset`. Upon receiving the status `ConfigNodeReset`, the node is deleted from the list of nodes [...]

👍  Yes, this I'm aware. I agree with you on the behavior, the result of this is would be because of improper disposing of the device after the removal. Technically the mesh can also perform the key refresh procedure.

Itxupengfei, 2021-12-30:
Hello, I also have an error in this "parseMeshPduNotifications "function, but it is not necessary. The ANR is more likely to occur when the status of the device is frequently reported

12-29 14:42:50.028 15924-22277/? E/DEBUG: --- --- --- --- --- --- --- --- --- --- --- --- --- --- --- ---
2021-12-29 14:42:50.028 15924-22277/? E/DEBUG: Process Name: 'com.hle.lhzm'
2021-12-29 14:42:50.028 15924-22277/? E/DEBUG: Thread Name: 'com.hle.lhzm'
2021-12-29 14:42:50.029 15924-22277/? E/DEBUG: ANR Time: 2021-12-29 14:42:50.026
2021-12-29 14:42:50.029 15924-22277/? E/DEBUG: ANR Info:
2021-12-29 14:42:50.029 15924-22277/? E/DEBUG: --- --- --- --- --- --- --- --- --- --- --- --- --- --- --- ---
2021-12-29 14:42:50.029 15924-22277/? E/DEBUG: anr traces:
2021-12-29 14:42:50.030 15924-22277/? E/DEBUG: generate begin time: 2021-12-29 14:42:49
    [DEBUG] dump art internal: 148
    [DEBUG] VMExt: 0x7017ca8380, i: 62, str: 0
    [DEBUG] aborting: 0x0, 0
    [DEBUG] Dump: 0x0, State: 0x7017317acc, JavaStack: 0x701731bcc4
    [DEBUG] current: 0x7017c10800, pid: 22264
    [DEBUG] List: 0x7017d27000
    [DEBUG] Each: 0x7017331c0c
    [DEBUG] err: 0x6ffd4fb990
    [DEBUG] begin each
    [DEBUG] dumping 0x7017c10800 ...
    "main" prio=5 tid=1 Runnable
      | group="main" sCount=0 dsCount=0 flags=0 obj=0x723dd6a8 self=0x7017c10800
      | sysTid=15924 nice=-10 cgrp=default sched=0/0 handle=0x709eb71ed0
      | state=? schedstat=( 0 0 0 ) utm=0 stm=0 core=0 HZ=100
      | stack=0x7fc44ac000-0x7fc44ae000 stackSize=8192KB
      | held mutexes= "mutator lock"(shared held)
        at no.nordicsemi.android.mesh.transport.BaseMeshMessageHandler.getMatchingK2Output(BaseMeshMessageHandler.java:176)
        at no.nordicsemi.android.mesh.transport.BaseMeshMessageHandler.parseMeshPduNotifications(BaseMeshMessageHandler.java:115)
        at no.nordicsemi.android.mesh.MeshMessageHandler.parseMeshPduNotifications(MeshMessageHandler.java:61)
        at no.nordicsemi.android.mesh.MeshManagerApi.parseNotifications(MeshManagerApi.java:282)
        at no.nordicsemi.android.mesh.MeshManagerApi.handleNotifications(MeshManagerApi.java:254)
        at com.lhzm.sigmeshlib.provisioned.MeshRespository$1.onDataReceived(MeshRespository.java:198)
        at com.lhzm.sigmeshlib.ble.BleMeshManager$BleMeshGattCallbacks.lambda$initialize$0$BleMeshManager$BleMeshGattCallbacks(BleMeshManager.java:122)
        at com.lhzm.sigmeshlib.ble.-$$Lambda$BleMeshManager$BleMeshGattCallbacks$z4YJsLtwd3--9r5reGqqi8yjLqA.onDataReceived(lambda:-1)
        at no.nordicsemi.android.ble.ValueChangedCallback.lambda$notifyValueChanged$0(ValueChangedCallback.java:147)
        at no.nordicsemi.android.ble.-$$Lambda$ValueChangedCallback$4rcq0bbOhcTVqpLmftoy8pUEJ-Q.run(lambda:-1)
        at android.os.Handler.handleCallback(Handler.java:883)
        at android.os.Handler.dispatchMessage(Handler.java:100)
        at android.os.Looper.loop(Looper.java:230)
        at android.app.ActivityThread.main(ActivityThread.java:7937)
        at java.lang.reflect.Method.invoke(Native method)
        at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:526)
        at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:1034)
    
    "Jit thread pool worker thread 0" daemon prio=5 tid=2 Native
      | group="main" sCount=0 dsCount=0 flags=0 obj=0x13cc02e0 self=0x700b00e000
      | sysTid=15933 nice=0 cgrp=default sched=0/0 handle=0x700c54bd40
      | state=? schedstat=( 0 0 0 ) utm=0 stm=0 core=0 HZ=100
      | stack=0x700c44d000-0x700c44f000 stackSize=1023KB
      | held mutexes=
      (no managed stack frames)
    
    "Signal Catcher" daemon prio=5 tid=6 WaitingInMainSignalCatcherLoop
      | group="system" sCount=0 dsCount=0 flags=0 obj=0x13cc0358 self=0x700b048000
      | sysTid=15938 nice=0 cgrp=default sched=0/0 handle=0x700b446d50
      | state=? schedstat=( 0 0 0 ) utm=0 stm=0 core=0 HZ=100
      | stack=0x700b350000-0x700b352000 stackSize=991KB
      | held mutexes=
      (no managed stack frames)
    
    "ADB-JDWP Connection Control Thread" daemon prio=0 tid=8 WaitingInMainDebuggerLoop
      | group="system" sCount=0 dsCount=0 flags=0 obj=0x13cc03d0 self=0x700b063800
      | sysTid=15939 nice=0 cgrp=default sched=0/0 handle=0x700b349d50
      | state=? schedstat=( 0 0 0 ) utm=0 stm=0 core=0 HZ=100
      | stack=0x700b253000-0x700b255000 stackSize=991KB
      | held mutexes=
      (no managed stack frames)
    
    "HeapTaskDaemon" daemon prio=5 tid=9 WaitingForTaskProcessor
      | group="system" sCount=0 dsCount=0 flags=0 obj=0x13ccb118 self=0x6fffea0800
      | sysTid=15940 nice=4 cgrp=default sched=0/0 handle=0x6fb7796d50
      | state=? schedstat=( 0 0 0 ) utm=0 stm=0 core=0 HZ=100
      | stack=0x6fb7694000-0x6fb7696000 stackSize=1039KB
      | held mutexes=
        at dalvik.system.VMRuntime.runHeapTasks(Native method)
        at java.lang.Daemons$HeapTaskDaemon.runInternal(Daemons.java:557)
        at java.lang.Daemons$Daemon.run(Daemons.java:140)
        at java.lang.Thread.run(Thread.java:919)
    
    "ReferenceQueueDaemon" daemon prio=5 tid=10 Waiting
      | group="system" sCount=0 dsCount=0 flags=0 obj=0x13cc0448 self=0x6fffea
2021-12-29 14:42:50.030 15924-22277/? E/DEBUG: --- --- --- --- --- --- --- --- --- --- --- --- --- --- --- ---
2021-12-29 14:42:50.030 15924-22277/? E/DEBUG: native stack:
2021-12-29 14:42:50.067 15924-22277/? E/DEBUG: Thread Name: 'com.hle.lhzm'
[... cut ...]
2021-12-29 14:42:50.220 15924-22277/? E/DEBUG:   ##00 pc 0000000000012000  /apex/com.android.runtime/lib64/libart.so (art_quick_imt_conflict_trampoline+32)
2021-12-29 14:42:50.220 15924-22277/? E/DEBUG:   ##01 pc 0000000000142654  /memfd:/jit-cache (deleted)
2021-12-29 14:42:50.224 15924-22277/? E/DEBUG: --- --- --- --- --- --- --- --- --- --- --- --- --- --- --- ---
2021-12-29 14:42:50.224 15924-22277/? E/DEBUG: meminfo:
2021-12-29 14:42:50.224 15924-22277/? E/DEBUG: MemTotal:        5750448 kB

roshanrajaratnam, 2021-12-30:
This issue have been fixed in dev branch. Could you check?

Rawa, 2022-01-26:
Yes, I tested as mentioned before. This seems to have resolved it.

PR #477 (2022-02-16): Release v3.1.7
This release includes the following changes.
* Fixes #432
* Fixes #438
* Fixes #441 
* #448 Adds support for Scheduler, Generic Transition time, PowerUpBehaviour, GenericProperties (Admin, User, Manufacturer) and Time Messages
* #456 maintains consistency.
* Fixes #457 
* Fixes #458, where the provisioner's sequence number seems to get reset when sending a message to 0xFF
* Fixes #461 and #462 that was causing the app/library to crash when de-obfuscating a received message.
* #464 adds support for Android 12 `BLUETOOTH_SCAN`, `BLUETOOTH_CONNECT` permissions and splash screen. Th [... cut]
closed 2022-02-16 by roshanrajaratnam