nordicsemi/Android-nRF-Mesh-Library #462: Crash app in sendBlockAck

R0m4in-dooz, 2021-12-17:
---

**Describe the bug**
When the library receives a segmented message that needs a block ack **from an unknown node**, the app will crash due to a `NullPointerException`.

**To Reproduce**
Steps to reproduce the behavior:
1. provision two nodes to the network
2. manually delete one node from the database without sending a `ConfigNodeReset`
3. connect to the second node of this network
4. make the first node to send a message that makes the library to trigger block ack answer

**Expected behavior**
Probably log some warning that the node is unknown instead of trying to send a block ack ?

**Platform details:**
 - Device: Any
 - OS: Android 8.0
 - Library Version 3.1.6

**Logs / Screenshots**
[image]

roshanrajaratnam, 2021-12-17:
Do you use the proxy filter in this case to receive the message on the phone that is send to other nodes?

R0m4in-dooz, 2021-12-17:
No, we still have to integrate this I think...at least I never worked on this feature, and I know we should (embedded team request)

R0m4in-dooz, 2021-12-17:
Sooo, sorry but the answer is maybe ? 😅 consider we don't

roshanrajaratnam, 2021-12-17:
so you basically delete the node, connect to and send some message that's segmented? I'll test this.

R0m4in-dooz, 2021-12-17:
> connect to and
connect to **another** one, yes

R0m4in-dooz, 2021-12-17:
it's not the phone that initiate the msg actually, it's the unkown node

R0m4in-dooz, 2021-12-17:
hmmm 🤔 

Well you can also simulate it with two phones I think :
- create one network with a phone, provision two nodes with it
- load the network with another phone, assign a new provisioner (the first phone won't know about this)
- send a segmented msg with second phone to (any?) node in the network **while the first phone is still connected**
- the first phone should trigger `sendBlockAck` but the ProvisionedNode (second provisioner) is unknown

roshanrajaratnam, 2021-12-17:
ok that should help me to produce this. What's the segmented message you are sending here?

R0m4in-dooz, 2021-12-17:
😅 I think it was proprietary message. So I never tested with other ones

roshanrajaratnam, 2021-12-17:
By just looking at the code I don't see how this could happen because the message is not processed any further if the node does not exist. https://github.com/NordicSemiconductor/Android-nRF-Mesh-Library/blob/e03b1eba6cf4d7bb41b1fa36d3ce8055e8df4208/mesh/src/main/java/no/nordicsemi/android/mesh/transport/BaseMeshMessageHandler.java#L127

R0m4in-dooz, 2021-12-17:
Well, yeh I would expect the library to do such check indeed... The thing is, it happens ahah.
To be more precise on our case, it was happening because we let an external BLE device to be a provisioner without declaring it on the phone (prototyping where phone was not compatible with this new device). This external device was sending a proprietary segmented message (extending `ApplicationMessage`) to a known node in order to read some config. As soon as we declared the provisioner and pass it to the external device, the library was not failing anymore.

R0m4in-dooz, 2021-12-17:
it can also happen if a provisioner change unicast without the phone knowing about it (shared network)

R0m4in-dooz, 2021-12-17:
> proprietary message

If it can help you, [here](https://github.com/OZEO-DOOZ/Android-nRF-Mesh-Library-1/blob/master/mesh/src/main/java/no/nordicsemi/android/mesh/transport/MagicLevelGet.java) is the implem for our fork of this library

roshanrajaratnam, 2022-02-14:
@R0m4in-dooz does the fix made in #461 fixed this one?

R0m4in-dooz, 2022-02-14:
Hello @roshanrajaratnam ! Unfortunately I'm not able to test it as of now..but as I explained earlier in this ticket, it's pretty easy to reproduce as long as you have two phones and two nodes. I may have time to test next week tho 👍

roshanrajaratnam, 2022-02-15:
I gave this a test now and I am not able to reproduce this, I used the mesh chat example from the nRF Connect SDK where you can send a long message to a specific node. The fix from #461 have solved the issue in my humble opinion.

closed 2022-02-16 by roshanrajaratnam