nordicsemi/Android-nRF-Mesh-Library #468: networkIdMatches method with only one param of class MeshManagerApi is missing in 2.4.3

ZubairAkber, 2022-01-12:
---

**Describe the bug**
I just pull the latest code and build it successfully, but when i create my own project and included 'no.nordicsemi.android:mesh:2.4.3' as library and try to call networkIdMatches method, i found that the overloaded method is missing with one argument that is present in the project.

**To Reproduce**
Simply add 'no.nordicsemi.android:mesh:2.4.3' in your project and try to call networkIdMatches method of MeshManagerApi class with one argument only

**Expected behavior**
The overloaded method should be available

roshanrajaratnam, 2022-01-12:
Hi is there a reason why you are not using 3.1.6? The sample app and the mesh library should be the same version.

ZubairAkber, 2022-01-12:
Hi, i am just trying to build my own wrapper class that will only do the provisioning part, so for that i just include the library

roshanrajaratnam, 2022-01-12:
I don't see an issue with version 2.4.3  https://github.com/NordicSemiconductor/Android-nRF-Mesh-Library/blob/1677257b1b23661b7e85b22da2c7e7cbdedce25b/mesh/src/main/java/no/nordicsemi/android/mesh/MeshManagerApi.java#L757

https://github.com/NordicSemiconductor/Android-nRF-Mesh-Library/blob/1677257b1b23661b7e85b22da2c7e7cbdedce25b/app/src/main/java/no/nordicsemi/android/nrfmesh/viewmodels/ScannerRepository.java#L86

Also I would upgrade using the latest 3.1.6 version as it adds support to the official Mesh Configuration Database as well.

closed 2022-01-12 by roshanrajaratnam

ZubairAkber, 2022-01-12:
@Override
    public boolean networkIdMatches(@Nullable final byte[] serviceData) {
        final byte[] advertisedNetworkId = getAdvertisedNetworkId(serviceData);
        if (advertisedNetworkId != null) {
            for (NetworkKey key : mMeshNetwork.getNetKeys()) {
                if (Arrays.equals(key.getNetworkId(), advertisedNetworkId) || Arrays.equals(key.getOldNetworkId(), advertisedNetworkId))
                    return true;
            }
        }
        return false;
    }

roshanrajaratnam, 2022-01-12:
Hi, if you check the source directory it is in the mesh project and has been since version 1.0. Check the following line, it's called from the app and the method is in the library.

https://github.com/NordicSemiconductor/Android-nRF-Mesh-Library/blob/1677257b1b23661b7e85b22da2c7e7cbdedce25b/app/src/main/java/no/nordicsemi/android/nrfmesh/viewmodels/ScannerRepository.java#L86