nordicsemi/Android-nRF-Mesh-Library #469: Add group is broken if provisioner have more than 1 group range

Rawa, 2022-01-31:
---

**Describe the bug**
If a provisioner has more than 1 groups range the add Group function will always fail.

**To Reproduce**
Steps to reproduce the behavior:
In mesh app:
1. Go to provisioner
2. Add new group range, e.g E000-EFFF
3. Try and add a new group to any of the ranges, it will always fail

**Expected behavior**
Group should work in either range.

Problem relies here, if the group is outside any of the ranges create group will fail.

https://github.com/NordicSemiconductor/Android-nRF-Mesh-Library/blob/e03b1eba6cf4d7bb41b1fa36d3ce8055e8df4208/mesh/src/main/java/no/nordicsemi/android/mesh/MeshNetwork.java#L434

Rawa, 2022-01-31:
Same goes for the other function as well.
https://github.com/NordicSemiconductor/Android-nRF-Mesh-Library/blob/e03b1eba6cf4d7bb41b1fa36d3ce8055e8df4208/mesh/src/main/java/no/nordicsemi/android/mesh/MeshNetwork.java#L454

Rawa, 2022-01-31:
Added a pull request for a fix:
https://github.com/NordicSemiconductor/Android-nRF-Mesh-Library/pull/470

philips77, 2022-01-31:
Good catch. Thank you for creating the issue and PR. We will look into it shortly.

philips77, 2022-01-31:
I managed to replicate the issue using the most recent version.

PR #477 (2022-02-16): Release v3.1.7
This release includes the following changes.
* Fixes #432
* Fixes #438
* Fixes #441 
* #448 Adds support for Scheduler, Generic Transition time, PowerUpBehaviour, GenericProperties (Admin, User, Manufacturer) and Time Messages
* #456 maintains consistency.
* Fixes #457 
* Fixes #458, where the provisioner's sequence number seems to get reset when sending a message to 0xFF
* Fixes #461 and #462 that was causing the app/library to crash when de-obfuscating a received message.
* #464 adds support for Android 12 `BLUETOOTH_SCAN`, `BLUETOOTH_CONNECT` permissions and splash screen. Th [... cut]
closed 2022-02-16 by roshanrajaratnam