nordicsemi/Android-nRF-Mesh-Library #473: Build error when running dev branch tests

gujor, 2022-02-09:
**Describe the bug**
Build error when running dev branch tests.

**To Reproduce**
Steps to reproduce the behavior:
1. Switch to dev branch.
2. Run tests.

**Expected behavior**
Tests should run.

**Platform details:**
 - Android Studio
 - dev branch

**Logs / Screenshots**
[image]

roshanrajaratnam, 2022-02-09:
Hi thanks for reporting this. Will fix this.

gujor, 2022-02-09:
Thanks. I have created a PR with the fixed we used. If it looks ok, feel free to use it, and maybe save some time :)

roshanrajaratnam, 2022-02-09:
There were some tests that was added in a pull request we got with the scheduler feature that may have had this issue. Thanks for your PR.

commit cfc3ba9f (2022-02-09): Fix #473 test build problem
PR #477 (2022-02-16): Release v3.1.7
This release includes the following changes.
* Fixes #432
* Fixes #438
* Fixes #441 
* #448 Adds support for Scheduler, Generic Transition time, PowerUpBehaviour, GenericProperties (Admin, User, Manufacturer) and Time Messages
* #456 maintains consistency.
* Fixes #457 
* Fixes #458, where the provisioner's sequence number seems to get reset when sending a message to 0xFF
* Fixes #461 and #462 that was causing the app/library to crash when de-obfuscating a received message.
* #464 adds support for Android 12 `BLUETOOTH_SCAN`, `BLUETOOTH_CONNECT` permissions and splash screen. Th [... cut]
closed 2022-02-09 by roshanrajaratnam