nordicsemi/Android-nRF-Mesh-Library #479: It seems like unicast addresses are assigned multiple times.

FluffyBunniesTasteTheBest, 2022-02-22:
---

**Describe the bug**
When provisioning a node it can happen that a previously assigned unicast address is reassigned.

**To Reproduce**
1. Use a fresh installation of the library's demo app to provision a node. It will get the unicast address **0x0002** assigned.
2. Erase the node using the programmer from nrf toolbox.
3. Use Segger Embedded Studio to reprogram the firmware. While technically the node is now unprovisioned, the app/library does not know about.
4. Provision the node again and the unicast address **0x0003** will be assigned - as expected.
5. Erase the node again using the programmer from nrf toolbox.
6. And use Segger Embedded Studio to reprogram the firmware again.
7. Provisioning the node for the third time and it gets the unicast address **0x0002** assigned again.

**Expected behavior**
Because the node with address **0x0002** was not unprovisioned through the app, the app/library can't know the node is gone for good, and hence the unicast addresses should not be reused. 
For the case that this is the expected behavior, step 4. would also be expected to reuse address **0x0002**. So there seems to be something off.

**Some background information**
It happened a few times that customers sent back their devices because they didn't work: The customers added them to their mesh network, but the nodes didn't do anything. When taking a closer look at those nodes it turns out that they are perfectly fine, which lead to the conclusion that there must be something wrong with the customer's mesh network. 

Now my hypothesis is that (maybe) the troublesome nodes suffer an address conflict - which would perfectly explain the error - and which lead to this report. But this is mainly speculation, so any clarifications are welcome. Thank you.

**Platform details:**
 - Device: Any Android device.
 - OS: Android 10, but will most likely happen on any Android version.
 - Library Version: 3.1.6 and 3.1.7

roshanrajaratnam, 2022-02-23:
@FluffyBunniesTasteTheBest so each node has a unique UUID and if the UUID is the same for each provisioned device you will be overwriting the same node. When address 0x0003 gets assigned the same node is in the network with 0x0002. However provisioning will replace the existing node with address 0x0003. When re-provisioning the same node it will check what addresses are in use and will find that 0x0002 is not in use anymore and 0x0002 will be reused. 

Addresses are not unique in a mesh network, the reason being any deleted node's address gets added to an exclusion list and once a Key Refresh Procedure has taken place the addresses in the exclusion list are free to be re-used.

Basically you have to make sure each node has a unique UUID.

FluffyBunniesTasteTheBest, 2022-02-24:
@roshanrajaratnam Thank you very much for the detailed explanation!

> Addresses are not unique in a mesh network, the reason being any deleted node's address gets added to an exclusion list and once a Key Refresh Procedure has taken place the addresses in the exclusion list are free to be re-used.

Under what circumstances does a Key Refresh Procedure happen? Does the library do it in background? My app doesn't handle it (at least not knowingly), so the previously used mesh addresses _should_ remain in the exclusion list, I believe.

Also thanks for the links to the nRF Connect SDK. Unfortunately I had to revert back to nRF5 SDK for Mesh because flash memory run out...

roshanrajaratnam, 2022-02-24:
@FluffyBunniesTasteTheBest Let me rephrase myself and I have corrected my previous comment. Addresses are not unique in a mesh network, the reason being any deleted node's address gets added to an exclusion list and once the IV Index Update procedure is completed they are free to be re-used. 

IV Index Update is automatically initiated by a node when it finds itself in the risk of exhausting it's sequence numbers or another node in the network is in the risk of exhausting their sequence numbers. Once the IV Index procedure completed meaning iv index of the network is incremented, all deleted nodes for a given iv index are freed to be reused. This is a bit of a complicated topic to discuss here so I recommend reading section 3.10.5 of the Bluetooth Mesh Specification version 1.0.1.

Deleting/Resetting a node will blacklist itself from being involved in a Key refresh procedure. Key refresh procedure involves distributing new keys to nodes in a mesh network. All excluded nodes are not aware of this allowing you to provision a brand new node with the latest keys. Once again this information can be read in the section 3.10.4 of the Bluetooth Mesh Specification version 1.0.1.

> Under what circumstances does a Key Refresh Procedure happen? Does the library do it in background? My app doesn't handle it (at least not knowingly), so the previously used mesh addresses should remain in the exclusion list, I believe.

A user has to initiate the Key refresh procedure and this is usually done when a key is compromised or if you decide to distribute new keys to your network. It is supported by the library and the user has to initiate this. Check the doc here https://github.com/NordicSemiconductor/Android-nRF-Mesh-Library/pull/381#issue-797675690.

> Also thanks for the links to the nRF Connect SDK. Unfortunately I had to revert back to nRF5 SDK for Mesh because flash memory run out...

FYI nRF5 SDK  for Mesh has similar behavior on the examples.

Hope this helps!

closed 2022-03-07 by roshanrajaratnam