nordicsemi/Android-nRF-Mesh-Library #480: Not able to reset Node

choi-seung-min, 2022-02-24:
I'm trying to use this app's features in flutter app.

I impletmeted scanning, connecting, identifying, provisioning but I cannot run reset function with app.

After i call ```createConfigMeshMessage()``` with ```ConfigNodeReset()```, nothing happens.
it's not removed from provisioned list and device is found on provisioned devices scanning.

I followed ```NodeConfigurationActivity``` for reset method calling process.
Before resetting, I set ```selectedNode```on ```NrfMeshRepository```.
I expect ```NrfMeshRepository```'s ```onDeviceDisconnected``` calling but there's not.

As there's no document and no flutter library, I have no way to figure it out.

What callBack/method can i check for trouble shooting? 
or Is there any missing settings for ```NrfMeshRepository``` or missing key fot something?

here's my repo 
https://github.com/choi-seung-min/MeshSample.git

R0m4in-dooz, 2022-02-24:
Hello @choi-seung-min,
We are using Flutter and have no issue sending node resets. But we don't use the method you mentioned, we are using `meshManagerApi.createMeshPdu(ConfigNodeReset(), unicastAddress)`..did you try this one ?

R0m4in-dooz, 2022-02-24:
The thing is, as long as you don't receive a feedback, nothing will happen. Checkout your node's logs if you can too 👍

choi-seung-min, 2022-02-24:
@R0m4in-dooz Thanks for reply.

I'm using this ```nrfMesh.mMeshManagerApi.createMeshPdu(node.getUnicastAddress(), new ConfigNodeReset());``` method to reset node.

choi-seung-min, 2022-02-24:
@R0m4in-dooz  Thanks for reply.

I’m using ````nrfMesh.mMeshManagerApi.createMeshPdu(node.getUnicastAddress(), new ConfigNodeReset());```` for node reset now.

I’ve checked this repo to implement my goal functions. Node reset too.

Is this repo’s source does not sure work on flutter?

My ```MashManagerApi.java``` file's ```createMeshPdu``` is like this:

```
@Override
public void createMeshPdu(final int dst, @NonNull final MeshMessage meshMessage) {
    if (!MeshAddress.isAddressInRange(dst)) {
        throw new IllegalArgumentException("Invalid address, destination address must be a valid 16-bit value.");
    }
    final Provisioner provisioner = mMeshNetwork.getSelectedProvisioner();
    if (provisioner != null && provisioner.getProvisionerAddress() != null) {
        UUID label = null;
        if (MeshAddress.isValidVirtualAddress(dst)) {
            label = mMeshNetwork.getLabelUuid(dst);
            if (label == null) {
                throw new IllegalArgumentException("Label UUID unavailable for the virtual address provided");
            }
        }
        mMeshMessageHandler.createMeshMessage(provisioner.getProvisionerAddress(), dst, label, meshMessage);
    } else {
        throw new IllegalArgumentException("Provisioner address not set, please assign an address to the provisioner.");
    }
}
```

Do i have to change my nrf mesh library’s version?

R0m4in-dooz, 2022-02-24:
Ah yes, i messed the parameters order, but it should work indeed.
For Flutter, everything is working, you have to wrap the native code with method channels and event streams and you should be able to have a mostly Dart api. We may release our plugin in the future but for now I can't tell much except that it is possible to use the Nordic libraries in Flutter 👍

For the library version, always try to use the latest (v3.1.7 today) like every dependency 😇

roshanrajaratnam, 2022-02-24:
@R0m4in-dooz Thanks for helping out!

choi-seung-min, 2022-03-07:
Thanks, I found that i did not set `unicastAddress` to network on provisioning process.

It’s working perfectly.

closed 2022-03-07 by choi-seung-min

R0m4in-dooz, 2022-03-18:
Hello @choi-seung-min,
We just open sourced our Flutter plugin that provides a Dart API to use this library and its iOS counterpart.
It still lacks some "get started" documentation, but the example app should be enough to jump in with basic mesh network features 👍 
If it can help you to develop your projects, feel free to give [our plugin](https://pub.dev/packages/nordic_nrf_mesh) a try !
Regards 👋