nordicsemi/Android-nRF-Mesh-Library #485: Can I keep connection with node after identify?

tomato8160, 2022-03-02:
# Question

We must disconnect from any nodes that we are connected to before we start scanning.

Why do this?

I want to keep connection with node after identify (without provisioning)

roshanrajaratnam, 2022-03-02:
Well you don't have to disconnect. The sample app is designed in this manner.

There is a 60 second timeout defined in the mesh profile specification version 1.0.1 section 5.4.4. This states that you should disconnect if no message is sent during a time frame of 60 seconds during the provisioning stages.

tomato8160, 2022-03-02:
Sorry, I think there is not enough of something. so I'm commenting more

while I connected to provisioned proxy nodes

[image]

I tried to scan devices
[image]

and I clicked item
[image]

and I got back to MainActivity without any progress(identify or provisioning)

I finally noticed that disconnetion to node
[image]

So, I chekced onItemClicked method in ScannerAcitvity, 
and there was disconnec method

[image]

Why do I need to disconnect all connected nodes before starting a scan?

I want to keep connection..

roshanrajaratnam, 2022-03-02:
The reason we disconnect here is that we did not want to stay connected to another proxy while provisioning. This approach allowed us not needing to figure out where a mesh packet arrived from and when sending provisioning message we did not have to verify which node to send to.

It's a sample app so we decided to take this approach which was easiest at that time. However you may decide to change the source and implement it the way you like :)