nordicsemi/Android-nRF-Mesh-Library #486: How to avoid multiple nodes with the same UUID after DFU?

FluffyBunniesTasteTheBest, 2022-03-10:
---

**Describe the bug**
The first version of the firmware was a simple generic level model, which will be replaced by a proprietary vendor model.
When updating the firmware from the generic level to the vendor model, the node automatically unprovisions itself because Softdevice/Mesh SDK detects that the (generic level) model data is invalid for the new vendor model.

Now, when reprovisioning the node after DFU, the library contains the node twice: Once with the generic level model and once with the vendor model. Unprovisioning any of them will remove both nodes. Question now is, is there a way to remove the old (generic level) node from the library without also removing the (new) node containing the vendor model?

Considering the following from [this discussion](https://github.com/NordicSemiconductor/Android-nRF-Mesh-Library/issues/479#issuecomment-1048531538):

> each node has a unique UUID and if the UUID is the same for each provisioned device you will be overwriting the same node during provisioning. 

Makes me think that this could be a bug in the library. 

**To Reproduce**
Steps to reproduce the behavior:
1. Open an example project from nRF5 SDK for Mesh in Segger Embedded Studio, compile and program on a device.
2. Provision the device.
3. Open another example project from nRF5 SDK for Mesh in Segger Embedded Studio, compile and program **onto the same device**. The device will start unprovisioned because `model_config_file_config_apply()` returns `NRF_ERROR_INVALID_DATA`. 
4. Reboot the device once. This is necessary, otherwise provisioning will fail.
5. Provision the device.
6. The app now shows both nodes. 

**Expected behavior**
Because the node's UUID did not change during DFU, reprovisioning the node should overwrite (remove) any other nodes with the same UUID.

**Platform details:**
 - Device: Any Android Device.
 - OS: Any Android Version.
 - Library Version 3.1.6

roshanrajaratnam, 2022-03-11:
I will try take a look at this beginning of next week and get back to you!

roshanrajaratnam, 2022-03-11:
Did you check if the UUIDs are the same after flashing both instances?

roshanrajaratnam, 2022-03-11:
@FluffyBunniesTasteTheBest I did speak to our mesh team. Check below.

Yes, because they did not fully erase the device before burning the second app. When device is not fully erased before burning the new application, the persistence memory data from old app gets preserved in such case and then this error is triggered. On the next reboot, stack will clean up dirty persistence storage data and start a fresh as unprovisioned device. This is part of the implicit spec requirement that once the device is provisioned, its composition (i.e. number of models, and elements, or any other persistently stored configuration) cannot be changed unless it is unprovisioned.

FluffyBunniesTasteTheBest, 2022-03-11:
@roshanrajaratnam Thanks for the quick response and the clarifications!

> Did you check if the UUIDs are the same after flashing both instances?

No, I just assumed that the UUIDs don't change because of [Tesc's first answer at this discussion at devzone](https://devzone.nordicsemi.com/f/nordic-q-a/38526/mesh-uuid). (The firmware uses the default implementation - i.e. does not provide UUIDs through `mesh_stack_init_params_t`, so the UUIDs can't change). But beside that, if the UUIDs would be different, both nodes would be "different" nodes and deleting one of them wouldn't also remove the other one from the app. 

> On the next reboot, stack will clean up dirty persistence storage data and start a fresh as unprovisioned device. This is part of the implicit spec requirement that once the device is provisioned, [...]

This automatic unprovisioning after the composition data changed is quite an awesome feature of the mesh stack. Thanks for the additional clarification. I believe that the library should behave the same here and also remove any previously provisioned nodes when a new node with a matching UUID is provisioned.

roshanrajaratnam, 2022-03-11:
If there are matching UUIDs library will overwrite it. That's how the spec is defined.  I believe you have two different UUIDs causing your old node not being overwritten.

FluffyBunniesTasteTheBest, 2022-03-11:
K, I'll check, but if the UUIDs are different, why are both nodes deleted when resetting one of the nodes?

closed 2022-03-13 by FluffyBunniesTasteTheBest

roshanrajaratnam, 2022-03-14:
No worries, glad it works! :)

R0m4in-dooz, 2022-03-18:
Hello @FluffyBunniesTasteTheBest,
We just open sourced our Flutter plugin that provides a Dart API to use this library and its iOS counterpart.
It still lacks some "get started" documentation, but the example app should be enough to jump in with basic mesh network features 👍 
If it can help you to develop your projects, feel free to give [our plugin](https://pub.dev/packages/nordic_nrf_mesh) a try !
Regards 👋

FluffyBunniesTasteTheBest, 2022-03-21:
@R0m4in-dooz Sounds promising. Thanks for sharing this! I'll have a look at it.