nordicsemi/Android-nRF-Mesh-Library #488: BaseMeshNetwork.id is never updated

R0m4in-dooz, 2022-03-23:
---

**Describe the bug**
The `id` field of `BaseMeshNetwork` seems to never be updated and is equals to the [initial value](https://github.com/NordicSemiconductor/Android-nRF-Mesh-Library/blob/main/mesh/src/main/java/no/nordicsemi/android/mesh/BaseMeshNetwork.java#L68) : `http://www.bluetooth.com/specifications/assigned-numbers/mesh-profile/cdb-schema.json#`.

**To Reproduce**
Steps to reproduce the behavior:
1. Load a mesh network
2. use the getter `meshNetwork.id` and see its value

**Expected behavior**
Maybe it's intended, but I would expect to have the same value as `meshUUID` field ? or this field has another purpose...? :thinking:

**Platform details:**
 - Device: ZenFone
 - OS: Android 8.0
 - Library Version : latest release

roshanrajaratnam, 2022-03-23:
Hi, it's not meant to be updated. It's a part of the Mesh CDB JSON schema and will point to a schema for that version.

closed 2022-03-23 by roshanrajaratnam

R0m4in-dooz, 2022-03-23:
Oh I see ! Thx for explanation