nordicsemi/Android-nRF-Mesh-Library #490: ConcurrentModificationException on insertNetwork

daretobeorjan, 2022-03-30:
Seems to be very hard to trigger, the is the first time ever for me, but still seems to be a similar timing issue in database writing to the others solved previously.

`Process: se.enequi.quipower.test.debug, PID: 1016
    java.util.ConcurrentModificationException
        at java.util.ArrayList$Itr.next(ArrayList.java:860)
        at java.util.Collections$UnmodifiableCollection$1.next(Collections.java:1089)
        at androidx.room.EntityInsertionAdapter.insert(EntityInsertionAdapter.java:95)
        at no.nordicsemi.android.mesh.data.ProvisionedMeshNodesDao_Impl.insert(ProvisionedMeshNodesDao_Impl.java:293)
        at no.nordicsemi.android.mesh.MeshNetworkDb.lambda$insertNetwork$0(MeshNetworkDb.java:153)
        at no.nordicsemi.android.mesh.MeshNetworkDb$$ExternalSyntheticLambda29.run(Unknown Source:16)
        at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1167)
        at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:641)
        at java.lang.Thread.run(Thread.java:920)`

I don't think Collections.unmodifiableList is enough to prevent reading a modified list. UnmodifiedList is just a wrapper that reads the backing collection. It only prevents modification by the receiver, so another instance that has access to the modifiable list can still cause concurrent modifications.

daretobeorjan, 2022-03-30:
This seems like a safer approach to insertNetwork:
```java
            meshNetworkDao.insert(meshNetwork);
            netKeysDao.insert(meshNetwork.getNetKeys());
            appKeysDao.insert(meshNetwork.getAppKeys());
            provisionersDao.insert(new ArrayList<>(meshNetwork.getProvisioners()));
            if (!meshNetwork.nodes.isEmpty()) {
                nodesDao.insert(new ArrayList<>(meshNetwork.getNodes()));
            }

            if (meshNetwork.groups != null) {
                groupsDao.insert(new ArrayList<>(meshNetwork.getGroups()));
            }

            if (meshNetwork.scenes != null) {
                scenesDao.insert(new ArrayList<>(meshNetwork.getScenes()));
            }
```

Or perhaps use new ArrayList in MeshNetwork.getNodes et all instead? Though I don't know if you would want to increase the overhead of those methods like that, although any iteration on the returned value may cause the exception, so perhaps it would be the safest.

roshanrajaratnam, 2022-03-30:
I think it will only increase the overhead IMHO, I can definitely try making a copy!

daretobeorjan, 2022-03-30:
It's probably fine if callers are the ones who decide if they need a copy or not.

roshanrajaratnam, 2022-03-30:
@daretobeorjan this is from 3.1.8? or 3.1.7?

daretobeorjan, 2022-03-30:
> @daretobeorjan is this is from 3.1.8? or 3.1.7? How did you trigger this again. I will try on my end as well.

We're actually running the mesh SDK as a submodule on our own internal dev-branch for features that are not merged into upstream. Our most recently merged upstream commit is 4ecb0ede561a16ccac7da9d8e14e270da981fbf4.

roshanrajaratnam, 2022-03-30:
ok so you have the latest! I've made the suggested changes. Let me know if that works!

roshanrajaratnam, 2022-04-07:
Possible fixes could be contained in #496

closed 2022-05-25 by roshanrajaratnam