nordicsemi/Android-nRF-Mesh-Library #494: Add option to subscribe to All Proxies, All Friends and All Relays

philips77, 2022-04-06:
Currently, when setting Subscription to a Model, one can only select a user-defined group.
However, spec 1.0.1 allows any Fixed Group Address EXCEPT All Nodes to be set as model subscription.
By design, messages targeting All XXX shall be processed by all modes on the primary element if they are subscribed to the Application Key (chapter 3.4.2.4). But is is possible to subscribe to any All XXX except All Nodes on a model on non-primary element. Current version of nRF Mesh (3.1.x) does not support that.

Note: bug found on iOS version, check how it is on Android.

philips77, 2022-04-06:
[image]
Confirmed, only use defined groups are allowed.

roshanrajaratnam, 2022-04-07:
As of now the app allows only to subscribing to groups and virtual addresses. Thanks for pointing out!

roshanrajaratnam, 2022-04-07:
Some more clarification:
1. All models on the primary element are automatically subscribed to all All XXX addresses, if the node supports such feature.
    E.g. If a node has Relay Feature enabled, all models on the Primary Element, that are bound to the App Key used to secure the message, will receive a message sent to All Relays encrypted with this App Key.
2. User may subscribe ANY model on ANY element to any fixed group address EXCEPT All Nodes.
    This "manual" subscription OVERRIDES the automatic one, and NO FEATURE CHECKS are done before giving the message for processing. That means, that if a model is manually subscribed to All Relays, it will always receive messages sent to All Relays given it is bound to the App Key. This applies to ANY model, including those on the Primary Element.

commit 42f26ad0 (2022-04-07): Fixes #494