nordicsemi/Android-nRF-Mesh-Library #503: Enable inheritance in VendorModelMessageStatus

likebobby, 2022-04-27:
**Is your feature request related to a problem? Please describe.**
When using vendor models the VendorModelMessageStatus is a final class and can not be used in class inheritance. Is there a reason for this or could it be un-finalized?

**Describe the solution you'd like**
Can it be changed as:
```java
 @SuppressWarnings({"WeakerAccess"})
-public final class VendorModelMessageStatus extends ApplicationStatusMessage implements Parcelable {
+public class VendorModelMessageStatus extends ApplicationStatusMessage implements Parcelable {
 
     private static final String TAG = VendorModelMessageStatus.class.getSimpleName();
     private final int mModelIdentifier;
@@ -66,7 +66,7 @@ public final class VendorModelMessageStatus extends ApplicationStatusMessage imp
     }
 
     @Override
-    void parseStatusParameters() {
+    public void parseStatusParameters() {
         Log.v(TAG, "Received Vendor model status: " + MeshParserUtils.bytesToHex(mParameters, false));
     }
```

**Describe alternatives you've considered**
The other option is to always parse the model as VendorModelMessageStatus and them read the parameters after the message is recieved, a bit less declarative when implementing VendorModels.

If the change can happen I am happy to contribute with a PR for it.

roshanrajaratnam, 2022-04-27:
Well I don't see why it shouldn't be public. The idea during implementation was just to return a array of bytes so that users can parse it in their app, but this could well be done by allowing VendorModelMessageStatus to be extended. I think this might be a good idea to open up the VendorModelMessageStatus.

I guess you plan on extending in your app?

likebobby, 2022-04-27:
Yes, the plan was to extending.

likebobby, 2022-05-17:
We saw some problems we the inheritance approached and used mapping of vendor status messages to vendor models instead.

closed 2022-05-17 by likebobby