nordicsemi/Android-nRF-Mesh-Library #514: nRF Mesh could not scaned my node device

chinli, 2022-06-02:
---

**Describe the bug**
I used Mynewt-nimble  to create node with pb_adv, but nRF Mesh could not scaned my device(in Scanner page not display).

**To Reproduce**
Steps to reproduce the behavior:
1. power on my ble device, and enable unprovisioned beacon adv;
2. use nRF Connect could scaned this device;
3.  use nRF Mesh, click the <ADD NODE> to enter the scanner page, but didn't displey always;

**Expected behavior**
nRF Mesh can scaned my device, and start to provisioning.

**Platform details:**
 - Device: HuaWei HONOR P6
 - OS: [e.g. Android 11.0]
 - Version of nRF Mesh: 3.2.4

**Logs / Screenshots**
adv data with hci log:
[image]

scan result with nRF Conenct
[image]

thanks~

roshanrajaratnam, 2022-06-03:
@chinli this is strange. Did you scan via "Add Node"?

chinli, 2022-06-06:
@roshanrajaratnam Yes, I'm a little confused that it must be enable pb_gatt or not?  I wonder if you could suggest some testing.
thanks ~

roshanrajaratnam, 2022-06-06:
Hi, I'm not sure if I could answer you on my-newt nimble. If the device is advertising you should be able to see the device in the scanner.

roshanrajaratnam, 2022-06-28:
@chinli are you still facing this issue?

chinli, 2022-07-01:
Unfortunately, this issue remains. but i enable both of pb_adv and pb_gatt, then nRF Mesh APP can display my device, provionsoning successful.
In my opinion, is APP not support pb_adv only or not?

closed 2022-07-01 by roshanrajaratnam

chinli, 2022-07-01:
Got it~~