nordicsemi/Android-nRF-Mesh-Library #517: What's the most efficient way to ping all nodes in a mesh network?

FluffyBunniesTasteTheBest, 2022-06-05:
The idea is to monitor the nodes accessibility. One way to implement it would be to add a ping message to the firmware, but the way it looks any broadcasted acknowledged message would do. What message would be a good fit in terms of efficiency? (The message should cause as little work as possible on the nodes). How would you implement it?

closed 2022-06-06 by philips77

FluffyBunniesTasteTheBest, 2022-06-07:
Thanks for taking the time clarifying this. Just mentioning a mesh message would have done...

philips77, 2022-06-07:
This is a bit complicated. I'm not aware of any ping type message from the spec. You may look into heartbeat config messages, which can configure a node to send periodical heartbeats. Or design vendor message. But perhaps the support team may have better ideas...