nordicsemi/Android-nRF-Mesh-Library #518: nRF Mesh only connects to last provisioned node

Brew317, 2022-06-05:
---

**Describe the bug**
The nRF mesh andriod app only will connect to the last node that it provisioned

**To Reproduce**
Steps to reproduce the behavior:
I have three nodes in my dev system all nRF5340DK, I provision each one everything is find.
one node is a client, the other two are identical server/client nodes

1. I provision all three nodes and close nRF Mesh
2. System is working as expected
3. I go to change the publication interval on a node
4. I open nRF Mesh and select "connect"
5. I see all three nodes with their proper MAC Address, I select a node
6. Phone screen shows that it is connecting to the node I selected
7. When it connects it is not the node I selected, it is the last node I provisioned.
8. I tried turning off the other two nodes
9. and at the "connect" step nRF Mesh only shows the active node
10. When I select it is again the last node I provisioned which is not even turned on.
11. I have to Clear Storage on the nRF Mesh app and start over

**Expected behavior**
A clear and concise description of what you expected to happen.

**Platform details:**
 - Device: Pixel 5
 - OS: Android 12
 - Library Version 3.2.4

**Logs / Screenshots**
Add any logs or screenshots that can help us debug the issue faster

roshanrajaratnam, 2022-06-10:
Hi, I'll take a look at this today, this seems a bit strange.

roshanrajaratnam, 2022-06-10:
I did this give this a go and I can't seem to produce this on my end. I can connect to the node I selected from the list of nodes irrespective of where I am within the app. To ensure this you can have nRF Connect running in the background with "Gatt Server enabled" and try connecting to the node you want to using nRF Mesh and if you switch back to nRF Connect you can see which device you are connected to based on the mac address.

roshanrajaratnam, 2022-06-21:
Hi, we don't have any internal test versions at the moment and if the version you have is the same as the one that's publicly available. I will try to have a look at this today.

roshanrajaratnam, 2022-06-22:
Are you using NCS based examples and version? Do you have any chance to try on 52840 DKs?

Brew317, 2022-06-23:
not using NCS using nrf examples nRF connect for desktop v1.9.1

roshanrajaratnam, 2022-06-24:
Have you tried to see if you notice the same behaviour? Or does it happen with your firmware?

Brew317, 2022-06-24:
I don't understand the question. I am using the Android nRF Mesh application downloaded from the app store "AS IS".
I have not cloned the nRF Mesh SDK from this site. I don't intend to modify the nRF mesh provisioner. 
it is just the tool (only tool) I have to provision the nRF5340DK eval boards

Brew317, 2022-06-24:
Ah! yes I misunderstood you.
I am running the nRF Connect v1.9.1\nrf\samples\bluetooth\mesh\light
and the nRF Connect v1.9.1\nrf\samples\bluetooth\mesh\light_switch
and modifying the Generic OnOFF server/client model and the Generic Level server/client model
below is my modified Node element structure.
the code works could having a level server and client in the same element be the issue?
too many elements ?
I will swap the ONOFF in element 1 wtih the Level server in element 2 and see what happens

struct bt_mesh_elem elements[] = {
	BT_MESH_ELEM(1,
		     BT_MESH_MODEL_LIST(
			     BT_MESH_MODEL_CFG_SRV,
			     BT_MESH_MODEL_HEALTH_SRV(&health_srv, &health_pub),
				 BT_MESH_MODEL_ONOFF_CLI(&rssi_detect.client)),
		     BT_MESH_MODEL_NONE),
	BT_MESH_ELEM(2,
		     BT_MESH_MODEL_LIST(
				 BT_MESH_MODEL_LVL_SRV(&cmd_stat.srv),
				 BT_MESH_MODEL_LVL_CLI(&rssi_subscribe.client)),
		     BT_MESH_MODEL_NONE),
	BT_MESH_ELEM(3,
		     BT_MESH_MODEL_LIST(
				 BT_MESH_MODEL_LVL_SRV(&rssi_publish.srv)),
	 	     BT_MESH_MODEL_NONE),
	BT_MESH_ELEM(4,
		     BT_MESH_MODEL_LIST(
	 		     BT_MESH_MODEL_LVL_SRV(&cmd_stat.srv)),
	 	     BT_MESH_MODEL_NONE),
};

roshanrajaratnam, 2022-06-25:
I suggest posting this issue on devzone as this is more related to the SDK and would help reaching a wider audience. Our dedicated tech support team should be able to help you on this.

closed 2022-06-25 by roshanrajaratnam