nordicsemi/Android-nRF-Mesh-Library #521: Provide option to process Nrf library logs

anees17861, 2022-06-13:
**Is your feature request related to a problem? Please describe.**
The Library relies on Android Native Logging. This way we can observe logs by connecting to laptop but can't observe for remote devices

**Describe the solution you'd like**
I can think of following two options where option 1 seems the easier approach)
1. Provide an option to set callback for logger interface
2. Allow to extend MeshManagerApi which has a default implementation of a log function. (Same as BleManager from the nordic ble library)

**Describe alternatives you've considered**
Couldn't find any other way

**Additional context**
The reason why we require this is to debug an issue where fetch composition data doesn't receive response if there are 20+ devices nearby that are provisioned in the same network.

anees17861, 2022-06-14:
Will it be alright if I submit a PR to the same so that you can integrate it as quickly as possible?

roshanrajaratnam, 2022-06-15:
Hey, we gladly accept pull requests :)

anees17861, 2022-06-22:
Hi @roshanrajaratnam 
Sorry for the delay but I have created a pull request #522 

Please go through the same and see if it has any issues

roshanrajaratnam, 2022-06-22:
Thanks I will take a look at this later today!

PR #522 (2022-06-28): feat: Custom logger added with option to attach handler
Added a static class MeshLogger.
This class accepts a LogHandler object providable by app.
If the LogHandler is null then it falls back to android.util.Log print statement.

closes #521
closed 2022-06-28 by roshanrajaratnam