nordicsemi/Android-nRF-Mesh-Library #523: IV index doesn't update on a imported network

anees17861, 2022-06-23:
---

**Describe the bug**
When a network is exported and imported to a new mobile device, the iv index of network is lost. Although it should update on receiving a secure network beacon after connecting to proxy, app discards the message and hence any message the app sends is ignored by the proxy.

Issue seems to be that when network is imported, IVIndex default transitionDate is set to currentDate and thus always failing the following condition
[image]

**To Reproduce**
Steps to reproduce the behavior:
1. Create a mesh network
2. Add multiple nodes
3. Generate enough traffic to increment iv index (or change it manually if possible)
4. Export network
5. Import on another phone
6. Change provisioner address on the new phone
7. Try provisioning new node in the vicinity of existing nodes.
8. It fails 90% of the time at Sending Composition Data Get as the node will ignore messages with older IV

**Expected behavior**
Expected The IV index to get updated on first secure beacon

**Platform details:**
 - Device: Redmi 9 power
 - OS: Android 11
 - Library Version : 3.2.4

**Logs / Screenshots**
```
2022-06-22 19:58:08 [INFO]  BleManager: Connected to D7:31:53:E5:D3:DE
2022-06-22 19:58:09 [INFO]  BleManager: PHY updated (TX: LE 2M, RX: LE 2M)
2022-06-22 19:58:09 [INFO]  BleManager: Connection parameters updated (interval: 30.0ms, latency: 0, timeout: 200ms)
2022-06-22 19:58:09 [INFO]  BleManager: Services discovered
2022-06-22 19:58:09 [INFO]  BleManager: MTU changed to: 247
2022-06-22 19:58:09 [INFO]  BleManager: Notification received from 00002ade-0000-1000-8000-00805f9b34fb, value: (0x) 01-01-00-DD-05-4B-24-03-2B-24-ED-00-00-00-03-0B-34-C1-F4-8B-B3-B7-A1
2022-06-22 19:58:09 [INFO]  BleManager: Data written to descr. 00002902-0000-1000-8000-00805f9b34fb, value: (0x) 01-00
2022-06-22 19:58:09 [INFO]  BleManager: Notifications enabled

** The problem point
2022-06-22 19:58:09 [OTHER]  MeshManagerApi/Received mesh beacon: SecureNetworkBeacon { KeyRefreshActive: false, IV Index: IV Index: 3, IV Update Active: false, Authentication Value: 0x0B34C1F48BB3B7A1}
2022-06-22 19:58:09 [OTHER]  MeshManagerApi/Secure Network Beacon authenticated.
2022-06-22 19:58:09 [OTHER]  MeshManagerApi/Last IV Index: 0
2022-06-22 19:58:09 [WARN]  MeshManagerApi: Discarding beacon IV Index: 3, IV Update Active: false, last 0, changed: 0h ago, test mode: false
**

2022-06-22 19:58:09 [INFO]  BleManager: Connection parameters updated (interval: 7.5ms, latency: 0, timeout: 5000ms)
2022-06-22 19:58:10 [INFO]  BleManager: Connection parameters updated (interval: 30.0ms, latency: 0, timeout: 200ms)
2022-06-22 19:58:11 [OTHER]  MeshTransport/Src address: 7FFF
2022-06-22 19:58:11 [OTHER]  MeshTransport/Dst address: 0021
2022-06-22 19:58:11 [OTHER]  MeshTransport/Key: EC3A1FA4903D5320049C69424C805AE3
2022-06-22 19:58:11 [OTHER]  MeshTransport/akf: 0
2022-06-22 19:58:11 [OTHER]  MeshTransport/aid: 0
2022-06-22 19:58:11 [OTHER]  MeshTransport/aszmic: 0
2022-06-22 19:58:11 [OTHER]  MeshTransport/Sequence number: 2
2022-06-22 19:58:11 [OTHER]  MeshTransport/Access message opcode: 8008
2022-06-22 19:58:11 [OTHER]  MeshTransport/Access message parameters: FF
2022-06-22 19:58:11 [OTHER]  AccessLayer/Created Access PDU 8008FF
2022-06-22 19:58:11 [OTHER]  UpperTransportLayer/Device nonce: 02000000027FFF002100000000
2022-06-22 19:58:11 [OTHER]  UpperTransportLayer/Encrypted upper transport pdu: 959D5EF58D5EBE
2022-06-22 19:58:11 [OTHER]  LowerTransportLayer/Unsegmented Lower transport access PDU 00959D5EF58D5EBE
2022-06-22 19:58:11 [OTHER]  NetworkLayer/Encryption key: 6E46DED43FC00F629C620A59C48E7265
2022-06-22 19:58:11 [OTHER]  NetworkLayer/Privacy key: 27D5F91E8D8FDDA3AEC347217878A354
2022-06-22 19:58:11 [OTHER]  NetworkLayer/Sequence Number: 000002
2022-06-22 19:58:11 [OTHER]  NetworkLayer/Encrypted Network payload: 6B6358D42F28B044FF90AA6E6A60
2022-06-22 19:58:11 [INFO]  BleManager: Data written to 00002add-0000-1000-8000-00805f9b34fb, value: (0x) 00-1E-64-67-B9-BB-F5-ED-6B-63-58-D4-2F-28-B0-44-FF-90-AA-6E-6A-60
2022-06-22 19:58:11 [OTHER]  MeshManagerApi/MeshNetwork pdu sent: 0x001E6467B9BBF5ED6B6358D42F28B044FF90AA6E6A60
2022-06-22 19:58:14 [INFO]  BleManager: Notification received from 00002ade-0000-1000-8000-00805f9b34fb, value: (0x) 01-01-00-DD-05-4B-24-03-2B-24-ED-00-00-00-03-0B-34-C1-F4-8B-B3-B7-A1
2022-06-22 19:58:14 [OTHER]  MeshManagerApi/Received mesh beacon: SecureNetworkBeacon { KeyRefreshActive: false, IV Index: IV Index: 3, IV Update Active: false, Authentication Value: 0x0B34C1F48BB3B7A1}
2022-06-22 19:58:14 [OTHER]  MeshManagerApi/Secure Network Beacon authenticated.
2022-06-22 19:58:14 [OTHER]  MeshManagerApi/Last IV Index: 0
2022-06-22 19:58:14 [WARN]  MeshManagerApi: Discarding beacon IV Index: 3, IV Update Active: false, last 0, changed: 0h ago, test mode: false
2022-06-22 19:58:35 [WARN]  BleManager: Error: (0x13): GATT CONN TERMINATE PEER USER
2022-06-22 19:58:35 [INFO]  BleManager: Disconnected
```

roshanrajaratnam, 2022-06-23:
Hi thanks for reporting this issue, I will take a look at this as soon as possible.

anees17861, 2022-06-23:
@R0m4in-dooz Thanks for this. Instead of removing the check completely, I decided to update the iv index date after import is done since that api is public. I set the year to 1970 :P.
The advantage was I did not need to put the library under source control. I could still use maven plugin

anees17861, 2022-06-28:
https://github.com/NordicSemiconductor/Android-nRF-Mesh-Library/blob/6d9e71ac432d33f67c30f1ee16ce342fcf094952/mesh/src/main/java/no/nordicsemi/android/mesh/MeshManagerApi.java#L334

@roshanrajaratnam Is there any specific reason why the previous transition date is put here. In case of iOS it just replaces the entire ivIndex with the one generated by the received beacon

roshanrajaratnam, 2022-06-28:
Ideally it should, I'll have to investigate this.

roshanrajaratnam, 2022-06-28:
@anees17861 this was a good catch, IV Index update is a tricky one! Please give it a test and let me know if this works for you as expected.

anees17861, 2022-06-28:
Although it fixes the issue of the transition date not updating at all, it doesn't solve the issue raised here of export import.
I observed that the initial ivIndex transition date is actually `nil` in iOS but in android it is `Calendar.getInstance()` which ends up taking the date when the network is imported.
In `canOverwrite()` if date is found to be `nil` then it returns true
Keeping ivIndex transitionDate initially null logically makes sense since the ivIndex never transitioned. Only thing is need to be careful of null safety

Another issue I observed while comparing with iOS is that on importing in iOS if there is an existing ivIndex stored for the mesh network uuid, then the existing ivIndex is reused. But in Android that does not happen (although sequence numbers are updated). So if you export and import the network within the same phone, then also it loses the ivIndex

https://github.com/NordicSemiconductor/Android-nRF-Mesh-Library/blob/d2292fac77cd195bb962c4be6b2cd245cc6f0e86/mesh/src/main/java/no/nordicsemi/android/mesh/MeshManagerApi.java#L938-L950

In this section I believe existing IvIndex needs to be ported to the new network

roshanrajaratnam, 2022-06-28:
Now I see your point, I thought it was doing this already but seems like I don't. Let me fix this.

roshanrajaratnam, 2022-06-28:
Hopefully this should work now!

anees17861, 2022-06-28:
Unfortunately it still doesn't work.

As you mentioned for iOS the transition date is stored in preferences but it's only stored when the update procedure actually takes place. For a new network this never happens until at-least one beacon is received.
Following is the only point where iv index transition date is updated by the library.
https://github.com/NordicSemiconductor/IOS-nRF-Mesh-Library/blob/4d1a40d924b036683122afd9e91aed1434432089/nRFMeshProvision/Classes/Layers/Network%20Layer/NetworkLayer.swift#L318-L319

canOverwrite() always passes for first beacon in a new network.
Practically speaking, it will always take whatever iv index it receives first time after connecting over proxy in a new network.

roshanrajaratnam, 2022-06-28:
you are correct, but have you had a chance to increment the IV Index of the existing network and trying to import the same network again?

anees17861, 2022-06-28:
Also is it safe to ignore the transition date altogether or provide an option for the app to set a flag to ignore it? It is a bit concerning that it will take the first packet and won't update for at-least 192 hours.
In case of provisioning + configuring a new node in existing network with higher iv index:

1. the proxy may first send secure beacon with iv index = 0
2. Nrf Library would update it's transitionDate to current date
3. The proxy may get beacons from other devices and update it's own iv
4. Then send new secure beacon with new iv
5. This time library would discard it since it had just updated

roshanrajaratnam, 2022-06-28:
> Also is it safe to ignore the transition date altogether or provide an option for the app to set a flag to ignore it? It is a bit concerning that it will take the first packet and won't update for at-least 192 hours.

It's only an IV Index recovery that will not take place for 192 hours which means that the received Index must be current IvIndex + 1. And normal operation must last for at least 96 hours. :)

anees17861, 2022-06-28:
> you are correct, but have you had a chance to increment the IV Index of the existing network and trying to import the same network again?

Ok confirmed. Reimport works

anees17861, 2022-06-28:
Hi @roshanrajaratnam I think PR #524 and your changes in https://github.com/NordicSemiconductor/Android-nRF-Mesh-Library/tree/hotfix/fix-iv-index-transition-date should together solve the issue fully

PR #524 (2022-07-01): fix: initial iv index transition date set to null
This PR partially fixes #523.

It sets the default transition date to null and thus accepts an iv index update if it's receiving a beacon with higher ivindex in the network for the very first time
commit 257618fe (2022-07-01): Fixes #523 When importing a network, load the ivIndex last known ivIndex if the network exists from the local database. This will be updated accordingly when the secure network beacon is received.
commit a233e5a0 (2022-07-01): Fixes #523 by replaces the secure network beacon in case it's higher than the existing, if not marks the current iv update active state to false. canOverwrite ensures this by discarding any invalid beacons.
closed 2022-07-01 by roshanrajaratnam