nordicsemi/Android-nRF-Mesh-Library #529: Provisioner sequence no incorrectly set upon visiting current provisioner

pmjagtap, 2022-08-06:
---

**Describe the bug**
Upon updating provisioner, the sequence number of provisioner is set to lower than current. This causes further mesh communication failure while interacting with provisioned node.

**To Reproduce**
Steps to reproduce the behavior:
1. Put break point in public final void setSequenceNumber(final int sequenceNumber) located in ProvisionedMeshNode.java
2. Put break point in public boolean updateProvisioner(@NonNull final Provisioner provisioner) in BaseMeshNetwork.java
3. Connect to provisioned node > Create mesh traffic (e.g Read Proxy State) using connected node > Note last seq no increment of provisioner node using break point setup in step 1
4. Disconnect from connected node(optional) > Go to Settings > Provisioners > nRF Mesh Provisioner > Press back (top left arrow) > it would cause break point set in step 2 to trigger
5. Watch sequence no set by line "provisioner.setSequenceNumber(sequenceNumber);" in updateProvisioner() function. It is lower than what was noted in step 3
6. Connect to provisioned node again and try reading proxy state again, it fails ! As seq no remembered by connected node that of provisioner is lower.

Potential fix
```
public boolean updateProvisioner(@NonNull final Provisioner provisioner) {
:
:
final int sequenceNumber,node_sequenceNumber;
  if (meshNode.getUnicastAddress() != provisioner.getProvisionerAddress()) {
      sequenceNumber = sequenceNumbers.get(provisioner.getProvisionerAddress(),0);
      node_sequenceNumber = provisioner.getSequenceNumber();
  } else {
      sequenceNumber = sequenceNumbers.get(node.getUnicastAddress(), 0);
      node_sequenceNumber = node.getSequenceNumber();
  }
  if(node_sequenceNumber > sequenceNumber)
      provisioner.setSequenceNumber(node_sequenceNumber);
  else
      provisioner.setSequenceNumber(sequenceNumber);
:
:
}
```

**Platform details:**
 - Device: [e.g. Galaxy A40]
 - OS: [e.g. Android 11.0]
 - Library Version (2.1.4)

roshanrajaratnam, 2022-08-08:
Hi thanks for reporting this issue, could you kindly elaborate the scenario here so I could try to reproduce and test on my network.

pmjagtap, 2022-08-09:
Hi Roshan,

I'm not sure if this is reproducable on latest release.

Least on V2.1.4 library, I think variable "sequenceNumbers[ ]" does not have latest sequence no of the provisioner so it sets to last known(which is lower than current) after visit to current provisioner dialog on app.

Basically after returning "Current Provisioner" dialog (which calls updateProvisioner( )) parovisioner is not able to talk with provisioned node as seq(of provisioner) is outdated.

roshanrajaratnam, 2022-08-10:
ok, this could have been fixed in the later versions.

I would really recommend upgrading to the latest library as there have been a number of bug fixes and improvements and also  implements OOB provisioning as required by the mesh profile specification.

pmjagtap, 2022-08-10:
Hi,

I'm afraid issue mentioned above still exist on nRF Mesh app V 3.2.4

To Reproduce on App v 3.2.4
1.	Install nRF Mesh app from play store (V 3.2.4)
2.	Provision a mesh node
3.	To increase provisioner’s sequence no, connect to provisioned node and read its proxy state then disconnect from node
4.	Go to Settings > Provisioners > nRF Mesh Provisioner > Once on “Edit Provisioner” dialog then press back (top left arrow)
5.	Connect to provisioned node again and try reading its proxy state it would fail.
6.	It would take 10+ tries of reading proxy state to re-establish communication with node. It implies seq no of the provisioner was set to lower than actual one after returing from Edit Provisioner dialog & takes a while to increment it to actual value.

roshanrajaratnam, 2022-08-11:
Hi thanks for reporting this, I will reopn the issue and look in to it as soon as possible.

roshanrajaratnam, 2023-05-25:
Hi, sorry for the delay, I've been giving this  a test and can't seem to reproduce. Could you pull the latest changes and let us know if its still happenning?

closed 2023-05-25 by roshanrajaratnam

pmjagtap, 2023-05-26:
I'm no longer working on my last position and thus don't have access to
required resources.