nordicsemi/Android-nRF-Mesh-Library #531: GenericBatteryGet and GenericBatteryStatus

JulesDommartin, 2022-09-07:
**Is your feature request related to a problem? Please describe.**
In my project, I would like to get the battery level of my device over mesh. It would be very nice if we could send a GenericBatteryGet message and get the status.

**Describe the solution you'd like**
Would it be possible to implement it ? I tried to extend in Java som classes but I could not because of the library restrictions.

**Describe alternatives you've considered**
I tried to implement it myself within a fork but it didn't work at all.

JulesDommartin, 2022-09-20:
Any update on this ? 

I saw that the GenericBatteryGet is implemented in the iOS project. I can retrieve battery level on iOS but not on Android.

roshanrajaratnam, 2022-09-20:
Hi, I haven't had time to look in to this. If you are looking in to implementing this yourself you could simple add the support to the library and send us a pull request. You could follow the same way as it is implemented for a GenericOnOffGet.

closed 2022-09-25 by JulesDommartin