nordicsemi/Android-nRF-Mesh-Library #532: Automatically connects to Mesh

Timmy5818, 2022-09-19:
Hello, I am new to Android development. I want to make a system that automatically connects to Mesh. How do I implement it?

Or can clearly explain the current code, because the file part is too complicated, I don't know where to start.

I would like to automate the input of the user pick, the Net Key, App Key, Groups and publication settings selected in the code section.

roshanrajaratnam, 2022-09-20:
Hi, Let me answer you last question first. 

> I want to make a system that automatically connects to Mesh. How do I implement it?

I would recommend you to prompt the user to select the configuration, i.e. which network key, app key, subscription and publication settings first and start provisioning and apply the selected configuration after. 

> I would like to automate the input of the user pick, the Net Key, App Key, Groups and publication settings selected in the code section.

When connecting to a mesh network you need to scan with the Mesh Proxy service uuid and check for the network id that is advertised by the mesh devices as described in the mesh profile specification chapter 7.  Here's the ScannerRepository class that does that. 

https://github.com/NordicSemiconductor/Android-nRF-Mesh-Library/blob/fa3d03941d076409fff1c036678c85ee61be05ea/app/src/main/java/no/nordicsemi/android/nrfmesh/viewmodels/ScannerRepository.java#L71-L110

Timmy5818, 2022-09-20:
Hello, I have imported the original project into our project using the Library method. I would like to ask if you want to refer to the API description?
Because we don't know how to use the API on our side

roshanrajaratnam, 2023-05-25:
Hi you might want to look [here ](https://github.com/NordicSemiconductor/Android-nRF-Mesh-Library/blob/main/mesh/src/main/java/no/nordicsemi/android/mesh/MeshMngrApi.java)

closed 2023-05-25 by roshanrajaratnam