nordicsemi/Android-nRF-Mesh-Library #534: Unknown Access PDU Received

cjagdish, 2022-09-27:
# Problem
I am using vendor model in my node. I have subscribed vendor model with a group. When I send a read request to node then i can receive reply from node in callback. But when I try same with group address to fetch data of multiple nodes then I'm getting `Unknown Access PDU Received` message.

**Steps to Reproduce**
1. Provision multiple node having vendor model.
2. Subscribe all vendor model in same group
3. Send a read request using group address so that all node will reply

**Expected behavior**
A message should be received in callback instead of Unknown Access PDU with group read requests also.
**Platform details:**
 - nRF Mesh Android Version (3.1.8)

**Logs**
```
V/MeshTransport: Src address: 01FC
V/MeshTransport: Dst address: CA7A
V/MeshTransport: Key: 8E42AC99126640AA9AEB03769C60AEA9
V/MeshTransport: akf: 1
V/MeshTransport: aid: 22
V/MeshTransport: aszmic: 0
V/MeshTransport: Sequence number: 896
V/MeshTransport: Access message opcode: 17
V/MeshTransport: Access message parameters: 
V/AccessLayer: Created Access PDU D7030C
V/UpperTransportLayer: Application nonce: 010000038001FCCA7A00000000
V/UpperTransportLayer: Encrypted upper transport pdu: 9A0AB20946A256
V/LowerTransportLayer: Unsegmented Lower transport access PDU 569A0AB20946A256
V/NetworkLayer: Encryption key: FBAEF6094DB3C91BD8B102C50431D325
V/NetworkLayer: Privacy key: 8BC445A7E9208068C40EEB8F449226C2
V/NetworkLayer: Sequence Number: 000380
V/NetworkLayer: Encrypted Network payload: 682A9472B121B18FAADA851F65C9
V/VendorModelMessageUnackedState: Sending acknowledged vendor model message
V/BleManager: Writing characteristic 00002add-0000-1000-8000-00805f9b34fb (WRITE COMMAND)
D/BleManager: gatt.writeCharacteristic(00002add-0000-1000-8000-00805f9b34fb)
I/BleManager: Data written to 00002add-0000-1000-8000-00805f9b34fb, value: (0x) 40-49-CF-B1-D2-36-5D-B5-68-2A-94-72-B1-21-B1-8F-AA-DA-85-1F
V/BleManager: Writing characteristic 00002add-0000-1000-8000-00805f9b34fb (WRITE COMMAND)
D/BleManager: gatt.writeCharacteristic(00002add-0000-1000-8000-00805f9b34fb)
I/BleManager: Data written to 00002add-0000-1000-8000-00805f9b34fb, value: (0x) C0-65-C9
I/BleManager: Notification received from 00002ade-0000-1000-8000-00805f9b34fb, value: (0x) 40-49-47-4B-6A-57-A8-07-76-3F-CC-EC-E5-C0-E2-87-2D-EF-F2-04
I/BleManager: Notification received from 00002ade-0000-1000-8000-00805f9b34fb, value: (0x) C0-57-83-47-48-FA-54-BF-47-21-72
V/MeshManagerApi: Received network pdu: 0x0049474B6A57A807763FCCECE5C0E2872DEFF20457834748FA54BF472172
V/BaseMeshMessageHandler: Sequence number of received Network PDU: 1524
V/NetworkLayer: TTL for received message: 20
V/NetworkLayer: Dst: 0x01FC
D/LowerTransportLayer: IV Index of received message: 0
D/LowerTransportLayer: SeqAuth: 1524
V/AccessLayer: Received Access PDU D8030C14500501011E0202
V/DefaultNoOperationMessageState: Unknown Access PDU Received: D8030C14500501011E0202
```

hipal31, 2022-09-29:
@roshanrajaratnam Can you help with this?

roshanrajaratnam, 2022-09-29:
I'll look into this later today.

roshanrajaratnam, 2022-09-29:
So the issue could be that you are sending an acknowledged message to a group address. The mesh specification states that we should avoid sending acknowledged messages to a group address.

The reason behind you get the response to your message as an unknown message is that the library doesn't wait for a response for a message send to a group. This applies to Config and Generic messages as well.

cjagdish, 2022-09-29:
So what should be my next step?
Can you provide unknown message to callback?

roshanrajaratnam, 2022-09-29:
IMHO I would not send a read request to a group message. You should rather send your read requests to each individual device. The other option is that you could parse your message when the UnknownAccessPdu is called.

hipal31, 2022-09-29:
@roshanrajaratnam Thank you for providing the information. We'll find another way to archive our goal.

closed 2022-09-29 by cjagdish

roshanrajaratnam, 2022-09-29:
Happy to help!